Ysurac/openmptcprouter #508: Multipath seems to be bocked on the connection

its27, 2019-06-10:
## Specifications

  - OpenMPTCProuter version: 0.50.1
  - OpenMPTCProuter VPS version: 0.995
  - OpenMPTCProuter platform: RPI3

I have two connections, WiFi + Android USB terhered phone. Both connections work fine and MPTCP utilises both links as it should however the status page tells me multipath may be blocked?

MPTCP support results below

1: 192.168.42.129 
2: * 
3: 192.168.213.21  [PARTIAL] +PartialTCP IP::DiffServicesCP IP::TotalLength IP::TTL IP::CheckSum 
4: 192.168.213.22  [PARTIAL] +PartialTCP IP::DiffServicesCP IP::TotalLength IP::TTL IP::CheckSum 
5: * 
6: * 
7: 63.130.104.194 TCP::DataOffset TCP::CheckSum IP::DiffServicesCP IP::TotalLength IP::TTL IP::CheckSum +TCPOptionMaxSegSize 
8: 195.66.226.25  [PARTIAL] +PartialTCP IP::TotalLength IP::TTL IP::CheckSum 
9: * 
10: 178.128.169.118 TCP::SrcPort TCP::DstPort TCP::SeqNumber TCP::AckNumber TCP::DataOffset TCP::Flags TCP::WindowsSize TCP::CheckSum IP::DiffServicesCP IP::TotalLength IP::Identification IP::TTL IP::CheckSum IP::SourceIP IP::DestinationIP +TCPOptionMaxSegSize TCPOptionMPTCPCapable::Sender's Key 

Result:
< Ethernet (14 bytes) :: DestinationMAC = 16:b4:9b:21:6b:20 , SourceMAC = 0e:d0:fd:34:46:36 , Type = 0x800 , >
< IP (20 bytes) :: Version = 4 , HeaderLength = 5 , DiffServicesCP = 10 , ExpCongestionNot = 0 , TotalLength = 56 , Identification = 0x0 , Flags = 2 , FragmentOffset = 0 , TTL = 54 , Protocol = 0x6 , CheckSum = 0xfdc8 , SourceIP = 178.128.169.118 , DestinationIP = 192.168.42.48 , >
< TCP (20 bytes) :: SrcPort = 65101 , DstPort = 27156 , SeqNumber = 3347535650 , AckNumber = 1476214551 , DataOffset = 9 , Reserved = 0 , Flags = ( SYN ACK ) , WindowsSize = 28800 , CheckSum = 0x110a , UrgPointer = 0 , >
< TCPOptionMaxSegSize (4 bytes) :: Kind = 2 , Length = 4 , MaxSegSize = 1320 , >
< TCPOptionMPTCPCapable (12 bytes) :: Kind = 30 , Length = 12 , Subtype = 0 , Version = 0 , Checksum = 1 (Checksum Enabled) , Flags = 0 , Crypto = 1 (HMAC-SHA1) , Sender's Key = Sender's Key = 5017006098200662591 , >

Is the issue with the status page only?

Thanks,

Ysurac, 2019-06-10:
Multipath can't always be detected as working well. Are you sure you test it on good interface ?
It's only a warning, no issue here.

its27, 2019-06-10:
I am sure both interfaces are working, and can confirm the results below are from the interface that the status page believes there is a problem with.

closed 2019-07-04 by Ysurac