Ysurac/openmptcprouter #509: Settings no longer persist between reboots

reedog117, 2019-06-10:
## Expected Behavior

Settings persist between reboots.

## Actual Behavior

Settings no longer persist between reboots (seems like everything resets to factory defaults on every reboot)

  - Upgraded from 0.49.7 to 0.50.1 using UI
  - 0.50.1 was reset to factory defaults
  - took SD card and reflashed a factory version of 0.50.2 (squashfs) - (not the upgrade image)
  - Set up my normal settings for 0.50.2
  - Rebooted and then all settings were lost and I'm back to a clean version
  - Set a root password
  - Rebooted and then password setting is lost again

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.50.2 r0+10127-3209f5ae3d
  - OpenMPTCProuter VPS version: n/a
  - OpenMPTCProuter platform: RPi 3 B+ / 64bit

Ysurac, 2019-06-10:
You can have a faulty SD card.
Check using "mount" via SSH that FS is set as readonly.

reedog117, 2019-06-11:
This doesn't look read-only to me.
```
root@OpenMPTCProuter:~# mount
/dev/root on /rom type squashfs (ro,relatime)
devtmpfs on /rom/dev type devtmpfs (rw,relatime,size=464020k,nr_inodes=116005,mode=755)
proc on /proc type proc (rw,nosuid,nodev,noexec,noatime)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,noatime)
tmpfs on /tmp type tmpfs (rw,nosuid,nodev,noatime)
tmpfs on /tmp/root type tmpfs (rw,noatime,mode=755)
overlayfs:/tmp/root on / type overlay (rw,noatime,lowerdir=/,upperdir=/tmp/root/upper,workdir=/tmp/root/work)
/dev/mmcblk0p1 on /boot type vfat (rw,noatime,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
tmpfs on /dev type tmpfs (rw,nosuid,relatime,size=512k,mode=755)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,mode=600,ptmxmode=000)
debugfs on /sys/kernel/debug type debugfs (rw,noatime)
```

Ysurac, 2019-06-11:
True. This can happen if there is not enough free space, can you give me the result of "df -h" ?
I need also kernel log.

reedog117, 2019-06-15:
potential race condition?
```
[    7.071355] mount_root: failed to mount -t f2fs /dev/loop0 /tmp/overlay: Invalid argument
[    7.081578] mount_root: overlay filesystem has not been fully initialized yet
[    7.089877] mount_root: switching to f2fs overlay
[    7.095589] mount_root: switching to f2fs failed - fallback to ramoverlay
```
Also `df -h` - this was on a 32GB card so i don't think it's a space issue
```
root@OpenMPTCProuter:~# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                31.3M     31.3M         0 100% /rom
devtmpfs                453.1M         0    453.1M   0% /rom/dev
tmpfs                   461.3M    932.0K    460.4M   0% /tmp
tmpfs                   461.3M    180.0K    461.1M   0% /tmp/root
overlayfs:/tmp/root     461.3M    180.0K    461.1M   0% /
/dev/mmcblk0p1           19.9M     15.1M      4.8M  76% /boot
tmpfs                   512.0K         0    512.0K   0% /dev
```

closed 2019-07-04 by Ysurac