Ysurac/openmptcprouter #510: Gros problème de VPN et comportement étrange du serveur DHCP

Randdrick, 2019-06-10:
## Expected Behavior
- Pas de problèmes rencontrés avec la version 0.45 sur le VPN, qui seraient liés à GloryTUN.
- Le serveur DHCP délivrait correctement ses adresses IP sans qu'il soit besoin de le forcer comme étant le serveur principal

## Actual Behavior
Depuis la version 0.50 et supérieur, les problèmes liés au VPN sont plus en plus fréquents. Ils semblent que ces derniers soient liées à GloryTun qui est devenu complètement instable.
J'utilise depuis le départ (ou presque), GloruTun TCP, la version UDP ne fonctionnant tout simplement pas.
De plus, le serveur DHCP délivre correctement ses adresses que si il est forcé comme étant le serveur DHCP principal. 

## Specifications

  - OpenMPTCProuter version: (V 0.52)
  - OpenMPTCProuter VPS version: (V 0.995)
  - OpenMPTCProuter platform: (RPI3)

Voici mes derniers logs :

Mon Jun 10 20:36:05 2019 daemon.err glorytun[4625]: read: Operation timed out
Mon Jun 10 20:36:05 2019 daemon.info glorytun[4625]: STOPPED tun0
Mon Jun 10 20:36:22 2019 daemon.err glorytun[4625]: 188.213.31.131.65001: connected
Mon Jun 10 20:36:32 2019 daemon.err glorytun[4625]: 188.213.31.131.65001: key exchange failed
Mon Jun 10 20:36:32 2019 daemon.info glorytun[4625]: STOPPED tun0
Mon Jun 10 20:36:33 2019 daemon.err glorytun[4625]: 188.213.31.131.65001: connected
Mon Jun 10 20:36:43 2019 daemon.err glorytun[4625]: 188.213.31.131.65001: key exchange failed
Mon Jun 10 20:36:43 2019 daemon.info glorytun[4625]: STOPPED tun0
Mon Jun 10 20:36:44 2019 daemon.err glorytun[4625]: 188.213.31.131.65001: connected
Mon Jun 10 20:36:54 2019 daemon.err glorytun[4625]: 188.213.31.131.65001: key exchange failed
Mon Jun 10 20:36:54 2019 daemon.info glorytun[4625]: STOPPED tun0
Mon Jun 10 20:36:55 2019 daemon.err glorytun[4625]: 188.213.31.131.65001: connected
Mon Jun 10 20:37:05 2019 daemon.err glorytun[4625]: 188.213.31.131.65001: key exchange failed
Mon Jun 10 20:37:05 2019 daemon.info glorytun[4625]: STOPPED tun0
Mon Jun 10 20:37:06 2019 daemon.err glorytun[4625]: 188.213.31.131.65001: connected
Mon Jun 10 20:37:16 2019 daemon.err glorytun[4625]: 188.213.31.131.65001: key exchange failed
Mon Jun 10 20:37:16 2019 daemon.info glorytun[4625]: STOPPED tun0
Mon Jun 10 20:37:17 2019 daemon.err glorytun[4625]: 188.213.31.131.65001: connected
Mon Jun 10 20:37:27 2019 daemon.err glorytun[4625]: 188.213.31.131.65001: key exchange failed
Mon Jun 10 20:37:27 2019 daemon.info glorytun[4625]: STOPPED tun0
Mon Jun 10 20:37:28 2019 daemon.err glorytun[4625]: 188.213.31.131.65001: connected
Mon Jun 10 20:37:38 2019 daemon.err glorytun[4625]: 188.213.31.131.65001: key exchange failed
Mon Jun 10 20:37:38 2019 daemon.info glorytun[4625]: STOPPED tun0
Mon Jun 10 20:37:39 2019 daemon.err glorytun[4625]: 188.213.31.131.65001: connected
Mon Jun 10 20:37:49 2019 daemon.err glorytun[4625]: 188.213.31.131.65001: key exchange failed
Mon Jun 10 20:37:49 2019 daemon.info glorytun[4625]: STOPPED tun0
Mon Jun 10 20:37:50 2019 daemon.err glorytun[4625]: 188.213.31.131.65001: connected
Mon Jun 10 20:38:00 2019 daemon.err glorytun[4625]: 188.213.31.131.65001: key exchange failed
Mon Jun 10 20:38:00 2019 daemon.info glorytun[4625]: STOPPED tun0
Mon Jun 10 20:38:41 2019 daemon.err glorytun[4625]: 188.213.31.131.65001: connected
Mon Jun 10 20:38:50 2019 daemon.info glorytun[4625]: STARTED tun0

Ysurac, 2019-06-10:
Pour le serveur DHCP, si il y a plusieurs serveurs sur le réseau c'est pas forcément étonnant.
Sinon pour le soucis avec glorytun il faut voir du côté du VPS, dans /var/log/daemon.log
Il faut aussi vérifier la configuration de glorytun tcp si il est bien sur le VPS configuré en chacha20.

Randdrick, 2019-06-10:
Pour le DHCP, c'est le seul serveur. Je désactive le DHCP de ma box avant chaque mise à jour pour être sûr que cela soit bien le routeur qui donne les adresses IP. Or, depuis la V 0.50, si je force pas le serveur DHCP comme serveur principal, il ne distribue pas d'adresse Ip.

Concernant GloryTun, il est configuré en chacha20, aussi côté client que VPS (et cela depuis le début).
D'ailleurs, côté serveur j'ai eu :  (quelques logs significatifs, je l'espère)

Jun 10 19:50:00 vps-42566 ss-server[12455]: 2019/06/10 19:50:00 tcp:2.6.80.105:3
5773 accepted tcp:127.0.0.1:0
Jun 10 19:50:02 vps-42566 ss-server[12455]: 2019/06/10 19:50:02 tcp:2.6.80.105:2
7811 accepted tcp:127.0.0.1:0
Jun 10 19:50:06 vps-42566 ss-server[12455]: 2019/06/10 19:50:06 tcp:2.6.80.105:3
5799 accepted tcp:127.0.0.1:0
Jun 10 19:50:07 vps-42566 ss-server[12455]: 2019/06/10 19:50:07 tcp:2.6.80.105:3
5807 accepted tcp:127.0.0.1:0
Jun 10 19:50:10 vps-42566 ss-server[12455]: 2019/06/10 19:50:10 tcp:193.56.243.7
:40125 accepted tcp:127.0.0.1:0
Jun 10 19:50:17 vps-42566 ss-server[12455]: 2019/06/10 19:50:17 tcp:2.6.80.105:3
5807 accepted tcp:127.0.0.1:0
Jun 10 19:50:25 vps-42566 ss-server[12455]: 2019/06/10 19:50:25 tcp:2.6.80.105:3
5691 accepted tcp:127.0.0.1:0
Jun 10 19:50:27 vps-42566 ss-server[12455]: 2019/06/10 19:50:27 tcp:2.6.80.105:3
5825 accepted tcp:127.0.0.1:0
Jun 10 19:50:28 vps-42566 ss-server[12455]:  2019-06-10 19:50:28 ERROR: remote r
ecv: Connection reset by peer
Jun 10 19:50:29 vps-42566 ss-server[12455]:  2019-06-10 19:50:29 ERROR: remote r
ecv: Connection reset by peer
Jun 10 19:50:37 vps-42566 ss-server[12455]: 2019/06/10 19:50:37 tcp:2.6.80.105:3
5825 accepted tcp:127.0.0.1:0
Jun 10 19:50:43 vps-42566 ss-server[12455]:  2019-06-10 19:50:43 ERROR: remote r
ecv: Connection reset by peer
Jun 10 19:50:45 vps-42566 ss-server[12455]:  2019-06-10 19:50:45 ERROR: remote r
ecv: Connection reset by peer
Jun 10 19:50:45 vps-42566 ss-server[12455]:  2019-06-10 19:50:45 ERROR: remote r
ecv: Connection reset by peer
Jun 10 19:50:45 vps-42566 ss-server[12455]: 2019/06/10 19:50:45 tcp:2.6.80.105:3
5449 accepted tcp:127.0.0.1:0
Jun 10 19:50:47 vps-42566 ss-server[12455]: 2019/06/10 19:50:47 tcp:2.6.80.105:3
5841 accepted tcp:127.0.0.1:0
Jun 10 19:50:49 vps-42566 ss-server[12455]:  2019-06-10 19:50:49 ERROR: remote r
ecv: Connection reset by peer
Jun 10 19:50:57 vps-42566 ss-server[12455]: 2019/06/10 19:50:57 tcp:2.6.80.105:3
5585 accepted tcp:127.0.0.1:0
Jun 10 19:50:57 vps-42566 ss-server[12455]: 2019/06/10 19:50:57 tcp:2.6.80.105:3
5841 accepted tcp:127.0.0.1:0
Jun 10 19:51:05 vps-42566 ss-server[12455]: 2019/06/10 19:51:05 tcp:2.6.80.105:3
5853 accepted tcp:127.0.0.1:0
Jun 10 19:51:07 vps-42566 ss-server[12455]: 2019/06/10 19:51:07 tcp:2.6.80.105:3
5861 accepted tcp:127.0.0.1:0
Jun 10 19:51:10 vps-42566 ss-server[12455]:  2019-06-10 19:51:10 ERROR: remote r
ecv: Connection reset by peer
Jun 10 19:51:13 vps-42566 ss-server[12455]:  2019-06-10 19:51:13 ERROR: remote r
ecv: Connection reset by peer
Jun 10 19:51:14 vps-42566 ss-server[12455]:  2019-06-10 19:51:14 ERROR: remote r
ecv: Connection reset by peer
Jun 10 19:51:16 vps-42566 ss-server[12455]:  2019-06-10 19:51:16 ERROR: remote r
ecv: Connection reset by peer
Jun 10 19:51:18 vps-42566 ss-server[12455]: 2019/06/10 19:51:18 tcp:2.6.80.105:3
5861 accepted tcp:127.0.0.1:0
Jun 10 19:51:19 vps-42566 ss-server[12455]:  2019-06-10 19:51:19 ERROR: remote r
ecv: Connection reset by peer
Jun 10 19:51:19 vps-42566 ss-server[12455]:  2019-06-10 19:51:19 ERROR: remote r
ecv: Connection reset by peer
Jun 10 19:51:21 vps-42566 ss-server[12455]:  2019-06-10 19:51:21 ERROR: remote r
ecv: Connection reset by peer
////////////////////////////////////////////////////////////////
Jun 10 19:41:17 vps-42566 glorytun-tcp-run[16553]: 2.6.80.105.28317: connected
Jun 10 19:41:17 vps-42566 glorytun-tcp-run[16553]: 2.6.80.105.28317: key exchang
e failed
Jun 10 19:41:18 vps-42566 glorytun-tcp-run[16553]: 2.6.80.105.28369: connected
Jun 10 19:41:18 vps-42566 glorytun-tcp-run[16553]: 2.6.80.105.28369: key exchang
e failed
[... 53 lines cut ...]
Jun 10 19:55:18 vps-42566 systemd-networkd[23815]: Enumeration completed
Jun 10 19:55:18 vps-42566 systemd[1]: Started Network Service.
Jun 10 19:55:18 vps-42566 named[17497]: no longer listening on 10.255.255.1#53
Jun 10 19:55:18 vps-42566 named[17497]: listening on IPv4 interface gt-tun0, 10.
255.255.1#53
Jun 10 19:55:18 vps-42566 named[17497]: no longer listening on 10.255.252.1#53
Jun 10 19:55:18 vps-42566 named[17497]: listening on IPv4 interface tun0, 10.255
.252.1#53
Jun 10 19:55:18 vps-42566 named[17497]: no longer listening on 10.255.253.1#53
Jun 10 19:55:18 vps-42566 named[17497]: listening on IPv4 interface mlvpn0, 10.2
55.253.1#53
Jun 10 19:55:18 vps-42566 named[17497]: no longer listening on 10.255.254.1#53
Jun 10 19:55:18 vps-42566 named[17497]: listening on IPv4 interface gt-udp-tun0,
 10.255.254.1#53
Jun 10 19:55:19 vps-42566 glorytun-tcp-run[23797]: 2.6.80.105.30485: connected
Jun 10 19:55:21 vps-42566 ss-server[12455]: 2019/06/10 19:55:21 tcp:2.6.80.105:3
6205 accepted tcp:127.0.0.1:0
Jun 10 19:55:25 vps-42566 ss-server[12455]: 2019/06/10 19:55:25 tcp:2.6.80.105:3
6217 accepted tcp:127.0.0.1:0
Jun 10 19:55:31 vps-42566 systemd-networkd[23815]: gt-tun0: Configured
Jun 10 19:55:31 vps-42566 systemd-networkd[23815]: gt-udp-tun0: Configured
Jun 10 19:55:31 vps-42566 systemd-networkd[23815]: mlvpn0: Configured
Jun 10 19:55:31 vps-42566 systemd-networkd[23815]: tun0: Configured
Jun 10 19:55:32 vps-42566 ss-server[12455]: 2019/06/10 19:55:32 tcp:2.6.80.105:3
6225 accepted tcp:127.0.0.1:0

Ysurac, 2019-06-10:
dnsmasq is the DHCP server, and it wasn't updated for latest releases.
Il me faudrait toute la configuration, je pense à un soucis de route.
Sinon il faudrait refaire une configuration depuis 0 sur OpenMPTCProuter pour voir si ça corrige.

Randdrick, 2019-06-11:
Bonjou Ysurac,
Voici toute la configuration. Pour des raison de sécurité, j'ai supprimer les clefs.

**Tous les paramètres d'OpenMPTCProuter**
ddns.global=ddns
ddns.global.ddns_dateformat='%F %R'
ddns.global.ddns_loglines='250'
ddns.global.upd_privateip='0'
ddns.myddns_ipv4=service
ddns.myddns_ipv4.lookup_host='yourhost.example.com'
ddns.myddns_ipv4.domain='yourhost.example.com'
ddns.myddns_ipv4.username='your_username'
ddns.myddns_ipv4.password='your_password'
ddns.myddns_ipv4.interface='wan'
ddns.myddns_ipv4.ip_source='network'
ddns.myddns_ipv4.ip_network='wan'
ddns.myddns_ipv4.service_name='dyn.com'
ddns.myddns_ipv6=service
ddns.myddns_ipv6.update_url='http://[USERNAME]:[PASSWORD]@your.provider.net/nic/update?hostname=[DOMAIN]&myip=[IP]'
ddns.myddns_ipv6.lookup_host='yourhost.example.com'
ddns.myddns_ipv6.domain='yourhost.example.com'
ddns.myddns_ipv6.username='your_username'
ddns.myddns_ipv6.password='your_password'
ddns.myddns_ipv6.use_ipv6='1'
ddns.myddns_ipv6.interface='wan6'
ddns.myddns_ipv6.ip_source='network'
ddns.myddns_ipv6.ip_network='wan6'
dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].server='127.0.0.1#5353'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].quietdhcp='1'
dhcp.@dnsmasq[0].notinterface='wan1'
dhcp.@dnsmasq[0].ipset='/googlevideo.com/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googlevideo.com/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googlevideo.com/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.cot/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.cot/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.cot/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr-cs2,omr_dscp-cs2,omr_dscp6-cs2' '/googletagmanager.com/omr-cs2,omr_dscp-cs2,omr_dscp6-cs2' '/googletagmanager.com/omr-cs2,omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr-cs2,omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr-cs2,omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr-cs2,omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr-cs2,omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr-cs2,omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr-cs2,omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr-cs2,omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr-cs2,omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr-cs2,omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr-cs2,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr-cs2,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr-cs2,omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr-cs2,omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr-cs2,omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr-cs2,omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr-cs2,omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr-cs2,omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr-cs2,omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr-cs2,omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr-cs2,omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr-cs2,omr_dscp-cs2,omr_dscp6-cs2' '/1e100.net/omr-cs2,omr_dscp-cs2,o [... cut ...]
vnstat.@vnstat[0].interface='wan1' 'wan2'
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.channel='11'
wireless.radio0.hwmode='11g'
wireless.radio0.path='platform/soc/3f300000.mmc/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
wireless.radio0.htmode='HT20'
wireless.radio0.disabled='1'

================================================
J'en profite pour te joindre aussi les routes actives IPv4 (je n'utilise pas l'IPv6). Pour ma part, je n'ai pas défini de routes autres que celles prévues par le système :

[image]

Ysurac, 2019-06-11:
Le VPS ne semble pas avoir été redémarré après la mise à jour. Tu peux déjà essayer et vérifier que le kernel est bien le 4.14.122

Randdrick, 2019-06-11:
En effet, je n'avais pas redémarré le VPS. Chose faite désormais. 
Il est bien à la version 4.14.122.

Je te tiens au courant dans le cas ou le problème persiste. merci pour ton aide.

Randdrick, 2019-06-12:
Bonjour,
Bon malgré le reboot du serveur, le problème persiste. Voici ce que j'ai relevé dans mon journal aujourd'hui : 

Wed Jun 12 15:09:38 2019 daemon.err glorytun[4625]: read: Operation timed out
Wed Jun 12 15:09:38 2019 daemon.info glorytun[4625]: STOPPED tun0
Wed Jun 12 15:10:04 2019 daemon.err glorytun[4625]: 188.213.31.131.65001: connected
Wed Jun 12 15:10:09 2019 user.notice post-tracking-post-tracking: Change master interface from wan2 (201 ms) to wan1 (36 ms)
Wed Jun 12 15:10:14 2019 daemon.err glorytun[4625]: 188.213.31.131.65001: key exchange failed
Wed Jun 12 15:10:14 2019 daemon.info glorytun[4625]: STOPPED tun0
Wed Jun 12 15:10:15 2019 daemon.err glorytun[4625]: 188.213.31.131.65001: connected
Wed Jun 12 15:10:17 2019 user.notice post-tracking-post-tracking: Set server vps (188.213.31.131) default route via 192.168.1.1
Wed Jun 12 15:10:25 2019 daemon.err glorytun[4625]: 188.213.31.131.65001: key exchange failed
Wed Jun 12 15:10:25 2019 daemon.info glorytun[4625]: STOPPED tun0
Wed Jun 12 15:10:26 2019 daemon.err glorytun[4625]: 188.213.31.131.65001: connected
Wed Jun 12 15:10:36 2019 daemon.err glorytun[4625]: 188.213.31.131.65001: key exchange failed
Wed Jun 12 15:10:36 2019 daemon.info glorytun[4625]: STOPPED tun0
Wed Jun 12 15:10:37 2019 daemon.err glorytun[4625]: 188.213.31.131.65001: connected
Wed Jun 12 15:10:47 2019 daemon.err glorytun[4625]: 188.213.31.131.65001: key exchange failed
Wed Jun 12 15:10:47 2019 daemon.info glorytun[4625]: STOPPED tun0
Wed Jun 12 15:10:49 2019 daemon.err glorytun[4625]: 188.213.31.131.65001: connected
Wed Jun 12 15:10:59 2019 daemon.err glorytun[4625]: 188.213.31.131.65001: key exchange failed
Wed Jun 12 15:10:59 2019 daemon.info glorytun[4625]: STOPPED tun0
Wed Jun 12 15:11:00 2019 daemon.err glorytun[4625]: 188.213.31.131.65001: connected
Wed Jun 12 15:11:10 2019 daemon.err glorytun[4625]: 188.213.31.131.65001: key exchange failed
Wed Jun 12 15:11:10 2019 daemon.info glorytun[4625]: STOPPED tun0
Wed Jun 12 15:11:11 2019 daemon.err glorytun[4625]: 188.213.31.131.65001: connected
Wed Jun 12 15:11:21 2019 daemon.err glorytun[4625]: 188.213.31.131.65001: key exchange failed
Wed Jun 12 15:11:21 2019 daemon.info glorytun[4625]: STOPPED tun0
Wed Jun 12 15:11:22 2019 daemon.err glorytun[4625]: 188.213.31.131.65001: connected
Wed Jun 12 15:11:32 2019 daemon.err glorytun[4625]: 188.213.31.131.65001: key exchange failed
Wed Jun 12 15:11:32 2019 daemon.info glorytun[4625]: STOPPED tun0
Wed Jun 12 15:11:33 2019 daemon.err glorytun[4625]: 188.213.31.131.65001: connected
Wed Jun 12 15:11:42 2019 daemon.info glorytun[4625]: STARTED tun0
Wed Jun 12 15:25:43 2019 daemon.info glorytun[4625]: STOPPED tun0
Wed Jun 12 15:25:44 2019 daemon.err glorytun[4625]: 188.213.31.131.65001: connected
Wed Jun 12 15:25:44 2019 daemon.info glorytun[4625]: STARTED tun0

Il doit y avoir quelque chose qui ne plait pas à Glorytun.

Ysurac, 2019-06-12:
Ca semble être quand OpenMPTCProuter change d'interface maitre.
Dans Systéme->OpenMPTCProuter puis paramètres avancés en mettant "Selection de l'interface maître" à autre chose que dynamique ça devrait limiter le soucis, au moins celui du "key exchange failed".
Sinon le soucis est ailleurs...

Randdrick, 2019-06-12:
Avant la version 0.50, je n'avais pas de problèmes avec le mode dynamique, dans le changement d'interface. Peut-être est-ce une piste ?

Dans le log, glorytun passe en timeout avant le changement d'interface : 

**Wed Jun 12 15:09:38 2019 daemon.err glorytun[4625]: read: Operation timed out**
Wed Jun 12 15:09:38 2019 daemon.info glorytun[4625]: STOPPED tun0
Wed Jun 12 15:10:04 2019 daemon.err glorytun[4625]: 188.213.31.131.65001: connected
Wed Jun 12 15:10:09 2019 user.notice post-tracking-post-tracking: Change master interface from wan2 (201 ms) to wan1 (36 ms)

C’est pour cela que je pense à un problème au niveau de Glorytun

Ysurac, 2019-06-12:
Il passe en timeout car l'interface wan2 répond mal, le tracker doit avoir un timeout un peu plus élevé, je vais augmenter un peu celui de glorytun tcp dans la prochaine version. Mais ce n'est pas réellement le soucis.

closed 2019-07-04 by Ysurac