Ysurac/openmptcprouter #511: Erreur dans journal system

bosseb, 2019-06-11:
- OpenMPTCProuter version: 0.50.2
  - OpenMPTCProuter VPS version: 0.995
  - OpenMPTCProuter platform: x86_64

j'ai ces erreurs dans mon journal systeme :

Tue Jun 11 18:29:49 2019 daemon.err miniupnpd[5092]: ioctl(s, SIOCGIFADDR, ...): Address not available
Tue Jun 11 18:29:49 2019 daemon.err miniupnpd[5092]: ioctl(s, SIOCGIFADDR, ...): Address not available
Tue Jun 11 18:29:49 2019 daemon.err miniupnpd[5092]: ioctl(s, SIOCGIFADDR, ...): Address not available
Tue Jun 11 18:29:49 2019 daemon.err miniupnpd[5092]: Failed to get ip address for interface 6in4-omr6in4
Tue Jun 11 18:29:50 2019 daemon.err miniupnpd[5092]: ioctl(s, SIOCGIFADDR, ...): Address not available
Tue Jun 11 18:29:50 2019 daemon.err miniupnpd[5092]: Failed to get ip address for interface 6in4-omr6in4
Tue Jun 11 18:29:51 2019 daemon.err miniupnpd[5092]: ioctl(s, SIOCGIFADDR, ...): Address not available
Tue Jun 11 18:29:51 2019 daemon.err miniupnpd[5092]: ioctl(s, SIOCGIFADDR, ...): Address not available
Tue Jun 11 18:29:51 2019 daemon.err miniupnpd[5092]: ioctl(s, SIOCGIFADDR, ...): Address not available
Tue Jun 11 18:29:51 2019 daemon.err miniupnpd[5092]: Failed to get ip address for interface 6in4-omr6in4
Tue Jun 11 18:29:51 2019 daemon.err miniupnpd[5092]: ioctl(s, SIOCGIFADDR, ...): Address not available
Tue Jun 11 18:29:51 2019 daemon.err miniupnpd[5092]: Failed to get ip address for interface 6in4-omr6in4
Tue Jun 11 18:29:51 2019 daemon.err miniupnpd[5092]: ioctl(s, SIOCGIFADDR, ...): Address not available
Tue Jun 11 18:29:51 2019 daemon.err miniupnpd[5092]: ioctl(s, SIOCGIFADDR, ...): Address not available
Tue Jun 11 18:29:51 2019 daemon.err miniupnpd[5092]: ioctl(s, SIOCGIFADDR, ...): Address not available
Tue Jun 11 18:29:51 2019 daemon.err miniupnpd[5092]: Failed to get ip address for interface 6in4-omr6in4
Tue Jun 11 18:29:51 2019 daemon.err miniupnpd[5092]: ioctl(s, SIOCGIFADDR, ...): Address not available
Tue Jun 11 18:29:51 2019 daemon.err miniupnpd[5092]: Failed to get ip address for interface 6in4-omr6in4
Tue Jun 11 18:29:52 2019 daemon.err miniupnpd[5092]: ioctl(s, SIOCGIFADDR, ...): Address not available
Tue Jun 11 18:29:52 2019 daemon.err miniupnpd[5092]: ioctl(s, SIOCGIFADDR, ...): Address not available
Tue Jun 11 18:29:52 2019 daemon.err miniupnpd[5092]: ioctl(s, SIOCGIFADDR, ...): Address not available
Tue Jun 11 18:29:52 2019 daemon.err miniupnpd[5092]: Failed to get ip address for interface 6in4-omr6in4

Ysurac, 2019-06-11:
C'est UPnP qui provoque l'erreur (il est fortement recommandé de ne pas activer cette horreur).
Sinon, vu que ça semble lié à l'interface 6in4-omr6in4 est-ce que l'IPv6 est activé ou pas ?

bosseb, 2019-06-11:
je viens donc de désactiver upnp les erreurs ont disparus .. ipv6 tjrs désactivé

Ysurac, 2019-06-11:
Il faudrait voir si les commandes suivantes permettent bien de plus avoir d'erreur tout en ayant UPnP qui fonctionne :
```
uci set upnpd.config.external_iface='omrvpn'
uci commit upnpd
```

closed 2019-07-04 by Ysurac