Ysurac/openmptcprouter #514: VPN IP (ovh) blocked by service

audren49, 2019-06-16:
Just a general question. All seem to work fine. But some services, like canal+ are restricting ovh IP.
How to get around this?

Thanks;

Ysurac, 2019-06-16:
You have to use omr-bypass, this will use a wan directly.

audren49, 2019-06-16:
It works, thanks.
Is there a way to include all subdomains?
Like *. 1fichier.com?

Thank you

Ysurac, 2019-06-16:
If you put "example.com" this will also bypass "www.example.com" or "sub.example.com", so any subdomains.

audren49, 2019-06-17:
Hi, did some further test.
Subdomains are not bypassed when just the domain is setup.

Easy test is jeuxvideo.com which blocks the OVH IP. the website loads its images on "images.jeuxvideo.com"
If this is not setup explicitely in the bypass table like this :
[image]

Then images will not load with a 403

Ysurac, 2019-06-17:
subdomains are really bypassed. Check that OpenMPTCProuter is used as DNS.

audren49, 2019-06-17:
That was the issue indeed. My routeur used 8.8.8.8 as DNS.
When set to OMR local IP, all works fine.

Thank you very much.

closed 2019-06-17 by audren49