Ysurac/openmptcprouter #515: 2 routeur, 2 wan, clients just use 1 wan

audren49, 2019-06-17:
## Expected Behavior

Speed should be the aggregated of wan1+wan2 (same speed adsl).

## Actual Behavior

Clients only get the speed of one of the wan

## Steps to Reproduce the Problem

- I use Google Wifi, set on 192.168.87.X . Wan set to 192.168.100.7 with gateway 192.168.100.1
- Google wifi is the only device which has the DHCP server turned on.
- OMR is kept on 192.168.100.1

Google Wifi speed test gets the aggregated speed. OMR gets the aggregated speed (from the iperf util in the interface). Clients don't get it.

shadowsocks transfert all ports (under advanced settings). Other than that, OMR settings are the originals one.
 

## Specifications

  - OpenMPTCProuter version: 50.2
  - OpenMPTCProuter VPS version: 0.995
  - OpenMPTCProuter platform: RPI3+

Ysurac, 2019-06-17:
WAN is on same network as OMR, this can't work. You need to set each wan on own network and use ip of each box as gateway.

audren49, 2019-06-17:
Thanks,

Box1 is 192.168.0.1
Box 2 is 192.168.1.254

In google wifi config you can either set WAN as based on DHCP or have it fixed.
if fixed then you cannot have it as your external IP address (vps IP) because the gateway will then not be in the same sub network. So in google wifi the config looks like this : 
- static WAN to 192.168.100.7
- gateway 192.168.100.1

Ysurac, 2019-06-17:
You use Google wifi as provider ? If yes, this config can't use it to aggregate. Config must be like other box, with a WAN on OMR set in same network.

audren49, 2019-06-17:
Google WiFi is only the routeur, with dhcp

Ysurac, 2019-06-17:
ok, I understand. So if Google wifi get full speed, there is no real reason clients of it can't get it too.

audren49, 2019-06-17:
Indeed...
I disabled some other settings on one of the box, and it seems to be
working fine now.
Let's see if that endures.

The settings was related to ipv6 and somethings called ula. Also disabled
upnp on those boxes.

I had already got it worked in the last few days but then eventually the
merge for the clients failed.
I'll keep you posted.
Thanks for this amazing tool.

audren49, 2019-06-19:
All good. Everything perfect. Thank you so much.

closed 2019-06-19 by audren49