Ysurac/openmptcprouter #516: Stability problem

JorgeMoort89, 2019-06-17:
Internet is very slow and seems like there are some problem on wan3, previously I've added a second connection on wan2 that had a lot of problem (the modem rebotted itself every hour), can be Glorytun the problem?

Those are the system log:
`Mon Jun 17 17:16:50 2019 kern.info kernel: [    0.613768] VFS: Mounted root (squashfs filesystem) readonly on device 8:2.
Mon Jun 17 17:16:50 2019 kern.info kernel: [    0.615107] devtmpfs: mounted
Mon Jun 17 17:16:50 2019 kern.info kernel: [    0.615633] Freeing unused kernel memory: 1064K
Mon Jun 17 17:16:50 2019 kern.info kernel: [    0.660028] usb 1-1: new high-speed USB device number 2 using ehci-pci
Mon Jun 17 17:16:50 2019 kern.info kernel: [    0.700038] usb 2-1: new high-speed USB device number 2 using ehci-pci
Mon Jun 17 17:16:50 2019 kern.info kernel: [    0.700045] Write protecting the kernel read-only data: 16384k
Mon Jun 17 17:16:50 2019 kern.info kernel: [    0.700533] Freeing unused kernel memory: 2016K
Mon Jun 17 17:16:50 2019 kern.info kernel: [    0.703081] Freeing unused kernel memory: 1556K
Mon Jun 17 17:16:50 2019 user.info kernel: [    0.760291] init: Console is alive
Mon Jun 17 17:16:50 2019 kern.info kernel: [    0.860448] hub 1-1:1.0: USB hub found
Mon Jun 17 17:16:50 2019 kern.info kernel: [    0.860553] hub 1-1:1.0: 6 ports detected
Mon Jun 17 17:16:50 2019 kern.info kernel: [    0.900446] hub 2-1:1.0: USB hub found
Mon Jun 17 17:16:50 2019 kern.info kernel: [    0.900553] hub 2-1:1.0: 6 ports detected
Mon Jun 17 17:16:50 2019 user.info kernel: [    1.055258] kmodloader: loading kernel modules from /etc/modules-boot.d/*
Mon Jun 17 17:16:50 2019 kern.info kernel: [    1.065246] pps_core: LinuxPPS API ver. 1 registered
Mon Jun 17 17:16:50 2019 kern.info kernel: [    1.065250] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
Mon Jun 17 17:16:50 2019 kern.info kernel: [    1.065634] PTP clock support registered
Mon Jun 17 17:16:50 2019 kern.info kernel: [    1.066432] Button Hotplug driver version 0.4.1
Mon Jun 17 17:16:50 2019 kern.info kernel: [    1.066672] ehci-platform: EHCI generic platform driver
Mon Jun 17 17:16:50 2019 kern.info kernel: [    1.066941] sdhci-pltfm: SDHCI platform and OF driver helper
Mon Jun 17 17:16:50 2019 user.info kernel: [    1.066967] kmodloader: done loading kernel modules from /etc/modules-boot.d/*
Mon Jun 17 17:16:50 2019 user.info kernel: [    1.072418] init: - preinit -
Mon Jun 17 17:16:50 2019 kern.notice kernel: [    1.175817] random: jshn: uninitialized urandom read (4 bytes read)
Mon Jun 17 17:16:50 2019 kern.notice kernel: [    1.178622] random: jshn: uninitialized urandom read (4 bytes read)
Mon Jun 17 17:16:50 2019 kern.notice kernel: [    1.179764] random: jshn: uninitialized urandom read (4 bytes read)
Mon Jun 17 17:16:50 2019 kern.info kernel: [    1.270010] tsc: Refined TSC clocksource calibration: 3092.973 MHz
Mon Jun 17 17:16:50 2019 kern.info kernel: [    1.270023] clocksource: tsc: mask: 0xffffffffffffffff max_cycles: 0x2c95594edb3, max_idle_ns: 440795220223 ns
Mon Jun 17 17:16:50 2019 kern.info kernel: [    2.280207] clocksource: Switched to clocksource tsc
Mon Jun 17 17:16:50 2019 kern.warn kernel: [    4.198028] urandom_read: 4 callbacks suppressed
Mon Jun 17 17:16:50 2019 kern.notice kernel: [    4.198029] random: jshn: uninitialized urandom read (4 bytes read)
Mon Jun 17 17:16:50 2019 kern.notice kernel: [    4.215941] random: procd: uninitialized urandom read (4 bytes read)
Mon Jun 17 17:16:50 2019 kern.notice kernel: [    4.331523] F2FS-fs (loop0): Mounted with checkpoint version = 3398767d
Mon Jun 17 17:16:50 2019 user.info kernel: [    4.332048] mount_root: switching to f2fs overlay
Mon Jun 17 17:16:50 2019 user.warn kernel: [    4.400539] urandom-seed: Seeding with /etc/urandom.seed
Mon Jun 17 17:16:50 2019 user.info kernel: [    4.415839] procd: - early -
Mon Jun 17 17:16:50 2019 kern.notice kernel: [    4.564548] random: jshn: uninitialized urandom read (4 bytes read)
Mon Jun 17 17:16:50 2019 user.info kernel: [    5.030056] procd: - ubus -
Mon Jun 17 17:16:50 2019 user.info kernel: [    5.079806] procd: - init -
Mon Jun 17 17:16:50 2019 user.info kernel: [    5.144080] kmodloader: loading kernel modules from /etc/modules.d/*
Mon Jun 17 17:16:50 2019 kern.info kernel: [    5.153256] natsemi dp8381x driver, version 2.1, Sept 11, 2006
Mon Jun 17 17:16:50 2019 kern.info kernel: [    5.153256]   originally by Donald Becker <becker@scyld.com>
Mon Jun 17 17:16:50 2019 kern.info kernel: [    5.153256]   2.4.x kernel port by Jeff Garzik, Tjeerd Mulder
Mon Jun 17 17:16:50 2019 kern.info kernel: [    5.153745] NET: Registered protocol family 8
Mon Jun 17 17:16:50 2019 kern.info kernel: [    5.153748] NET: Registered protocol family 20
Mon Jun 17 17:16:50 2019 kern.info kernel: [    5.154342] Modular ISDN core version 1.1.29
Mon Jun 17 17:16:50 2019 kern.info kernel: [    5.154376] NET: Registered protocol family 34
Mon Jun 17 17:16:50 2019 kern.info kernel: [    5.154533] mISDN: Layer-1-over-IP driver Rev. 2.00
Mon Jun 17 17:16:50 2019 kern.info kernel: [    5.154681] 0 virtual devices registered
Mon Jun 17 17:16:50 2019 kern.info kernel: [    5.154870] DSP module 2.0
Mon Jun 17 17:16:50 2019 kern.info kernel: [    5.154872] mISDN_dsp: DSP clocks every 80 samples. This equals 1 jiffies.
Mon Jun 17 17:16:50 2019 kern.info kernel: [    5.155894] tun: Universal TUN/TAP device driver, 1.6
Mon Jun 17 17:16:50 2019 kern.info kernel: [    5.156265] mISDN: HFC-multi driver 2.03
Mon Jun 17 17:16:50 2019 kern.info kernel: [    5.157358] sctp: Hash tables configured (bind 256/256)
Mon Jun 17 17:16:50 2019 kern.info kernel: [    5.157594] sit: IPv6, IPv4 and MPLS over IPv4 tunneling driver
Mon Jun 17 17:16:50 2019 kern.info kernel: [    5.158207] e1000: Intel(R) PRO/1000 Network Driver - version 7.3.21-k8-NAPI
Mon Jun 17 17:16:50 2019 kern.info kernel: [    5.158210] e1000: Copyright (c) 1999-2006 Intel Corporation.
Mon Jun 17 17:16:50 2019 kern.info kernel: [    5.158566] igb: Intel(R) Gigabit Ethernet Network Driver - version 5.4.0-k
Mon Jun 17 17:16:50 2019 kern.info kernel: [    5.158568] igb: Copyright (c) 2007-2014 Intel Corporation.
Mon Jun 17 17:16:50 2019 kern.notice kernel: [    5.160530] ISDN subsystem Rev: 1.1.2.3/
Mon Jun 17 17:16:50 2019 kern.warn kernel: [    5.160531] 1.1.2.2/
Mon Jun 17 17:16:50 2019 kern.warn kernel: [    5.160534] 1.1.2.3/
Mon Jun 17 17:16:50 2019 kern.warn kernel: [    5.160535] 1.1.2.2/
Mon Jun 17 17:16:50 2019 kern.warn kernel: [    5.160536] 1.1.2.2
Mon Jun 17 17:16:50 2019 kern.warn kernel: [    5.160537]  loaded
Mon Jun 17 17:16:50 2019 kern.info kernel: [    5.160838] dss1_divert module successfully installed
Mon Jun 17 17:16:50 2019 kern.info kernel: [    5.160937] PPP BSD Compression module registered
Mon Jun 17 17:16:50 2019 kern.info kernel: [    5.162010] ip6_tables: (C) 2000-2006 Netfilter Core Team
[... 2084 lines cut ...]
upnpd.config.log_output='0'
upnpd.config.download='1024'
upnpd.config.upload='512'
upnpd.config.internal_iface='lan'
upnpd.config.port='5000'
upnpd.config.upnp_lease_file='/var/run/miniupnpd.leases'
upnpd.config.igdv1='0'
upnpd.@perm_rule[0]=perm_rule
upnpd.@perm_rule[0].action='allow'
upnpd.@perm_rule[0].ext_ports='1024-65535'
upnpd.@perm_rule[0].int_addr='0.0.0.0/0'
upnpd.@perm_rule[0].int_ports='1024-65535'
upnpd.@perm_rule[0].comment='Allow high ports'
upnpd.@perm_rule[1]=perm_rule
upnpd.@perm_rule[1].action='deny'
upnpd.@perm_rule[1].ext_ports='0-65535'
upnpd.@perm_rule[1].int_addr='0.0.0.0/0'
upnpd.@perm_rule[1].int_ports='0-65535'
upnpd.@perm_rule[1].comment='Default deny'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='eth0' 'eth1' 'eth2' 'eth3' 'eth4' 'eth5'`

  - OpenMPTCProuter version: 4.14.110
  - OpenMPTCProuter VPS version: 0.994
  - OpenMPTCProuter platform: x86_64

Ysurac, 2019-06-17:
What is OpenMPTCProuter version ? you give th kernel version but not OMR release version.

JorgeMoort89, 2019-06-17:
Sorry, 0.49.6

Ysurac, 2019-06-17:
So update both VPS and OpenMPTCProuter.

JorgeMoort89, 2019-06-17:
In this moment I can't update, the router is not near to me, I fear that I can made some mistake during the upgrade process remotely. I have that problem only when there is a lot of traffic on the router (15/20 people surfing the web/watching videos), can maybe be an hardware problem? The router have Xeon e3 1220 v2 with 4Gb of RAM

Ysurac, 2019-06-17:
Can also be a problem on the VPS side.

JorgeMoort89, 2019-06-17:
The max CPU usage on VPS is 30% of 1 core CPU, glorytun is using just 10% of it, ss is using 15%.
How can I find where I have the bottleneck? All the speedtests are fine, but with a lot of people using it I have that problem

Ysurac, 2019-06-17:
You can check the number of connections open, and on the VPS the logs.

JorgeMoort89, 2019-06-17:
Where I can find the VPS logs? the maximum connections that I ever seen was roughly 13k connections

Ysurac, 2019-06-17:
in /var/log

JorgeMoort89, 2019-06-17:
Have I to watch some specific log?

Ysurac, 2019-06-17:
begin with daemon.log if you are on debian, check also syslog

JorgeMoort89, 2019-06-17:
In daemon.log I see a lot of `ERROR: remote recv: Connection reset by peer`, also in syslog by ss-server

JorgeMoort89, 2019-06-18:
Now I have this message every 10 seconds `getpeername: Socket not connected`

Ysurac, 2019-06-18:
"getpeername: Socket not connected" means an IP cannot be reached by your server.

JorgeMoort89, 2019-06-18:
Can maybe be the ICMP blocked on the modem? What can be `ERROR: remote recv: Connection reset by peer`? I'm thinking about firewall leaved working on the modems

Ysurac, 2019-06-18:
It's some network errors.
If firewall let connections open from LAN side, no problems with it.

JorgeMoort89, 2019-06-18:
How I can find the IP not reachable by the router? I'll see today with a lot of people if I have the same problem, can I maybe set a static master interface to avoid the problems presented on the issue?

Ysurac, 2019-06-18:
You can enable verbose mode in shadowsocks on the VPS in ```/etc/shadowsocks-libev/config.json```
Master interface is not set as dynamic in your settings, so this can only change after config changes.

JorgeMoort89, 2019-06-18:
What verbose mode do in ss? Can maybe resolve my problem?

Ysurac, 2019-06-18:
it's more verbose, so you will know what IPs are not reachable.

JorgeMoort89, 2019-06-18:
Same issues:
`ue Jun 18 09:27:55 2019 daemon.err /usr/bin/ss-local[7943]: getpeername: Socket not connected
Tue Jun 18 09:27:58 2019 user.notice post-tracking-post-tracking: omrvpn down. Replace default route by xxx.xxx.xxx.xxx dev eth1
Tue Jun 18 09:27:58 2019 user.notice post-tracking-post-tracking: omrvpn down because gateway down
Tue Jun 18 09:28:00 2019 user.notice post-tracking-post-tracking: Master up : Replace default route by xxx.xxx.xxx.xxx dev eth3
Tue Jun 18 09:28:05 2019 daemon.err /usr/bin/ss-local[7943]: getpeername: Socket not connected
Tue Jun 18 09:28:14 2019 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0
Tue Jun 18 09:28:16 2019 daemon.err /usr/bin/ss-local[7943]: getpeername: Socket not connected
Tue Jun 18 09:28:26 2019 daemon.err /usr/bin/ss-local[7943]: getpeername: Socket not connected
Tue Jun 18 09:28:36 2019 daemon.err /usr/bin/ss-local[7943]: getpeername: Socket not connected
Tue Jun 18 09:28:46 2019 daemon.err /usr/bin/ss-local[7943]: getpeername: Socket not connected
Tue Jun 18 09:28:56 2019 daemon.err /usr/bin/ss-local[7943]: getpeername: Socket not connected
Tue Jun 18 09:29:06 2019 daemon.err /usr/bin/ss-local[7943]: getpeername: Socket not connected
Tue Jun 18 09:29:13 2019 daemon.err glorytun[5964]: write: Operation timed out
Tue Jun 18 09:29:13 2019 daemon.info glorytun[5964]: STOPPED tun0
Tue Jun 18 09:29:14 2019 daemon.err glorytun[5964]: xxx.xxx.xxx.xxx.65001: connected
Tue Jun 18 09:29:14 2019 daemon.info glorytun[5964]: STARTED tun0`

Ysurac, 2019-06-19:
There is problem on the connection between OpenMPTCProuter and the VPS, as you can see there is a "socket not connected" error from ss-local, so shadowsocks, and after a few second the VPS is detected as down.
You can increase values in omr-tracker for omrvpn, you will lost some packets but it will not be detected as down.

JorgeMoort89, 2019-06-20:
Ok I'll try it, can I update the router remotely? I've seen that warning: `Upgrade on x86_64 from version between 0.49 and 0.50.2 using luci interface doesn't work. Save config and import it on a new install.`

Ysurac, 2019-06-20:
It's often not a good idea to update remotely. At least for now.

JorgeMoort89, 2019-06-24:
How can I enable verbose mode on `/etc/shadowsocks-libev/config.json` ?

JorgeMoort89, 2019-06-24:
I've noticed that while i'm downloading at full speed the latency to the OMR-TRACKER IPs (8.8.8.8, 8.8.4.4 ecc.) increases a lot, sometimes it goes on timeout error, can that be the problem?

Ysurac, 2019-06-24:
Yes it's a problem. You can enable SQM.

JorgeMoort89, 2019-06-25:
I've enabled it, but when I save the settings the checkbox return unchecked.
[image]

Ysurac, 2019-07-04:
This will be fixed in next release.

closed 2019-07-04 by Ysurac