Ysurac/openmptcprouter #517: IPv6 routing/ OPKG error

jwthecolorist, 2019-06-17:
## Expected Behavior

I've noticed when I run opkg update, it fails because it cannot route ipv6 (the default address that returns is ipv6 instead of ipv4). My ISP's do not serve ipv6, only ipv4. However, my VPS has both. Is it possible to route ipv6 by routing through the VPS?

It looks like that is supposed to happen with omr6in4 but it didn't seem to be working so I disabled all "Use builtin IPv6-management" on the lan/wan interfaces and that didn't fix it either. I also disabled ipv6 in shadowsocks and that didn't help opkg either.

My preference would be to set up my network so I can have IPV6 on all client routes out the ipv6 via the ipv6 on the vps but I disabled it on the lan DHCP and the interface settings because it didn't seem to be working. Plus I see no traffic on omr6in4

Is that currently available on the release 0.50.2? Do I need to configure something specific to make that happen or is it enabled by default?

I'm very unfamiliar with ipv6 so I may just have a configuration error. Just let me know what you need from me. As far as the issue with opkg, I'm wondering if this is an issue with DNS as it's returning ipv6 instead of ipv4.

Thanks!!

[image]

[image]

[image]

[image]

[image]

## Specifications

  - OpenMPTCProuter version: 0.50.2
  - OpenMPTCProuter VPS version: 0.995
  - OpenMPTCProuter platform: x86_64

Ysurac, 2019-06-17:
Yes it's available. You need to check ipv6 in the wizard, and if it's configured on the server you should see the IPv6 in OpenMPTCProuter status page.

jayanta525, 2019-06-18:
Replace `http://downloads.openwrt.org/` with mirrors from `https://openwrt.org/downloads#mirrors` to workaround the IPv6 issue.

Ysurac, 2019-06-18:
I next release I will force IPv4 for opkg.

jwthecolorist, 2019-06-18:
Thanks for the feedback! 

I enabled ipv6 (after a fresh install, had disabled ipv6 everywhere and would take too long to re-enable everything) and now it appears to be working fine. Even opkg is updating correctly without changing the repos.

@jayanta525 Thanks for the suggestion! I hadn't seen that option but I had tried changing to a mirror that only resolved to ipv4. That worked but didn't resolve the greater issue. Enabling ipv6 in the wizard did though.

As a side note: I'm currently updating my VPS to 0.996. Looking forward to seeing how it performs.

jwthecolorist, 2019-06-18:
Incase this is related, I now have lots of these in my log:

[image]

Any thoughts?

Ysurac, 2019-06-18:
It's a test in status page that fail. You enabled IPv6 in the wizard ?

jwthecolorist, 2019-06-22:
Sorry for the late reply! Yes, I had it enabled in the wizard. Just turned it off though because it was causing a strange problem where all the connections would complete drop periodically. Not sure why yet, I don't think it was your software but something with my network. 

That said (this is not a complaint, just an observation), version 0.50.2 does appear to be far more unstable than the previous versions. It appears to drop connections faster, reboots dnsmasq more etc. It could just be something with my system though.

closed 2019-07-04 by Ysurac