Ysurac/openmptcprouter #519: Restauration impossible et perte du fichier du fichier de configuration

Randdrick, 2019-06-21:
## Expected Behavior

  Dans la version 0.49.5 RPI 3+ le fichier de configuration se sauvegarde et se restaure correctement. La configuration du routeur n'est pas perdue après un redémarrage.

## Actual Behavior
Depuis la version 0.50.1 et 2, il est possible de faire un backup de la configuration. Par contre, dès que le routeur redémarre sa configuration est perdue. Il en va de même si on recharge son fichier de sauvegarde.

## Steps to Reproduce the Problem

  1. Installer la version de  openmptcprouter-v0.50.2-r010127-3209f5ae3d-brcm2708-bcm2710-rpi-3-squashfs-factory.img.gz ou openmptcprouter-v0.50.2-r010127-3209f5ae3d-brcm2708-bcm2710-rpi-3-squashfs-sysupgrade.img.gz
  2. Configurer le routeur et sauvegarder les changements
  3. Redémarrer le routeur

## Specifications

  - OpenMPTCProuter version: (0.50.2)
  - OpenMPTCProuter VPS version: (0.995)
  - OpenMPTCProuter platform: (RPI3)

## Notes complémentaires.
Le problème se produit même si je flash de nouveau mon firmware.
Il n'y a pas de problème avec la version 0.49.5. La restauration de son fichier de configuration et le redémarrage du routeur ne pose pas de problème.
L'upgrade entre la version 0.49.5 vers la version v.50.2 avec conservation des paramètres de configuration se passe sans problème. Par contre, il ne faut pas redémarrer le routeur sous peine d'une perte total de la configuration.

Ysurac, 2019-06-22:
Comment le routeur est redémarré ? en débranchant et rebranchant ou via le menu Système->Redémarrage ?

Ext4 fonctionne correctement, mais effectivement il y a toute une série de soucis avec F2FS...

Randdrick, 2019-06-22:
En appuyant sur le bouton Système->Redémarrage. Mais cela se produit aussi si je débranche et rebranche le routeur.
Pour le moment j'ai basculé de nouveau en 0.49.5 pour éviter de perdre ma configuration en cas de problème de coupure électrique (par exemple)

Ysurac, 2019-06-22:
Tu peux aussi essayer la versions en testing : https://www.openmptcprouter.com/files/archive/
Il faut que je fasse quelques tests et ça devrait être la 0.51

Randdrick, 2019-06-22:
Très bonne nouvelle : Le problème semble être résolu avec la version openmptcprouter-v0.50.2-27-g4403030-r0+10298-eaad2211db-brcm2708-bcm2710-rpi-3-squashfs-sysupgrade.img.gz
L'upgrade à partir de la version 0.49.5 se fait sans problème. Tous les paramètres sont bien conservés. La sauvegarde, la restauration et le redémarrage par le bouton Système->Redémarrage se fait sans problème.
Je n'ai pas fait le test en débranchant et rebranchant le routeur.

Ysurac, 2019-06-22:
Cool. Je vais tester un peu et ça pourra être la prochaine version.

Randdrick, 2019-06-22:
Je garde cette version et je te tiens au courant si je rencontre d'autres problèmes.

Randdrick, 2019-06-22:
Quelques erreurs (dans cette version de test) que je n'avais pas encore eu jusqu'à présent :

Sat Jun 22 18:32:00 2019 kern.err kernel: [ 8353.758904] mptcp_detect_mapping Mappings do not match!
Sat Jun 22 18:32:00 2019 kern.err kernel: [ 8353.765317] mptcp_detect_mapping dseq 3472393377 mdseq 3472393377, sseq 1124014982 msseq 1124012302 dlen 14088 mdlen 14087 dfin 1 mdfin 1
Sat Jun 22 18:32:01 2019 kern.err kernel: [ 8354.144491] mptcp_detect_mapping Mappings do not match!
Sat Jun 22 18:32:01 2019 kern.err kernel: [ 8354.150875] mptcp_detect_mapping dseq 859991362 mdseq 859991362, sseq 3771113695 msseq 3771111015 dlen 3594 mdlen 3593 dfin 1 mdfin 1
Sat Jun 22 18:32:03 2019 kern.err kernel: [ 8356.191255] mptcp_detect_mapping Mappings do not match!
Sat Jun 22 18:32:03 2019 kern.err kernel: [ 8356.197631] mptcp_detect_mapping dseq 3070053188 mdseq 3070053188, sseq 2399252241 msseq 2399249561 dlen 8160 mdlen 8159 dfin 1 mdfin 1
Sat Jun 22 18:32:04 2019 kern.err kernel: [ 8356.941716] mptcp_detect_mapping Mappings do not match!
Sat Jun 22 18:32:04 2019 kern.err kernel: [ 8356.948086] mptcp_detect_mapping dseq 4065126454 mdseq 4065126454, sseq 79934448 msseq 79931768 dlen 7334 mdlen 7333 dfin 1 mdfin 1
Sat Jun 22 18:32:04 2019 kern.err kernel: [ 8357.527531] mptcp_detect_mapping Mappings do not match!
Sat Jun 22 18:32:04 2019 kern.err kernel: [ 8357.533875] mptcp_detect_mapping dseq 3812032856 mdseq 3812032856, sseq 68983769 msseq 68981089 dlen 14528 mdlen 14527 dfin 1 mdfin 1
Sat Jun 22 18:32:04 2019 kern.err kernel: [ 8357.702210] mptcp_detect_mapping Mappings do not match!
Sat Jun 22 18:32:04 2019 kern.err kernel: [ 8357.708576] mptcp_detect_mapping dseq 3812044216 mdseq 3812044216, sseq 2679399770 msseq 2679396930 dlen 3168 mdlen 3167 dfin 1 mdfin 1
Sat Jun 22 18:32:05 2019 kern.err kernel: [ 8358.186859] mptcp_detect_mapping Mappings do not match!
Sat Jun 22 18:32:05 2019 kern.err kernel: [ 8358.193255] mptcp_detect_mapping dseq 1455965386 mdseq 1455965386, sseq 278975959 msseq 278973279 dlen 6998 mdlen 6997 dfin 1 mdfin 1
Sat Jun 22 18:32:05 2019 kern.err kernel: [ 8358.777684] mptcp_detect_mapping Mappings do not match!
Sat Jun 22 18:32:05 2019 kern.err kernel: [ 8358.784150] mptcp_detect_mapping dseq 1219892190 mdseq 1219892190, sseq 1157577143 msseq 1157574463 dlen 4882 mdlen 4881 dfin 1 mdfin 1
Sat Jun 22 18:32:05 2019 kern.err kernel: [ 8358.843555] mptcp_detect_mapping Mappings do not match!
Sat Jun 22 18:32:05 2019 kern.err kernel: [ 8358.849981] mptcp_detect_mapping dseq 2728844621 mdseq 2728844621, sseq 2863891282 msseq 2863888442 dlen 14146 mdlen 14145 dfin 1 mdfin 1
Sat Jun 22 18:32:06 2019 kern.err kernel: [ 8359.305992] mptcp_detect_mapping Mappings do not match!
Sat Jun 22 18:32:06 2019 kern.err kernel: [ 8359.312380] mptcp_detect_mapping dseq 3277105851 mdseq 3277105851, sseq 840637807 msseq 840634967 dlen 4020 mdlen 4019 dfin 1 mdfin 1
Sat Jun 22 18:32:06 2019 kern.err kernel: [ 8359.368645] mptcp_detect_mapping Mappings do not match!
Sat Jun 22 18:32:06 2019 kern.err kernel: [ 8359.375038] mptcp_detect_mapping dseq 3277105851 mdseq 3277105851, sseq 2632592924 msseq 2632590244 dlen 4020 mdlen 4019 dfin 1 mdfin 1
Sat Jun 22 18:32:07 2019 kern.err kernel: [ 8360.016898] mptcp_detect_mapping Mappings do not match!
Sat Jun 22 18:32:07 2019 kern.err kernel: [ 8360.023269] mptcp_detect_mapping dseq 963856900 mdseq 963856900, sseq 2600946322 msseq 2600943642 dlen 34756 mdlen 34755 dfin 1 mdfin 1
Sat Jun 22 18:32:08 2019 kern.err kernel: [ 8361.152617] mptcp_detect_mapping Mappings do not match!
Sat Jun 22 18:32:08 2019 kern.err kernel: [ 8361.159023] mptcp_detect_mapping dseq 3343692808 mdseq 3343692808, sseq 1850315072 msseq 1850312392 dlen 12310 mdlen 12309 dfin 1 mdfin 1
Sat Jun 22 18:32:08 2019 kern.err kernel: [ 8361.227000] mptcp_detect_mapping Mappings do not match!
Sat Jun 22 18:32:08 2019 kern.err kernel: [ 8361.233385] mptcp_detect_mapping dseq 3343692808 mdseq 3343692808, sseq 538113572 msseq 538110732 dlen 12310 mdlen 12309 dfin 1 mdfin 1
Sat Jun 22 18:32:49 2019 daemon.err uhttpd[2330]: luci: accepted login on / for root from 192.168.100.230
Sat Jun 22 18:34:37 2019 kern.err kernel: [ 8509.983737] mptcp_detect_mapping Mappings do not match!
Sat Jun 22 18:34:37 2019 kern.err kernel: [ 8509.990141] mptcp_detect_mapping dseq 2268828693 mdseq 2268828693, sseq 376767515 msseq 376764675 dlen 6578 mdlen 6577 dfin 1 mdfin 1
Sat Jun 22 18:34:37 2019 kern.err kernel: [ 8510.686277] mptcp_detect_mapping Mappings do not match!
Sat Jun 22 18:34:37 2019 kern.err kernel: [ 8510.692687] mptcp_detect_mapping dseq 655306649 mdseq 655306649, sseq 2521928917 msseq 2521926077 dlen 10376 mdlen 10375 dfin 1 mdfin 1
Sat Jun 22 18:34:48 2019 kern.err kernel: [ 8521.061569] mptcp_detect_mapping Mappings do not match!
Sat Jun 22 18:34:48 2019 kern.err kernel: [ 8521.067964] mptcp_detect_mapping dseq 2086238727 mdseq 2086238727, sseq 3644774446 msseq 3644771606 dlen 6350 mdlen 6349 dfin 1 mdfin 1
Sat Jun 22 18:34:54 2019 kern.err kernel: [ 8527.552916] mptcp_detect_mapping Mappings do not match!
Sat Jun 22 18:34:54 2019 kern.err kernel: [ 8527.559354] mptcp_detect_mapping dseq 1698474296 mdseq 1698474296, sseq 1618551361 msseq 1618548521 dlen 2968 mdlen 2967 dfin 1 mdfin 1
Sat Jun 22 18:34:55 2019 kern.err kernel: [ 8528.505248] mptcp_detect_mapping Mappings do not match!
[... cut ...]
Sat Jun 22 18:34:59 2019 kern.err kernel: [ 8532.286757] mptcp_detect_mapping Mappings do not match!
Sat Jun 22 18:34:59 2019 kern.err kernel: [ 8532.293143] mptcp_detect_mapping dseq 1388001004 mdseq 1388001004, sseq 521278948 msseq 521276108 dlen 3245 mdlen 3244 dfin 1 mdfin 1
Sat Jun 22 18:34:59 2019 kern.err kernel: [ 8532.752188] mptcp_detect_mapping Mappings do not match!
Sat Jun 22 18:34:59 2019 kern.err kernel: [ 8532.758624] mptcp_detect_mapping dseq 249960397 mdseq 249960397, sseq 877783261 msseq 877780421 dlen 23324 mdlen 23323 dfin 1 mdfin 1

closed 2019-12-20 by Ysurac