Ysurac/openmptcprouter #522: 5Ghz WiFi [WRT3200ACM]

rubaz100, 2019-06-23:
## Expected Behavior

WiFi 5Ghz  AC fullspeed connection

## Actual Behavior

Not working 

[image]

## Steps to Reproduce the Problem

Every WRT3200ACM

I found this: https://dc502wrt.org/

https://forum.openwrt.org/t/davidc502-wrt1200ac-wrt1900acx-wrt3200acm-wrt32x-builds/15839

Could you try it with that openwrt build?

## Specifications

  - OpenMPTCProuter version: 0.50.2
  - OpenMPTCProuter VPS version: 0.996.4.14.122-mptcp
  - OpenMPTCProuter platform: WRT3200ACM

Ysurac, 2019-06-23:
I don't know what is your issue, this is working for me. Try another channel.
dc502wrt seems to be only a compiled image of OpenWrt.

rubaz100, 2019-06-23:
Its not working. 

`Sun Jun 23 21:09:44 2019 kern.info kernel: [106495.660179] br-lan: port 3(wlan0) entered disabled state
Sun Jun 23 21:09:44 2019 kern.info kernel: [106495.703444] br-lan: port 3(wlan0) entered disabled state
Sun Jun 23 21:09:45 2019 user.notice mac80211: Failed command: iw phy phy0 set antenna 0xffffffff 0xffffffff
Sun Jun 23 21:09:45 2019 daemon.notice netifd: radio0 (11941): command failed: Not supported (-95)
Sun Jun 23 21:09:45 2019 user.notice mac80211: Failed command: iw phy phy0 set distance 0
Sun Jun 23 21:09:45 2019 daemon.notice netifd: radio0 (11941): sh: out of range
Sun Jun 23 21:09:45 2019 daemon.err hostapd: Configuration file: /var/run/hostapd-phy0.conf
Sun Jun 23 21:09:46 2019 kern.debug kernel: [106497.093599] ieee80211 phy0: change: 0xffffffff
Sun Jun 23 21:09:46 2019 kern.info kernel: [106497.197325] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Sun Jun 23 21:09:46 2019 kern.info kernel: [106497.204432] br-lan: port 3(wlan0) entered blocking state
Sun Jun 23 21:09:46 2019 kern.info kernel: [106497.209858] br-lan: port 3(wlan0) entered disabled state
Sun Jun 23 21:09:46 2019 daemon.notice hostapd: wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Sun Jun 23 21:09:51 2019 daemon.notice hostapd: wlan0: interface state COUNTRY_UPDATE->HT_SCAN
Sun Jun 23 21:09:51 2019 kern.debug kernel: [106502.217719] ieee80211 phy0: change: 0x100
Sun Jun 23 21:09:51 2019 kern.debug kernel: [106502.225327] ieee80211 phy0: change: 0x40
Sun Jun 23 21:09:51 2019 kern.debug kernel: [106502.435338] ieee80211 phy0: change: 0x40
Sun Jun 23 21:09:51 2019 daemon.notice hostapd: wlan0: interface state HT_SCAN->DFS
Sun Jun 23 21:09:51 2019 daemon.notice hostapd: wlan0: DFS-CAC-START freq=5300 chan=60 sec_chan=1, width=2, seg0=50, seg1=0, cac_time=60s
Sun Jun 23 21:09:51 2019 daemon.err hostapd: DFS start_dfs_cac() failed, -1
Sun Jun 23 21:09:51 2019 daemon.err hostapd: Interface initialization failed
Sun Jun 23 21:09:51 2019 daemon.notice hostapd: wlan0: interface state DFS->DISABLED
Sun Jun 23 21:09:51 2019 daemon.notice hostapd: wlan0: AP-DISABLED
Sun Jun 23 21:09:51 2019 daemon.notice hostapd: wlan0: interface state DISABLED->DISABLED
Sun Jun 23 21:09:51 2019 daemon.notice hostapd: wlan0: AP-DISABLED
Sun Jun 23 21:09:51 2019 daemon.notice hostapd: wlan0: CTRL-EVENT-TERMINATING
Sun Jun 23 21:09:51 2019 daemon.err hostapd: hostapd_free_hapd_data: Interface wlan0 wasn't started
Sun Jun 23 21:09:51 2019 daemon.notice hostapd: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Sun Jun 23 21:09:51 2019 kern.debug kernel: [106502.645414] ieee80211 phy0: change: 0x40
Sun Jun 23 21:09:51 2019 kern.info kernel: [106502.685536] br-lan: port 3(wlan0) entered disabled state
Sun Jun 23 21:09:51 2019 kern.debug kernel: [106502.744517] ieee80211 phy0: change: 0x100
Sun Jun 23 21:09:51 2019 daemon.notice hostapd: ELOOP: remaining socket: sock=22 eloop_data=0xb6ff4560 user_data=0 handler=0x2b220
Sun Jun 23 21:09:51 2019 daemon.notice netifd: radio0 (11941): cat: can't open '/var/run/wifi-phy0.pid': No such file or directory
Sun Jun 23 21:09:51 2019 daemon.notice netifd: radio0 (11941): WARNING (wireless_add_process): executable path /usr/sbin/wpad does not match process  path (/proc/exe)
Sun Jun 23 21:09:51 2019 daemon.notice netifd: radio0 (11941): Command failed: Invalid argument
Sun Jun 23 21:09:51 2019 kern.debug kernel: [106502.921517] ieee80211 phy0: change: 0xffffffff
Sun Jun 23 21:09:51 2019 kern.info kernel: [106503.024582] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Sun Jun 23 21:09:51 2019 kern.info kernel: [106503.041570] br-lan: port 3(wlan0) entered blocking state
Sun Jun 23 21:09:51 2019 kern.info kernel: [106503.047095] br-lan: port 3(wlan0) entered disabled state
`

closed 2019-07-21 by Ysurac