Ysurac/openmptcprouter #524: Pi 4 image

zicochaos, 2019-06-24:
Any chance for Pi 4 images or Pi 3 are fine?

Ysurac, 2019-06-24:
I'm adding support to the tree. But can't be out before I receive my Raspberry PI 4 this week.

audren49, 2019-06-28:
What will be pi 4 top speed by the way?

xinnanlyu, 2019-07-01:
tested with  **OMR_TARGET="rpi4" ./build.sh**   , not booting up from squashfs image.

Ysurac, 2019-07-01:
RPI4 target is for the 64bits version, for now this doesn't work at all.
32bits version use the RPI2 target, but this fail after kernel load for now.

I crashed my PC motherboard during tests, so dev/tests have to wait I receive the new one...

xinnanlyu, 2019-07-01:
Looking forward to it. Can't wait to see how much AES performance boost by RPI4 BCM2711.

Ysurac, 2019-07-01:
But, sadly, AES is not enabled on RPI4: https://www.raspberrypi.org/forums/viewtopic.php?t=243410
You can check using raspbian image: no AES flag on /proc/cpuinfo...

adrianmace, 2019-07-02:
I have just found this project and think that it is absolutely fantastic! Amazing work! I have previously built my own implementation using a pair of Raspberry Pi 3B+ running ZeroShell but I think this will be much better and eager to switch over to it.

Will likely buy a Pi 4 for this project, so adding this comment in support of a Pi 4 image.

Ysurac, 2019-07-02:
For now same problem as https://github.com/raspberrypi/linux/issues/3032
```
mmc1: new high speed SDIO card at address 0001
```
This should be mmc0 and this should not hang.

Ysurac, 2019-07-11:
In master branch, RPI4 booting in 32 and 64bits. 
Some problems:
* Limited to 1Go of RAM for 64bits version (need that upstream fix that).
* Wifi not yet working

Ysurac, 2019-07-11:
Wifi fixed

xinnanlyu, 2019-07-11:
nice one!

Ysurac, 2019-07-12:
By using ```coherent_pool=128M cma=256M@128M swiotlb=force``` to RPI4 64bits command line, this seems to boot without errors and with full memory. I think this can be tuned...

Ysurac, 2019-07-13:
Current images at https://www.openmptcprouter.com/release/4.19/ are ready for tests.
VPS script at http://www.openmptcprouter.com/server/debian9-x86_64-test.sh

Ysurac, 2019-07-21:
RPI4 image released, with a dirty hack for 64bits while there is no fix from official RPI kernel.

closed 2019-07-21 by Ysurac