Ysurac/openmptcprouter #525: vpn disconnect on saturate

Florent4223, 2019-06-24:
## Expected Behavior

  stable vpn

## Actual Behavior
On transfert ping increase ...disconnect. ...reconnect. ...etc
The transfert demiss

## Steps to Reproduce the Problem

  1. Glorytun tcp ok
  2. Transfert file
  3.

## Specifications

  - OpenMPTCProuter version: (Last is not a version) 0.50.2
  - OpenMPTCProuter VPS version: (Last is not a version) 0.998
  - OpenMPTCProuter platform: (RPI2/RPI3/x86/x86_64) rpi3

Ysurac, 2019-06-24:
You can try to enable SQM in wizard.
And check system log.

Florent4223, 2019-06-24:
SQM but... i have 1 key 4G

Ysurac, 2019-06-24:
You can try but when there is only 4G connections it's difficult to know the bandwidth that is always variable.

Florent4223, 2019-06-24:
ok thanks i will try
You know if there are changes because in 0.49.7 it worked well

In case I can get the vps in 0.993?

Ysurac, 2019-06-24:
As I already said, check system log.

Florent4223, 2019-06-25:
try avec 0.50.2 et vps 0.995

[image]

journal system:

Tue Jun 25 10:18:51 2019 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Tue Jun 25 10:18:51 2019 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Tue Jun 25 10:18:51 2019 daemon.err omr-tracker[21016]: udhcpc: started, v1.30.1
Tue Jun 25 10:18:51 2019 daemon.err omr-tracker[21016]: udhcpc: sending discover
Tue Jun 25 10:18:54 2019 daemon.err omr-tracker[21016]: udhcpc: no lease, failing
Tue Jun 25 10:18:54 2019 daemon.info dnsmasq[32726]: started, version 2.80 cachesize 150
Tue Jun 25 10:18:54 2019 daemon.info dnsmasq[32726]: DNS service limited to local subnets
Tue Jun 25 10:18:54 2019 daemon.info dnsmasq[32726]: compile time options: IPv6 GNU-getopt no-DBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC no-ID loop-detect inotify dumpfile
Tue Jun 25 10:18:54 2019 daemon.info dnsmasq-dhcp[32726]: DHCP, IP range 192.168.100.100 -- 192.168.100.249, lease time 12h
Tue Jun 25 10:18:54 2019 daemon.info dnsmasq[32726]: using local addresses only for domain test
Tue Jun 25 10:18:54 2019 daemon.info dnsmasq[32726]: using local addresses only for domain onion
Tue Jun 25 10:18:54 2019 daemon.info dnsmasq[32726]: using local addresses only for domain localhost
Tue Jun 25 10:18:54 2019 daemon.info dnsmasq[32726]: using local addresses only for domain local
Tue Jun 25 10:18:54 2019 daemon.info dnsmasq[32726]: using local addresses only for domain invalid
Tue Jun 25 10:18:54 2019 daemon.info dnsmasq[32726]: using local addresses only for domain bind
Tue Jun 25 10:18:54 2019 daemon.info dnsmasq[32726]: using nameserver 127.0.0.1#5353
Tue Jun 25 10:18:54 2019 daemon.info dnsmasq[32726]: using local addresses only for domain lan
Tue Jun 25 10:18:54 2019 daemon.info dnsmasq[32726]: read /etc/hosts - 4 addresses
Tue Jun 25 10:18:54 2019 daemon.info dnsmasq[32726]: read /tmp/hosts/odhcpd - 0 addresses
Tue Jun 25 10:18:54 2019 daemon.info dnsmasq[32726]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Tue Jun 25 10:18:54 2019 daemon.info dnsmasq-dhcp[32726]: read /etc/ethers - 0 addresses
Tue Jun 25 10:18:54 2019 user.notice omr-bypass: OMR-ByPass is running
Tue Jun 25 10:19:04 2019 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0
Tue Jun 25 10:19:04 2019 user.notice post-tracking-post-tracking: Tunnel up disable use of ShadowSocks for UDP
Tue Jun 25 10:19:04 2019 user.notice Shadowsocks: Rules DOWN
Tue Jun 25 10:19:04 2019 user.notice Shadowsocks: Rules UP
Tue Jun 25 10:19:05 2019 user.notice Shadowsocks: Reload omr-bypass rules
Tue Jun 25 10:19:05 2019 user.notice omr-bypass: Starting OMR-ByPass...
Tue Jun 25 10:19:06 2019 daemon.info dnsmasq[32726]: exiting on receipt of SIGTERM
Tue Jun 25 10:19:06 2019 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Tue Jun 25 10:19:06 2019 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Tue Jun 25 10:19:09 2019 daemon.info dnsmasq[1914]: started, version 2.80 cachesize 150
Tue Jun 25 10:19:09 2019 daemon.info dnsmasq[1914]: DNS service limited to local subnets
Tue Jun 25 10:19:09 2019 daemon.info dnsmasq[1914]: compile time options: IPv6 GNU-getopt no-DBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC no-ID loop-detect inotify dumpfile
Tue Jun 25 10:19:09 2019 daemon.info dnsmasq-dhcp[1914]: DHCP, IP range 192.168.100.100 -- 192.168.100.249, lease time 12h
Tue Jun 25 10:19:09 2019 daemon.info dnsmasq[1914]: using local addresses only for domain test
Tue Jun 25 10:19:09 2019 daemon.info dnsmasq[1914]: using local addresses only for domain onion
Tue Jun 25 10:19:09 2019 daemon.info dnsmasq[1914]: using local addresses only for domain localhost
Tue Jun 25 10:19:09 2019 daemon.info dnsmasq[1914]: using local addresses only for domain local
Tue Jun 25 10:19:09 2019 daemon.info dnsmasq[1914]: using local addresses only for domain invalid
Tue Jun 25 10:19:09 2019 daemon.info dnsmasq[1914]: using local addresses only for domain bind
Tue Jun 25 10:19:09 2019 daemon.info dnsmasq[1914]: using nameserver 127.0.0.1#5353
Tue Jun 25 10:19:09 2019 daemon.info dnsmasq[1914]: using local addresses only for domain lan
Tue Jun 25 10:19:09 2019 daemon.info dnsmasq[1914]: read /etc/hosts - 4 addresses
Tue Jun 25 10:19:09 2019 daemon.info dnsmasq[1914]: read /tmp/hosts/odhcpd - 0 addresses
Tue Jun 25 10:19:09 2019 daemon.info dnsmasq[1914]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Tue Jun 25 10:19:09 2019 daemon.info dnsmasq-dhcp[1914]: read /etc/ethers - 0 addresses
Tue Jun 25 10:19:09 2019 user.notice omr-bypass: OMR-ByPass is running
Tue Jun 25 10:19:51 2019 user.notice post-tracking-post-tracking: Get status and settings for LTE0002... Done
Tue Jun 25 10:19:55 2019 user.notice post-tracking-post-tracking: Get status and settings for LTE0001... Done
Tue Jun 25 10:20:00 2019 user.notice post-tracking-post-tracking: Get status and settings for LTE0001... Done
Tue Jun 25 10:20:03 2019 user.notice post-tracking-post-tracking: omrvpn down. Replace default route by 192.168.8.1 dev eth1
Tue Jun 25 10:20:03 2019 user.notice post-tracking-post-tracking: omrvpn down because gateway down
Tue Jun 25 10:20:03 2019 user.notice post-tracking-post-tracking: Tunnel down use ShadowSocks for UDP
Tue Jun 25 10:20:03 2019 user.notice Shadowsocks: Rules DOWN
Tue Jun 25 10:20:04 2019 user.notice Shadowsocks: Rules UP
[... cut ...]
Tue Jun 25 10:24:37 2019 daemon.info dnsmasq[20170]: using local addresses only for domain lan
Tue Jun 25 10:24:37 2019 daemon.info dnsmasq[20170]: read /etc/hosts - 4 addresses
Tue Jun 25 10:24:37 2019 daemon.info dnsmasq[20170]: read /tmp/hosts/odhcpd - 0 addresses
Tue Jun 25 10:24:37 2019 daemon.info dnsmasq[20170]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Tue Jun 25 10:24:37 2019 daemon.info dnsmasq-dhcp[20170]: read /etc/ethers - 0 addresses
Tue Jun 25 10:24:37 2019 user.notice omr-bypass: OMR-ByPass is running

Florent4223, 2019-06-25:
and SQM nothing different

Florent4223, 2019-06-25:
I do not know if it's related but I have these anomalies too in log core:

[ 4249.537735] ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[ 4249.548148] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-110
[ 4255.615745] ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[ 4255.626155] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-110
[ 4261.691114] ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[ 4261.702214] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-110
[ 4267.755898] ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[ 4267.767056] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-110
[ 4270.810551] ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[ 4270.821983] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-110
[ 4273.839798] ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[ 4273.851756] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-110
[ 4276.882272] ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[ 4276.893717] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-110
[ 4282.930843] ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[ 4282.942423] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-110
[ 4289.022475] ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[ 4289.033967] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-110
[ 4292.066994] ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[ 4292.078410] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-110
[ 4298.142570] ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[ 4298.153951] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-110
[ 4304.211885] ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[ 4304.223393] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-110
[ 4310.270197] ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[ 4310.281607] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-110
[ 4316.313835] ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[ 4316.325237] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-110

Florent4223, 2019-06-25:
I detail a little more works since the implementation of version 0.50.2.

The transfer of a file via glorytun tcp goes well, with the maximum possible bitrate. But with saturation, the ping increases until the VPN disconnection.

Nevertheless in congestion bbr (and it is the only one) the transfer continues a few tens of seconds later after reconnection of the vpn ... to pause at the next disconnection of the vpn etc etc

ArChlOgY, 2019-06-25:
I have the same issue with 2 VDSL link, I set-up OpenMPTCProuter last week and use version 0.50.2 - never use previous version so I cannot confirm if it was more stable but my connection fail every 1 min.

Here is my log :

> Tue Jun 25 17:17:31 2019 user.notice omr-bypass: OMR-ByPass is running
Tue Jun 25 17:17:31 2019 daemon.info dnsmasq[20252]: read /tmp/hosts/dhcp.cfg01411c - 5 addresses
Tue Jun 25 17:17:31 2019 daemon.info dnsmasq-dhcp[20252]: read /etc/ethers - 0 addresses
Tue Jun 25 17:17:47 2019 user.notice post-tracking-post-tracking: omrvpn down. Replace default route by 192.168.0.1 dev wan1
Tue Jun 25 17:17:47 2019 user.notice post-tracking-post-tracking: omrvpn down because gateway down
Tue Jun 25 17:17:48 2019 user.notice post-tracking-post-tracking: Tunnel down use ShadowSocks for UDP
Tue Jun 25 17:17:48 2019 user.notice Shadowsocks: Rules DOWN
Tue Jun 25 17:17:48 2019 user.notice Shadowsocks: Rules UP
Tue Jun 25 17:17:49 2019 daemon.info omr-tracker-ss: Reload Shadowsocks rules
Tue Jun 25 17:17:49 2019 user.notice Shadowsocks: Reload omr-bypass rules
Tue Jun 25 17:17:49 2019 user.notice omr-bypass: Starting OMR-ByPass...
Tue Jun 25 17:17:50 2019 user.notice post-tracking-post-tracking: Master up : Replace default route by 192.168.1.1 dev wan2
Tue Jun 25 17:17:52 2019 daemon.info dnsmasq[20252]: exiting on receipt of SIGTERM
Tue Jun 25 17:17:52 2019 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Tue Jun 25 17:17:52 2019 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Tue Jun 25 17:17:52 2019 daemon.err omr-tracker[4887]: udhcpc: started, v1.30.1
Tue Jun 25 17:17:52 2019 daemon.err omr-tracker[4887]: udhcpc: sending discover
Tue Jun 25 17:17:55 2019 daemon.err omr-tracker[4887]: udhcpc: no lease, failing
Tue Jun 25 17:17:55 2019 daemon.info dnsmasq[22130]: started, version 2.80 cachesize 150
Tue Jun 25 17:17:55 2019 daemon.info dnsmasq[22130]: DNS service limited to local subnets
Tue Jun 25 17:17:55 2019 daemon.info dnsmasq[22130]: compile time options: IPv6 GNU-getopt no-DBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC no-ID loop-detect inotify dumpfile
Tue Jun 25 17:17:55 2019 daemon.info dnsmasq-dhcp[22130]: DHCP, IP range 192.168.100.100 -- 192.168.100.249, lease time 12h
Tue Jun 25 17:17:55 2019 daemon.info dnsmasq[22130]: using local addresses only for domain test
Tue Jun 25 17:17:55 2019 daemon.info dnsmasq[22130]: using local addresses only for domain onion
Tue Jun 25 17:17:55 2019 daemon.info dnsmasq[22130]: using local addresses only for domain localhost
Tue Jun 25 17:17:55 2019 daemon.info dnsmasq[22130]: using local addresses only for domain local
Tue Jun 25 17:17:55 2019 daemon.info dnsmasq[22130]: using local addresses only for domain invalid
Tue Jun 25 17:17:55 2019 daemon.info dnsmasq[22130]: using local addresses only for domain bind
Tue Jun 25 17:17:55 2019 daemon.info dnsmasq[22130]: using nameserver 127.0.0.1#5353
Tue Jun 25 17:17:55 2019 daemon.info dnsmasq[22130]: using local addresses only for domain lan
Tue Jun 25 17:17:55 2019 daemon.info dnsmasq[22130]: read /etc/hosts - 4 addresses
Tue Jun 25 17:17:55 2019 daemon.info dnsmasq[22130]: read /tmp/hosts/dhcp.cfg01411c - 5 addresses
Tue Jun 25 17:17:55 2019 daemon.info dnsmasq-dhcp[22130]: read /etc/ethers - 0 addresses
Tue Jun 25 17:17:55 2019 user.notice omr-bypass: OMR-ByPass is running
Tue Jun 25 17:18:06 2019 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0
Tue Jun 25 17:18:06 2019 user.notice post-tracking-post-tracking: Tunnel up disable use of ShadowSocks for UDP
Tue Jun 25 17:18:06 2019 user.notice Shadowsocks: Rules DOWN
Tue Jun 25 17:18:07 2019 user.notice Shadowsocks: Rules UP
Tue Jun 25 17:18:07 2019 user.notice Shadowsocks: Reload omr-bypass rules
Tue Jun 25 17:18:07 2019 user.notice omr-bypass: Starting OMR-ByPass...
Tue Jun 25 17:18:08 2019 daemon.err omr-tracker-ss[4889]: Another app is currently holding the xtables lock. Perhaps you want to use the -w option?
Tue Jun 25 17:18:08 2019 daemon.info omr-tracker-ss: Reload Shadowsocks rules
Tue Jun 25 17:18:10 2019 daemon.info dnsmasq[22130]: exiting on receipt of SIGTERM
Tue Jun 25 17:18:10 2019 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Tue Jun 25 17:18:10 2019 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Tue Jun 25 17:18:14 2019 daemon.info dnsmasq[23627]: started, version 2.80 cachesize 150
Tue Jun 25 17:18:14 2019 daemon.info dnsmasq[23627]: DNS service limited to local subnets
Tue Jun 25 17:18:14 2019 daemon.info dnsmasq[23627]: compile time options: IPv6 GNU-getopt no-DBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC no-ID loop-detect inotify dumpfile
Tue Jun 25 17:18:14 2019 daemon.info dnsmasq-dhcp[23627]: DHCP, IP range 192.168.100.100 -- 192.168.100.249, lease time 12h
Tue Jun 25 17:18:14 2019 daemon.info dnsmasq[23627]: using local addresses only for domain test
Tue Jun 25 17:18:14 2019 daemon.info dnsmasq[23627]: using local addresses only for domain onion
Tue Jun 25 17:18:14 2019 daemon.info dnsmasq[23627]: using local addresses only for domain localhost
Tue Jun 25 17:18:14 2019 daemon.info dnsmasq[23627]: using local addresses only for domain local
Tue Jun 25 17:18:14 2019 daemon.info dnsmasq[23627]: using local addresses only for domain invalid
[... cut ...]
Tue Jun 25 17:18:14 2019 daemon.info dnsmasq[23627]: using nameserver 127.0.0.1#5353
Tue Jun 25 17:18:14 2019 daemon.info dnsmasq[23627]: using local addresses only for domain lan
Tue Jun 25 17:18:14 2019 daemon.info dnsmasq[23627]: read /etc/hosts - 4 addresses
Tue Jun 25 17:18:14 2019 daemon.info dnsmasq[23627]: read /tmp/hosts/dhcp.cfg01411c - 5 addresses
Tue Jun 25 17:18:14 2019 daemon.info dnsmasq-dhcp[23627]: read /etc/ethers - 0 addresses
Tue Jun 25 17:18:14 2019 user.notice omr-bypass: OMR-ByPass is running

>

Florent4223, 2019-06-25:
Yannick  j aimerai vraiment ré tester avec la version précédente du vps ce devait être la 0.993 
Ça me permettrait d etre sur que le problème rencontré n était pas présent
As tu cette possibilité? 
Merci

Ysurac, 2019-06-25:
Toutes les versions sont ici : https://github.com/Ysurac/openmptcprouter-vps
Le seul changement important est la mise à jour du noyau.

Florent4223, 2019-06-25:
Merci
Par contre pour réinstaller l ancienne version dans son ensemble comment je m y prends?

oui c est ca que je comprends pas y a peu de modifs et c est aussi pour cela que j aimerai être sur que le problème date pas d avant

J ignore si tu as pu regarder mais dans l s log systeme à part des déconnexions du vpn qui relance tout le réseau (y compris server dhcp) y a rien

Ysurac, 2019-06-25:
La version 0.993 est ici : https://github.com/Ysurac/openmptcprouter-vps/raw/5cd78a9401c59e3216bcea8a5691cef0e249d309/debian9-x86_64.sh

Pour le redémarrage de dnsmasq, ça sera réglé dans la prochaine version.
Niveau VPS, tu peux regarder dans /var/log/daemon.log (sous debian) si il y a quelque chose lors des déconnexions. 
Le VPS a t'il assez de bande passante ?
Il faudrait aussi voir la charge CPU niveau routeur et VPS quand tu télécharges.

Florent4223, 2019-06-25:
Merci de ton aide

Non c est sur que le vps n a pas suffisement de bande passante
J ai regarde le daemon.log ras. Je mettrais une copie demain mais au premier abord je n ai rien vu d anormal
En realité c est un simple serveur de test avec une connexion 8M down et 1M up
Une seule clé 4g suffirait à saturer

Néanmoins le vpn devrait s adapter non?

Par qd j upload un fichier sur un serveur web présent derrière l omr via internet dc j utilisé le upload des clé 4g  le down de la connexion adsl et enfin le up de la connexion adsl j ai un débit de 1M en continu sans problème et aucun déconnexion du vpn

le pb est vraiment quand j upload directement entre l omr et le réseau ou est le serveur vps. La les 8M down sont saturés

Ysurac, 2019-06-25:
Ca peut expliquer le soucis. J'ai toujours considéré que le VPS devait avoir une BP > à la BP que l'on peut agréger.
Il n'y a donc rien sur le serveur pour gérer le cas où il n'a pas assez de BP :)

Florent4223, 2019-06-25:
Sur le serveur définitif j espere qu il y aura une fibre
Néanmoins deux clés 4g à plein régime cela turbine
Y aura t il un.moyen d intégrer un mécanisme? 

Je suis qd même surpris que glorytun ne sache pas faire le max de lui meme

Florent4223, 2019-06-26:
after test same pb with vps 0.993 et omr 0.49.7

Can we imagine a solution to limit the impact?

I also have the impression when I force in 3G so debit aggregate lower than 8M, the tunnel has cut too

tentious, 2019-06-28:
I also have the same or similar problem. With glorytun tcp enabled and when connection is saturated, the glorytun VPN drops for 5-8 seconds, stopping all UDP traffic. This seems to have started in version 0.49 or 0.50. I upgraded from 0.47(not sure of vps script version from before) to latest and now the problems starts. I can provide more details later.

tentious, 2019-06-28:
So I can confirm that after downgrading only the VPS to script version 0.992 using kernel 4.14.110-mptcp the problem goes away. I've kept the raspberry pi at OMR version 0.5.2

https://raw.githubusercontent.com/Ysurac/openmptcprouter-vps/616cb5c4944a1cf648133195bb702c7cadb5f50d/debian9-x86_64.sh

@Florent4223 - Make sure after you downgrade that you restart the VPS and that you're using the old kernel. You can verify with **"uname** -r".

I will now try fork the vps script and attempt to downgrade only the mptcp kernel and test.

tentious, 2019-06-28:
@Florent4223 I've modified the vps script here:https://github.com/tentious/openmptcprouter-vps/blob/master/debian9-x86_64.sh
This is latest commit from @Ysurac  with only the kernel downgrade.

@Ysurac -Thanks for you very hard work on this project!

Florent4223, 2019-06-28:
Thanks for your work
yes with 0.993 the kernel is 4.14.110 but glorytun not stable...strange

With the script on last vps version 0.996, only change the kernel?  Keys changed or save?
Or i do a new install with script?

tentious, 2019-06-28:
So, actually it appears that the old kernel doesn't fix my disconnect problems, but with the old kernel it seems that the saturation is fixed. Glorytun still disconnects every so often, even with zero/no load/saturation.
Fri Jun 28 12:00:00 2019 user.info vnstat_backup: backing up database
Fri Jun 28 12:21:47 2019 user.notice post-tracking-post-tracking: Get status and settings for wan1...
Fri Jun 28 12:22:08 2019 daemon.info glorytun[3368]: STOPPED tun0
Fri Jun 28 12:22:09 2019 daemon.err glorytun[3368]: 45.32.219.105.65001: connected
Fri Jun 28 12:22:09 2019 daemon.info glorytun[3368]: STARTED tun0
Fri Jun 28 12:22:15 2019 user.notice post-tracking-post-tracking: Get status and settings for wan1... Done
Fri Jun 28 12:37:52 2019 daemon.info glorytun[3368]: STOPPED tun0
Fri Jun 28 12:37:53 2019 daemon.err glorytun[3368]: 45.32.219.105.65001: connected
Fri Jun 28 12:37:53 2019 daemon.info glorytun[3368]: STARTED tun0
Fri Jun 28 12:53:33 2019 user.notice post-tracking-post-tracking: Get status and settings for verizon...
Fri Jun 28 12:53:35 2019 daemon.info glorytun[3368]: STOPPED tun0
Fri Jun 28 12:53:36 2019 daemon.err glorytun[3368]: 45.32.219.105.65001: connected
Fri Jun 28 12:53:36 2019 daemon.info glorytun[3368]: STARTED tun0
Fri Jun 28 12:56:21 2019 user.notice post-tracking-post-tracking: Get status and settings for verizon... Done
Fri Jun 28 13:09:18 2019 daemon.info glorytun[3368]: STOPPED tun0
Fri Jun 28 13:09:20 2019 daemon.err glorytun[3368]: 45.32.219.105.65001: connected
Fri Jun 28 13:09:20 2019 daemon.info glorytun[3368]: STARTED tun0

tentious, 2019-06-28:
@Florent4223 

Right, so after some investigation, appears glorytun is reconnecting every time my VPS asks dhcp for a new address(even though it never changes). Notice the times match up perfectly. I'll set my VPS to static IP and see if the problem goes away.

**From my VPS:**

```
Jun 28 13:56:27 XX.XX systemd-networkd[21714]: ens3: DHCPv4 address XX.XX.219.105/22 via XX.XX.216.1
Jun 28 14:17:40 XX.XX systemd-networkd[23624]: ens3: DHCPv4 address XX.XX.219.105/22 via XX.XX.216.1
Jun 28 14:29:24 XX.XX systemd-networkd[24620]: ens3: DHCPv4 address XX.XX.219.105/22 via XX.XX.216.1
Jun 28 15:26:05 XX.XX systemd-networkd[32098]: ens3: DHCPv4 address XX.XX.219.105/22 via XX.XX.216.1
Jun 28 15:47:08 XX.XX systemd-networkd[1605]: ens3: DHCPv4 address XX.XX.219.105/22 via XX.XX.216.1
```

**From OMR:**

```
Fri Jun 28 13:56:27 2019 daemon.info glorytun[6373]: STOPPED tun0
Fri Jun 28 13:56:28 2019 daemon.err glorytun[6373]: XX.XX219.105.65001: connected
Fri Jun 28 13:56:28 2019 daemon.info glorytun[6373]: STARTED tun0
Fri Jun 28 14:17:39 2019 daemon.info glorytun[6373]: STOPPED tun0
Fri Jun 28 14:17:40 2019 daemon.err glorytun[6373]: XX.XX219.105.65001: connected
Fri Jun 28 14:17:41 2019 daemon.info glorytun[6373]: STARTED tun0
Fri Jun 28 14:29:24 2019 daemon.info glorytun[6373]: STOPPED tun0
Fri Jun 28 14:29:25 2019 daemon.err glorytun[6373]: XX.XX219.105.65001: connected
Fri Jun 28 14:29:25 2019 daemon.info glorytun[6373]: STARTED tun0
Fri Jun 28 14:29:25 2019 daemon.info glorytun[6373]: STOPPED tun0
Fri Jun 28 14:29:26 2019 daemon.err glorytun[6373]: XX.XX219.105.65001: connected
Fri Jun 28 14:29:26 2019 daemon.info glorytun[6373]: STARTED tun0
Fri Jun 28 15:26:05 2019 daemon.info glorytun[6373]: STOPPED tun0
Fri Jun 28 15:26:06 2019 daemon.err glorytun[6373]: XX.XX219.105.65001: connected
Fri Jun 28 15:26:06 2019 daemon.info glorytun[6373]: STARTED tun0
Fri Jun 28 15:47:08 2019 daemon.info glorytun[6373]: STOPPED tun0
Fri Jun 28 15:47:09 2019 daemon.err glorytun[6373]: XX.XX219.105.65001: connected
Fri Jun 28 15:47:09 2019 daemon.info glorytun[6373]: STARTED tun0
Fri Jun 28 15:47:09 2019 daemon.info glorytun[6373]: STOPPED tun0
Fri Jun 28 15:47:10 2019 daemon.err glorytun[6373]: XX.XX219.105.65001: connected
Fri Jun 28 15:47:10 2019 daemon.info glorytun[6373]: STARTED tun0
```

ArChlOgY, 2019-06-28:
That's good input , I will have a try . Thanks !

Florent4223, 2019-06-29:
My vps is static address and glorytun disconnect are present

tentious, 2019-06-29:
After changing my VPS to static I also still have the glorytun disconnects every 15 - 30 mintues. There must be some relation to the VPS WAN NIC because the DHCP request matches nearly perfect when DHCP is enable on the VPS. Does the glorytun disconnects cause the local NIC to also perform a DHCP renewal? Is there a verbose setting that could be enabled for glorytun to be more verbose?

Ysurac, 2019-06-29:
You can try latest alpha release: https://www.openmptcprouter.com/release/4.19/
And latest alpha VPS script http://www.openmptcprouter.com/server/debian9-x86_64-test.sh

This is not yet tested at all, but there is many changes.

tentious, 2019-06-29:
@Ysurac 
I've tried latest alpha release, both OMR and VPS script and also still get glorytun TCP disconnects.

`
Sat Jun 29 15:44:47 2019 daemon.info glorytun[20910]: STOPPED tun0
Sat Jun 29 15:44:49 2019 daemon.err glorytun[20910]: XX.XX.219.105.65001: connected
Sat Jun 29 15:44:49 2019 daemon.info glorytun[20910]: STARTED tun0
Sat Jun 29 16:00:00 2019 cron.err crond[2958]: USER root pid 7695 cmd /etc/init.d/vnstat_backup backup
Sat Jun 29 16:00:00 2019 user.info vnstat_backup: backing up database
Sat Jun 29 16:12:17 2019 daemon.info glorytun[20910]: STOPPED tun0
Sat Jun 29 16:12:18 2019 daemon.err glorytun[20910]: XX.XX.219.105.65001: connected
Sat Jun 29 16:12:18 2019 daemon.info glorytun[20910]: STARTED tun0
Sat Jun 29 16:22:52 2019 user.notice post-tracking-post-tracking: Get status and settings for wan1...
Sat Jun 29 16:23:21 2019 user.notice post-tracking-post-tracking: Get status and settings for wan1... Done
Sat Jun 29 16:30:55 2019 user.notice post-tracking-post-tracking: Get status and settings for verizon...
Sat Jun 29 16:33:43 2019 user.notice post-tracking-post-tracking: Get status and settings for verizon... Done
Sat Jun 29 16:42:30 2019 daemon.info glorytun[20910]: STOPPED tun0
Sat Jun 29 16:42:31 2019 daemon.err glorytun[20910]: XX.XX.219.105.65001: connected
Sat Jun 29 16:42:31 2019 daemon.info glorytun[20910]: STARTED tun0
Sat Jun 29 16:42:31 2019 daemon.info glorytun[20910]: STOPPED tun0
Sat Jun 29 16:42:32 2019 daemon.err glorytun[20910]: XX.XX.219.105.65001: connected
Sat Jun 29 16:42:32 2019 daemon.info glorytun[20910]: STARTED tun0
`

tentious, 2019-07-20:
@Florent4223 
I was able to eliminate my glorytun issues by using Debian 9 vs Ubuntu 18.04. I'm now running latest official release(not beta/alpha) without any problems. Apparently glorytun tcp does not like Ubuntu 18.04.

Florent4223, 2019-07-20:
Thanks for yourte help
But i have debian 9...

Marctraider, 2019-08-10:
Hi! I have same problem, after 23 hour full uptime no problem, glorytun got DC. 

No WAN did go down in this time.. Also using Ubuntu 18.04 on VPS anyway.

closed 2019-12-20 by Ysurac