Ysurac/openmptcprouter #526: forward port problem after change vpn

Florent4223, 2019-06-24:
## Expected Behavior

Forward port on client behind omr

## Actual Behavior

Configuration forward port on omr
Forward port on firewall
Forward ok

Change type vpn
But nothing forward

## Steps to Reproduce the Problem

  1. Forward port 80 on client behind omr
  2. Try connect with pc on same vps network  ok
  3. Change type vpn
 4.  Try port forward ....nothing

## Specifications

  - OpenMPTCProuter version: (Last is not a version) 0.50.2
  - OpenMPTCProuter VPS version: (Last is not a version)  0.998
  - OpenMPTCProuter platform: (RPI2/RPI3/x86/x86_64) rpi3

Ysurac, 2019-06-24:
For now you need to wait a little, a script on the VPS detect current VPN.
This will be better in next release.

Florent4223, 2019-06-24:
Thanks
No Problem

closed 2019-12-20 by Ysurac