Ysurac/openmptcprouter #529: glorytun udp low flow

Florent4223, 2019-06-24:
## Expected Behavior

 transfer with aggregation of 2 4G keys of a file

## Actual Behavior

the transfer of a file will only rarely 600KB / s aggregated 2 4G connections.
We should have less 1MB / s in almost continuous

## Steps to Reproduce the Problem

  1. configure glorytun udp
  2. transfert a file
  3. see flow

## Specifications

  - OpenMPTCProuter version: (Last is not a version) 0.49.7
  - OpenMPTCProuter VPS version: (Last is not a version) 0.993
  - OpenMPTCProuter platform: (RPI2/RPI3/x86/x86_64) rpi3

Ysurac, 2019-06-24:
Now you know why it's not the default ;)

Florent4223, 2019-06-24:
sorry but I do not understand...

Ysurac, 2019-06-24:
I don't always have good result with Glorytun UDP, so it's why it's not the default VPN.

Florent4223, 2019-06-24:
I see
But to have a good agregation multioperator (bouygue blocks the mptcp apaprement on my subscription) that remains t it?

Ysurac, 2019-06-24:
No French operator block MPTCP to my knowledge.

Florent4223, 2019-06-24:
I have with Bouygue I have the message of "probable blocking of the mptcp"

and actually with this connection agregation works very badly

Florent4223, 2019-06-24:
I would do tests tomorrow
but even in glory tun tcp I have a pb of stability of vpn (cf other issue)

Ysurac, 2019-06-24:
To check if it's really the case, there is tracebox installed. Available in "Network->MPTCP->MPTCP Support Check", this may give (or not) some info.

Florent4223, 2019-06-24:
yes and I had losses with bouygues

with orange OK

closed 2019-12-20 by Ysurac