Ysurac/openmptcprouter #530: NIC shutting down itself

JorgeMoort89, 2019-06-25:
Randomly the nic, setted in DHCP, reboot itself, I've monitored it also with Observium via snmp
I've this problem on 0.49.6 and 0.50.2.
That's the configuration on 0.49.6:

``collectd.globals=globals
collectd.globals.alt_config_file='/etc/collectd.conf'
dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].boguspriv='1'
dhcp.@dnsmasq[0].filterwin2k='0'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].resolvfile='/tmp/resolv.conf.auto'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].server='127.0.0.1#5353'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].ipset='/hwcdn.net/omr-cs2' '/s3.ll.dash.row.aiv-cdn.net/omr-cs4' '/drive.google.com/omr-cs1' '/docs.googleusercontent.com/omr-cs1' '/docs.googleusercontent.com/omr-cs1' '/download.qq.com/omr-cs1' '/googleusercontent.com/omr-cs2' '/gvt1.com/omr-cs1' '/googletagmanager.com/omr-cs2' '/d25xi40x97liuc.cloudfront.net/omr-cs4' '/dropboxstatic.com/omr-cs1' '/sndcdn.com/omr-cs4' '/vevo.com/omr-cs4' '/akamaihd.net/omr-cs2' '/whatsapp.com/omr-cs2' '/mmg-fna.whatsapp.net/omr-cs1' '/deezer.com/omr-cs4' '/docs.google.com/omr-cs1' '/google.com/omr-cs2' '/aiv-delivery.net/omr-cs4' '/googlevideo.com/omr-cs4' '/aiv-delivery.net/omr-cs4' '/gvt1.com/omr-cs1' '/upload.youtube.com/omr-cs1' '/dropboxstatic.com/omr-cs1' '/fbcdn.net/omr-cs4' '/gvt1.com/omr-cs1' '/drive-thirdparty.googleusercontent.com/omr-cs1' '/nflxvideo.net/omr-cs4' '/upload.youtube.com/omr-cs1' '/google.com/omr-cs2' '/upload.video.google.com/omr-cs1' '/upload.video.google.com/omr-cs1' '/ttvnw.net/omr-cs4' '/googleapis.com/omr-cs2' '/audio-fa.scdn.cot/omr-cs4' '/hwcdn.net/omr-cs2' '/audio-fa.scdn.cot/omr-cs4' '/google.com/omr-cs2' '/dropbox-dns.com/omr-cs1' '/log.getdropbox.com/omr-cs1' '/vevo.com/omr-cs4' '/mmg-fna.whatsapp.net/omr-cs1' '/windowsupdate.com/omr-cs1' '/fbcdn.net/omr-cs2' '/hwcdn.net/omr-cs2' '/windowsupdate.com/omr-cs1' '/update.microsoft.com/omr-cs1' '/drive.google.com/omr-cs1' '/log.getdropbox.com/omr-cs1' '/deezer.com/omr-cs4' '/drive-thirdparty.googleusercontent.com/omr-cs1' '/deezer.com/omr-cs4' '/deezer.com/omr-cs4' '/audio-fa.scdn.cot/omr-cs4' '/akamaihd.net/omr-cs2' '/s3.ll.dash.row.aiv-cdn.net/omr-cs4' '/upload.youtube.com/omr-cs1' '/s3.ll.dash.row.aiv-cdn.net/omr-cs4' '/googlevideo.com/omr-cs4' '/docs.google.com/omr-cs1' '/update.microsoft.com/omr-cs1' '/deezer.com/omr-cs4' '/whatsapp.net/omr-cs2' '/gs2.ww.prod.dl.playstation.net/omr-cs1' '/sndcdn.com/omr-cs4' '/sndcdn.com/omr-cs4' '/last.fm/omr-cs4' '/docs.googleusercontent.com/omr-cs1' '/whatsapp.com/omr-cs2' '/last.fm/omr-cs4' '/whatsapp.net/omr-cs2' '/s3.ll.dash.row.aiv-cdn.net/omr-cs4' '/fbcdn.net/omr-cs4' '/gvt1.com/omr-cs1' '/v.redd.it/omr-cs4' '/upload.video.google.com/omr-cs1' '/dropbox.com/omr-cs1' '/d25xi40x97liuc.cloudfront.net/omr-cs4' '/ttvnw.net/omr-cs4' '/googleapis.com/omr-cs2' '/last.fm/omr-cs4' '/googletagmanager.com/omr-cs2' '/mmg-fna.whatsapp.net/omr-cs1' '/log.getdropbox.com/omr-cs1' '/sndcdn.com/omr-cs4' '/googleusercontent.com/omr-cs2' '/1e100.net/omr-cs2' '/sndcdn.com/omr-cs4' '/google.com/omr-cs2' '/v.redd.it/omr-cs4' '/hwcdn.net/omr-cs2' '/d25xi40x97liuc.cloudfront.net/omr-cs4' '/last.fm/omr-cs4' '/upload.video.google.com/omr-cs1' '/upload.youtube.com/omr-cs1' '/fbcdn.net/omr-cs2' '/steamcontent.com/omr-cs1' '/fbcdn.net/omr-cs2' '/akamaihd.net/omr-cs2' '/download.qq.com/omr-cs1' '/whatsapp.net/omr-cs2' '/windowsupdate.com/omr-cs1' '/upload.video.google.com/omr-cs1' '/fbcdn.net/omr-cs4' '/aiv-delivery.net/omr-cs4' '/googleusercontent.com/omr-cs2' '/steamcontent.com/omr-cs1' '/whatsapp.com/omr-cs2' '/download.qq.com/omr-cs1' '/update.microsoft.com/omr-cs1' '/windowsupdate.com/omr-cs1' '/googleapis.com/omr-cs2' '/fbcdn.net/omr-cs2' '/gs2.ww.prod.dl.playstation.net/omr-cs1' '/1e100.net/omr-cs2' '/ttvnw.net/omr-cs4' '/akamaihd.net/omr-cs2' '/download.qq.com/omr-cs1' '/update.microsoft.com/omr-cs1' '/hwcdn.net/omr-cs2' '/docs.googleusercontent.com/omr-cs1' '/fbcdn.net/omr-cs4' '/d25xi40x97liuc.cloudfront.net/omr-cs4' '/googlevideo.com/omr-cs4' '/gs2.ww.prod.dl.playstation.net/omr-cs1' '/googletagmanager.com/omr-cs2' '/dropbox.com/omr-cs1' '/whatsapp.com/omr-cs2' '/googlevideo.com/omr-cs4' '/aiv-delivery.net/omr-cs4' '/nflxvideo.net/omr-cs4' '/dropboxstatic.com/omr-cs1' '/drive.google.com/omr-cs1' '/whatsapp.net/omr-cs2' '/dropboxstatic.com/omr-cs1' '/last.fm/omr-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr-cs4' '/ttvnw.net/omr-cs4' '/googleusercontent.com/omr-cs2' '/windowsupdate.com/omr-cs1' '/dropbox-dns.com/omr-cs1' '/v.redd.it/omr-cs4' '/whatsapp.com/omr-cs2' '/steamcontent.com/omr-cs1' '/d25xi40x97liuc.cloudfront.net/omr-cs4' '/dropbox-dns.com/omr-cs1' '/nflxvideo.net/omr-cs4' '/drive.google.com/omr-cs1' '/ttvnw.net/omr-cs4' '/ttvnw.net/omr-cs4' '/akamaihd.net/omr-cs2' '/log.getdropbox.com/omr-cs1' '/steamcontent.com/omr-cs1' '/aiv-delivery.net/omr-cs4' '/google.com/omr-cs2' '/gvt1.com/omr-cs1' '/drive.google.com/omr-cs1' '/googleapis.com/omr-cs2' '/vevo.com/omr-cs4' '/dropbox.com/omr-cs1' '/fbcdn.net/omr-cs4' '/fbcdn.net/omr-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr-cs4' '/ttvnw.net/omr-cs4' '/mmg-fna.whatsapp.net/omr-cs1' '/drive-thirdparty.googleusercontent.com/omr-cs1' '/gs2.ww.prod.dl.playstation.net/omr-cs1' '/googleapis.com/omr-cs2' '/d25xi40x97liuc.cloudfront.net/omr-cs4' '/ttvnw.net/omr-cs4' '/docs.google.com/omr-cs1' '/gs2.ww.prod.dl.playstation.net/omr-cs1' '/vevo.com/omr-cs4' '/ttvnw.net/omr-cs4' '/audio-fa.scdn.cot/omr-cs4' '/drive-thirdparty.googleusercontent.com/omr-cs1' '/docs.googleusercontent.com/omr-cs1' '/update.microsoft.com/omr-cs1' '/vevo.com/omr-cs4' '/drive-thirdparty.googleusercontent.com/omr-cs1' '/v.redd.it/omr-cs4' '/gvt1.com/omr-cs1' '/audio-fa.scdn.cot/omr-cs4' '/docs.google.com/omr-cs1' '/mmg-fna.whatsapp.net/omr-cs1' '/aiv-delivery.net/omr-cs4' '/whatsapp.net/omr-cs2' '/deezer.com/omr-cs4' '/sndcdn.com/omr-cs4' '/vevo.com/omr-cs4' '/upload.youtube.com/omr-cs1' '/1e100.net/omr-cs2' '/last.fm/omr-cs4' '/dropbox.com/omr-cs1' '/googletagmanager.com/omr-cs2' '/upload.video.google.com/omr-cs1' '/ttvnw.net/omr-cs4' '/windowsupdate.com/omr-cs1' '/dropboxstatic.com/omr-cs1' '/v.redd.it/omr-cs4' '/log.getdropbox.com/omr-cs1' '/ttvnw.net/omr-cs4' '/update.microsoft.com/omr-cs1' '/ttvnw.net/omr-cs4' '/upload.youtube.com/omr-cs1' '/googletagmanager.com/omr-cs2' '/googlevideo.com/omr-cs4' '/audio-fa.scdn.cot/omr-cs4' '/googleusercontent.com/omr-cs2' '/googlevideo.com/omr-cs4' '/nflxvideo.net/omr-cs4' '/docs.google.com/omr-cs1' '/s3.ll.dash.row.aiv-cdn.net/omr-cs4' '/dropbox-dns.com/omr-cs1' '/dropbox.com/omr-cs1' '/google.com/omr-cs2' '/fbcdn.net/omr-cs2' '/googletagmanager.com/omr-cs2' '/d25xi40x97liuc.cloudfront.net/omr-cs4' '/akamaihd.net/omr-cs2' '/aiv-delivery.net/omr-cs4' '/whatsapp.net/omr-cs2' '/fbcdn.net/omr-cs4' '/whatsapp.com/omr-cs2' '/ttvnw.net/omr-cs4' '/vevo.com/omr-cs4' '/googleapis.com/omr-cs2' '/audio-fa.scdn.cot/omr-cs4' '/1e100.net/omr-cs2' '/deezer.com/omr-cs4' '/hwcdn.net/omr-cs2' '/sndcdn.com/omr-cs4' '/download.qq.com/omr-cs1' '/last.fm/omr-cs4' '/steamcontent.com/omr-cs1' '/v.redd.it/omr-cs4' '/gs2.ww.prod.dl.playstation.net/omr-cs1' '/ttvnw.net/omr-cs4' '/dropbox.com/omr-cs1' '/dropboxstatic.com/omr-cs1' '/googletagmanager.com/omr-cs2' '/googleusercontent.com/omr-cs2' '/dropbox-dns.com/omr-cs1' '/log.getdropbox.com/omr-cs1' '/google.com/omr-cs2' '/drive.google.com/omr-cs1' '/fbcdn.net/omr-cs2' '/drive-thirdparty.googleusercontent.com/omr-cs1' '/akamaihd.net/omr-cs2' '/docs.google.com/omr-cs1' '/whatsapp.net/omr-cs2' '/docs.googleusercontent.com/omr-cs1' '/whatsapp.com/omr-cs2' '/gvt1.com/omr-cs1' '/googleapis.com/omr-cs2' '/mmg-fna.whatsapp.net/omr-cs1' '/1e100.net/omr-cs2' '/upload.youtube.com/omr-cs1' '/hwcdn.net/omr-cs2' '/download.qq.com/omr-cs1' '/upload.video.google.com/omr-cs1' '/windowsupdate.com/omr-cs1' '/steamcontent.com/omr-cs1' '/update.microsoft.com/omr-cs1' '/gs2.ww.prod.dl.playstation.net/omr-cs1' '/dropbox.com/omr-cs1' '/dropboxstatic.com/omr-cs1' '/dropbox-dns.com/omr-cs1' '/log.getdropbox.com/omr-cs1' '/drive.google.com/omr-cs1' '/googlevideo.com/omr-cs4' '/nflxvideo.net/omr-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr-cs4' '/d2 [... cut ...]
Tue Jun 25 09:27:04 2019 user.notice post-tracking-post-tracking: Get status and settings for wan4... Done
Tue Jun 25 09:27:06 2019 daemon.notice netifd: Network device 'eth4' link is up
Tue Jun 25 09:27:06 2019 daemon.notice netifd: Interface 'wan4' has link connectivity
Tue Jun 25 09:27:06 2019 kern.info kernel: [60481.885696] e1000e: eth4 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: Rx/Tx`

Also sometimes the DHCP server on the LAN seems doesen't working

## Specifications

  - OpenMPTCProuter version: 0.50.2
  - OpenMPTCProuter VPS version: 0.996
  - OpenMPTCProuter platform: x86_64

JorgeMoort89, 2019-06-25:
Deleted redundant parts, I hope it is more readable now

Ysurac, 2019-06-25:
True eth4 is see as down. Check that you don't have a switch that put the interface down and that your DHCP doesn't give an IP already take on network.
This doesn't seems to be a software problem on OMR side.

JorgeMoort89, 2019-06-25:
There is no switch putting down the interface, I had this problem with two different wan with two different router, can maybe be an hardware problem? I've already seen a lot of MTU change, can that be a part of the problem?

Ysurac, 2019-06-25:
Should not, but you can fix the MTU in Network config of each wan.

JorgeMoort89, 2019-06-27:
I've noticed that with previous version I don't loose the IP in DHCP every about 60 minutes. I've alao tried 5 different routers. Only with this version I have this problem

JorgeMoort89, 2019-06-27:
[image]
How you can notice the router have 6 days uptime but the WAN2 only 23 minutes. I've tried a lot of different router (x64 not efi) and different WANs

JorgeMoort89, 2019-06-27:
[image]
This is my router with 4 WANs and OpenMPTCProuter 0.48

JorgeMoort89, 2019-06-27:
@Ysurac I've just tried 0.48, 0.49 and 0.50 on the same router with the same two WANs, with those versions I have no problem at all with the WANs loosing IP, only with 0.50 with 0.994 VPS version I have some speed stability problem. On 0.52, every 60 minutes, the WANs reboot itself, probably cause DHCP renew.

closed 2019-12-20 by Ysurac