Ysurac/openmptcprouter #531: Error Wan : "The document has moved"

xslite, 2019-06-26:
## Expected Behavior

in luci/admin/system/openmptcprouter/status  should display all WAN that are good and running with a green tick.

## Actual Behavior

Shows a Red X for Wan2, and error message : " the document has moved "
while the Wan2 interface is correctly running.

[image]

## Specifications

  - OpenMPTCProuter version: 0.49.3
  - OpenMPTCProuter VPS version: 0.993
  - OpenMPTCProuter platform: x86_64

Ysurac, 2019-06-26:
For the "document has moved", this will be fixed alone in next IP check (A problem on site called to get IP address).
No idea for the red X, I need to display more info here...

closed 2019-06-26 by Ysurac

xslite, 2019-06-26:
And when does the next ip check should happen? 

What is really strange is that the same error message (the document has loves) is still displayed even after a full reboot of omptcp router...

Ysurac, 2019-06-26:
IP check is done every hours.
But I think you have another problem on wan2. You try to submit the wizard again, this should restart all checks.