Ysurac/openmptcprouter #532: Possibilité d'installer un outil de QoS

silver9195, 2019-06-28:
## Expected Behavior

Add a new tool

## Actual Behavior

## Steps to Reproduce the Problem

  1.
  2.
  3.

## Specifications

  - OpenMPTCProuter version: 50.2
  - OpenMPTCProuter VPS version: VPS 0.991 
  - OpenMPTCProuter platform: RPI3/x86

Bonjour Ysurac, 

Merci encore pour tes nouvelles mises à jours et ton travail.

Actuellement comme pas mal de personne j'ai un acces 4G en plus de ma connexion ADSL (5Mb...)  .

Tout fonctionne bien maintenant mais sur mes 100Go de data 4G je suis en moyenne à 10G par jour ce qui me fait pas tenir très longtemps.

Je voulais donc installer un outil de QoS pour regarder qui prend le plus de bande passante et en faisant quoi.

Je voulais installer ntop par exemple mais je vois que sur le raspberry je ne peux plus faire de apt-get ou de aptitude.

As tu une idée pour me dépanner ?

Merci.

Ysurac, 2019-06-28:
Il y a déjà SQM QoS d'installer, mais je pense que ce n'est pas ce que tu veux ;)
Tu peux par contre regarder dans Etat->graphiques temps réels pour avoir ce genre de donnée.
Sinon le système utilisé comme base est OpenWrt et non Debian, donc pas de apt-get mais il y a opkg ou via l'interface graphique.

silver9195, 2019-07-01:
ok je vais regarder merci

silver9195, 2019-07-02:
Je me permet de re écrire dans le même ticket, mais j'ai fais les mises à jours mais depuis mes redirections de ports ne fonctionnent plus. J'ai eu plusieurs soucis avec le vpn qui ne montait pas j'ai relancer plusieurs fois le script d'instal sur le vps mais maintenant tout est au vert.

Mais il reste juste mes redirections de port qui ne fonctionnent pas :

[image]

[image]

Dragonox77, 2019-07-03:
Salut pour tes redirections, as tu bien coché la case "Rediriger tous les ports du serveur vers OpenMPTCProuter" sur la page:
http://192.168.100.1/cgi-bin/luci/admin/system/openmptcprouter/settings

silver9195, 2019-07-03:
Merci de ta réponse ,

Yes, je l'ai coché mais rien ... une autre idée ?

silver9195, 2019-07-05:
Après un reboot et la case c'est tout bon. 

On peut clore cet issue. 

Merci les amis

closed 2019-07-07 by Ysurac