Ysurac/openmptcprouter #533: WiFi ap mode not working

mshost, 2019-06-29:
## Expected Behavior
WiFi ap as access point in PI 3b+
  In English please.

## Actual Behavior
When i am trying to configure pi as WiFi router , it's successfully create the interface in wireless as AP but after couple of minutes it is changed to wireless client mode.

## Steps to Reproduce the Problem

  1.delete wireless interface,
  2.restart
  3.again install , but no luck

## Specifications

  - OpenMPTCProuter version: (0.50)
  - OpenMPTCProuter VPS version: (0.995)
  - OpenMPTCProuter platform: (/RPI3b+)

mshost, 2019-06-30:
VPS version 0.996

Ysurac, 2019-06-30:
I'm not able to reproduce this issue.

mshost, 2019-06-30:
Is it possible to enable WiFi ap in raspberry pi 3b+ ?

Ysurac, 2019-06-30:
Yes. Set country, canal, ssid and this should work. Else change settings.

closed 2019-07-21 by Ysurac