Ysurac/openmptcprouter #534: static route problem

Anthony010101, 2019-06-30:
Hello,

I'm trying to replace the OverTheBox with the OpenTCProuter.
We have OTB site-to-site IPSec VPNs behind the router. All with three static routes, allowing us to access remote sites.

When, I activate the OTR, and after making the necessary configurations, the VPNs work in part.

From the router that initiates VPNs, I can join all the remote machines, but not from the PCs. If from a PC, I make a tracert to a remote site, I see that the OTR takes the route of the VPN output, instead of taking the one indicated in "Route static".
I tried to put the static route in question directly on Windows 10, and it works. I ping each remote PCs.

I feel that the roads are not well taken into account. Or I forgot something ..

Thank you.

## Specifications

  - OpenMPTCProuter version: ( v0.50)
  - OpenMPTCProuter VPS version: (0.995)
  - OpenMPTCProuter platform: (x86_64)

Ysurac, 2019-06-30:
I think custom routes are not working for now. You can still check using 'ip route' via ssh on openmptcprouter if they are applied. 
I Will try to fix this asap

Anthony010101, 2019-07-22:
Hello,

In the last update, is the static route problem resolved?

Thank you

Ysurac, 2019-07-22:
Should be. Need some tests.

closed 2019-12-20 by Ysurac