Ysurac/openmptcprouter #535: Dynamic IP (DDNS) support

xicciodj, 2019-06-30:
Hello!
I'm currently using a home server as a "VPS" for OpenMPTCP.
The problem is that the IP is dynamic, so it changes if the connection crashes or something.
Now, to solve that problem I tried assigning my no-ip host to the IP address of the VPN in the OpenMPTCPRouter settings, but that didn't work, it threw a lot of errors and it slowed down the RPI.
So, does or will OpenMPTCPRouter support Dynamic IP in the future?

  - OpenMPTCProuter version: 0.50.2
  - OpenMPTCProuter VPS version: 0.996
  - OpenMPTCProuter platform: RPI3

Ysurac, 2019-06-30:
For now vps part with dynamics ip is not supported . I need to make some code changes and tests if it's possible.

suyuan168, 2019-07-01:
nrgok  OR frp

xicciodj, 2019-07-01:
Ok, thanks!

closed 2019-07-21 by Ysurac