Ysurac/openmptcprouter #539: Status page is green when tracebox returns -TCPOptionMPTCPCapable

VoightKampff, 2019-07-06:
## Expected Behavior

The status page should indicate if an interface is flagged as -TCPOptionMPTCPCapable in the tracebox results.

## Actual Behavior
The status page is "green", but the non-mptcp-capable interface is not used for bandwidth aggregation. This is the case even if the interface is selected as the mptcp master. 

## Steps to Reproduce the Problem

  1. Put a middle box that blocks mptcp on one of the interfaces
  2. The status page is "green"
  3. 

## Specifications

  - OpenMPTCProuter version: 0.50.2
  - OpenMPTCProuter VPS version: 0.996
  - OpenMPTCProuter platform: x86_64

Ysurac, 2019-07-06:
Status page should put a warning if interface is detected as not MPTCP capable.
I can't be sure that it's really the case...

VoightKampff, 2019-07-06:
Here's the output of MPTCP Support Check (tracebox). I believe it's WAN2 that's the culprit. Do you need any additional information?

WAN1:
```
1: 192.168.5.1 
2: * 
3: * 
4: 10.220.105.30  [PARTIAL] +PartialTCP IP::DiffServicesCP IP::TotalLength IP::TTL IP::CheckSum 
5: 212.2.99.245 TCP::DataOffset TCP::CheckSum IP::TotalLength IP::TTL IP::CheckSum +TCPOptionMaxSegSize 
6: 212.2.102.209 TCP::DataOffset TCP::CheckSum IP::TotalLength IP::TTL IP::CheckSum +TCPOptionMaxSegSize 
7: 54.36.50.108 TCP::DataOffset TCP::CheckSum IP::TotalLength IP::TTL IP::CheckSum +TCPOptionMaxSegSize 
8: 213.186.32.203 TCP::DataOffset TCP::CheckSum IP::TotalLength IP::TTL IP::CheckSum +TCPOptionMaxSegSize 
9: * 
10: * 
11: * 
12: * 
13: * 
14: VPS_IP.VPS_IP.VPS_IP.VPS_IP TCP::SrcPort TCP::DstPort TCP::SeqNumber TCP::AckNumber TCP::DataOffset TCP::Flags TCP::WindowsSize TCP::CheckSum IP::TotalLength IP::Identification IP::TTL IP::CheckSum IP::SourceIP IP::DestinationIP +TCPOptionMaxSegSize TCPOptionMPTCPCapable::Sender's Key 

Result:
< Ethernet (14 bytes) :: DestinationMAC = 52:54:00:31:32:9a , SourceMAC = 34:ba:9a:60:f8:38 , Type = 0x800 , >
< IP (20 bytes) :: Version = 4 , HeaderLength = 5 , DiffServicesCP = 0 , ExpCongestionNot = 0 , TotalLength = 56 , Identification = 0x0 , Flags = 2 , FragmentOffset = 0 , TTL = 51 , Protocol = 0x6 , CheckSum = 0xc999 , SourceIP = VPS_IP.VPS_IP.VPS_IP.VPS_IP , DestinationIP = 192.168.5.5 , >
< TCP (20 bytes) :: SrcPort = 65101 , DstPort = 55519 , SeqNumber = 1560261519 , AckNumber = 2039645864 , DataOffset = 9 , Reserved = 0 , Flags = ( SYN ACK ) , WindowsSize = 28800 , CheckSum = 0xff05 , UrgPointer = 0 , >
< TCPOptionMaxSegSize (4 bytes) :: Kind = 2 , Length = 4 , MaxSegSize = 1400 , >
< TCPOptionMPTCPCapable (12 bytes) :: Kind = 30 , Length = 12 , Subtype = 0 , Version = 0 , Checksum = 1 (Checksum Enabled) , Flags = 0 , Crypto = 1 (HMAC-SHA1) , Sender's Key = Sender's Key = 12928355026262416321 , >
```

WAN2:
```
1: 192.168.8.1 
2: 172.18.130.201  [PARTIAL] +PartialTCP IP::TTL IP::CheckSum 
3: VPS_IP.VPS_IP.VPS_IP.VPS_IP TCP::SrcPort TCP::DstPort TCP::SeqNumber TCP::AckNumber TCP::DataOffset TCP::Flags TCP::WindowsSize TCP::CheckSum IP::TotalLength IP::Identification IP::TTL IP::CheckSum IP::SourceIP IP::DestinationIP +TCPOptionMaxSegSize -TCPOptionMPTCPCapable 

Result:
< Ethernet (14 bytes) :: DestinationMAC = 52:54:00:31:32:9a , SourceMAC = a4:ca:a0:63:d0:1f , Type = 0x800 , >
< IP (20 bytes) :: Version = 4 , HeaderLength = 5 , DiffServicesCP = 0 , ExpCongestionNot = 0 , TotalLength = 44 , Identification = 0xf244 , Flags = 2 , FragmentOffset = 0 , TTL = 252 , Protocol = 0x6 , CheckSum = 0xb00 , SourceIP = VPS_IP.VPS_IP.VPS_IP.VPS_IP , DestinationIP = 192.168.8.101 , >
< TCP (20 bytes) :: SrcPort = 65101 , DstPort = 29028 , SeqNumber = 563973445 , AckNumber = 368382868 , DataOffset = 6 , Reserved = 0 , Flags = ( SYN ACK ) , WindowsSize = 4380 , CheckSum = 0xbd55 , UrgPointer = 0 , >
< TCPOptionMaxSegSize (4 bytes) :: Kind = 2 , Length = 4 , MaxSegSize = 1460 , >
```

Ysurac, 2019-07-08:
WAN2 trace is really short and there is no public address.
What is the ISP ?

VoightKampff, 2019-07-08:
Yes it is, that's the full output - I only anonymized my vps ip, I didn't remove anything else.
The ISP is 37.47.0.0/16 - PL-TPSA-20120111 AS5617 Orange Polska Spolka Akcyjna

VoightKampff, 2019-07-20:
I found an additional, related bug in the status page - aggregation works despite the "Multipath seems to be blocked on the connection" message. Here is the output from tracebox (ran from the web UI).

```
1: 192.168.8.1 
2: 172.18.130.201  [PARTIAL] +PartialTCP IP::TTL IP::CheckSum 
3: 172.18.40.115  [PARTIAL] +PartialTCP IP::TTL IP::CheckSum 
4: 172.18.40.81  [PARTIAL] +PartialTCP IP::TTL IP::CheckSum 
5: 172.18.40.2  [PARTIAL] +PartialTCP IP::TTL IP::CheckSum 
6: 80.50.105.25  [PARTIAL] +PartialTCP IP::TTL IP::CheckSum 
7: * 
8: * 
```
...
(snip - lines are identical)
...
```
63: * 
64: * 

Result:
nil
```

I realize that tracebox giving up doesn't give a lot of options on how to handle this. If there is no way to obtain the information in a different way, I think a different message might be less confusing to users, such as "Unable to determine MPTCP support for this connection" or something along those lines. It's just a minor detail, but I think it would improve usability even if in a minuscule way.

Ysurac, 2019-07-20:
It's not a bug. It's why the page say "seems" I can't always detect when MPTCP work or not.
It's difficult to really make better tests because I don't have a connection with the problem for now.
I may change the message but in fact there is several tests, one using multipath-tcp.org that should give the real result, but seems not.

closed 2019-12-20 by Ysurac