Ysurac/openmptcprouter #540: Feature request: support for connections which are not TCPOptionMPTCPCapable

VoightKampff, 2019-07-06:
Would it be possible to provide MPTCP support for connections where the ISP blocks MPTCP by running additional layer 2/3 tunnels for each WAN interface to the VPS? This way, an ISP would not be able to mangle the TCP packets inside the individual tunnels. I think this is worth trying despite the additional overhead and possible packet fragmentation issues, because glorytun UDP is not a viable alternative - performance is very poor. For example, using the scenario with two wan interfaces - WAN1 and WAN2, two tunnel interfaces would need to be added on the router, TUN1 and TUN2. Then, the MPTCP settings on the router would use TUN1 and TUN2 instead of WAN1 and WAN2.

Ysurac, 2019-07-06:
True. But for now I don't know any ISP that really block MPTCP.

suyuan168, 2019-07-08:
There are currently no problems.

VoightKampff, 2019-07-08:
> There are currently no problems.

I posted my tracebox results in #539 - I am pretty sure one of my ISPs started blocking MPTCP.

PxxBxx, 2019-07-19:
Hello,
I live in France and I have a strange behaviour with one of my 3 ISPs.
Altice/SFR worked like a charm with OMR for the last year, but since 3 weeks (and 0.50.2, is there a link?) the wan2 (=SFR) stops being aggregated (MPTCP not supported in status, and -TCPOptionMPTCPCapable :-/

Sometime, when I reboot the router (SFR), the support comes back. But it disappear some hours later.

Do you know how I can monitor what is done on the MPTCP Support Check page ? do you have the command that gives the TCPOptions, that I could run on a regular basis (eg. crontab) and monitor precisely when it occurs ?

Of course I'd love an option to automaticaly aggregate WANs that does not support MPTCP, but in my case, I want to monitor/debug my specific WAN. It's totally anormal that the MPTCP support goes on and off randomly.....

Ysurac, 2019-07-20:
The information in status is only an info, MPTCP try to use the link in all case. Is it really not agregated ? In this case this should show a red cross. SFR always reported some strange informations. Test is done only sometimes by background script, so this can appear and disappear, but it's again only an info.

To test for TCPOptionMPTCPCapable there is several script that give needed options to tracebox : ```omr-tracebox wan1``` (give all details from wan1 to multipath-tcp.org), and ```omr-tracebox-mptcp wan1 vpsip``` (give only if MPTCP is up or down on wan1 to VPS)

closed 2019-12-20 by Ysurac