Ysurac/openmptcprouter #541: OMR BYPASS and iptv not working

nonameboy007, 2019-07-09:
## Specifications

  - OpenMPTCProuter version: 0.50.2
  - OpenMPTCProuter VPS version: 0.996 4.14.127-mtpcp
  - OpenMPTCProuter platform: RPI3

Since last update, my iptv seems not working anymore with mptcp ( adsl ok, 4G ok)

indeed, before this last update, i was using the OMR BYPASS to add my domain url (premium-ott.com with the interface wan1 or default interface)

But now, its not working, something has changed?

Thanks for your help.

Some command that may help you:

root@OpenMPTCProuter:~# uci show omr-bypass
omr-bypass.all=interface
omr-bypass.@domains[0]=domains
omr-bypass.@domains[0].name='premium-ott.com'
omr-bypass.@domains[0].interface='all'
omr-bypass.@ip[0]=ip
omr-bypass.@ip[0].interface='all'
omr-bypass.lo=interface
omr-bypass.lo.id='1'
omr-bypass.eth0=interface
omr-bypass.eth0.id='2'
omr-bypass.wan1=interface
omr-bypass.wan1.id='3'
omr-bypass.tun0=interface
omr-bypass.tun0.id='4'
omr-bypass.wan2=interface
omr-bypass.wan2.id='6'

root@OpenMPTCProuter:~# uci  show network
network.loopback=interface
network.loopback.ifname='lo'
network.loopback.proto='static'
network.loopback.ipaddr='127.0.0.1'
network.loopback.netmask='255.0.0.0'
network.loopback.multipath='off'
network.loopback.macaddr='00:00:00:00:00:00'
network.loopback.metric='1'
network.globals=globals
network.globals.ula_prefix='fd14:0e82:8ffe::/48'
network.globals.multipath='enable'
network.globals.mptcp_path_manager='fullmesh'
network.globals.mptcp_scheduler='default'
network.globals.mptcp_checksum='0'
network.globals.mptcp_fullmesh_num_subflows='1'
network.globals.mptcp_fullmesh_create_on_err='1'
network.globals.mptcp_ndiffports_num_subflows='1'
network.globals.mptcp_debug='0'
network.globals.congestion='bbr'
network.globals.mptcp_syn_retries='1'
network.lan=interface
network.lan.ifname='eth0'
network.lan.proto='static'
network.lan.ipaddr='192.168.100.1'
network.lan.netmask='255.255.255.0'
network.lan.ip6assign='60'
network.lan.multipath='off'
network.lan.ip4table='lan'
network.lan.macaddr='b8:27:eb:91:1b:bf'
network.lan.metric='2'
network.lan.ipv6='0'
network.wan1=interface
network.wan1.proto='static'
network.wan1.ip4table='wan'
network.wan1.defaultroute='0'
network.wan1.type='macvlan'
network.wan1.ifname='wan1'
network.wan1.masterintf='eth0'
network.wan1.ipaddr='192.168.1.10'
network.wan1.netmask='255.255.255.0'
network.wan1.gateway='192.168.1.1'
network.wan1.macaddr='26:da:23:10:d2:3e'
network.wan1.dns='192.168.1.1'
network.wan1.metric='3'
network.wan1.multipath='master'
network.wan1.peerdns='0'
network.wan1.ipv6='0'
network.wan1_dev=device
network.wan1_dev.name='wan1'
network.wan1_dev.type='macvlan'
network.wan1_dev.ifname='eth0'
network.wan1_dev.macaddr='26:da:23:10:d2:3e'
network.omrvpn=interface
network.omrvpn.ifname='tun0'
network.omrvpn.ip4table='vpn'
network.omrvpn.multipath='off'
network.omrvpn.leasetime='12h'
network.omrvpn.metric='4'
network.omrvpn.ipv6='0'
network.omrvpn.proto='none'
network.omr6in4=interface
network.omr6in4.proto='6in4'
network.omr6in4.ip4table='vpn'
network.omr6in4.multipath='off'
network.omr6in4.peeraddr='10.255.255.1'
network.omr6in4.gateway='fe80::a00:1'
network.omr6in4.ip6addr='fe80::a00:2'
network.omr6in4.ipaddr='10.255.255.2'
network.omr6in4.metric='5'
network.wwan_dev=device
network.wwan_dev.name='wwan'
network.wwan_dev.type='macvlan'
network.wwan_dev.ifname='wwan'
network.wan2=interface
network.wan2.proto='static'
network.wan2.type='macvlan'
network.wan2.ifname='wan2'
network.wan2.ipaddr='192.168.10.10'
network.wan2.netmask='255.255.255.0'
network.wan2.gateway='192.168.10.1'
network.wan2.defaultroute='0'
network.wan2.dns='8.8.8.8'
network.wan2.masterintf='eth0'
network.wan2.macaddr='f2:f6:34:02:9a:35'
network.wan2.metric='6'
network.wan2.multipath='on'
network.wan2.peerdns='0'
network.wan2.ipv6='0'
network.wan2_dev=device
network.wan2_dev.name='wan2'
network.wan2_dev.type='macvlan'
network.wan2_dev.ifname='eth0'
network.wan2_dev.macaddr='f2:f6:34:02:9a:35'
network.omr6in4_route6_default=route6
network.omr6in4_route6_default.interface='omr6in4'
network.omr6in4_route6_default.target='::'
network.omr6in4_route6_default.gateway='fe80::a00:1'

Thanks
Regards

Ysurac, 2019-07-09:
There is no change on OMR-ByPass.
You should remove DNS on all interfaces, and you must use OpenMPTCProuter as DNS on your computer. Domains are bypassed after a DNS request is made to OpenMPTCProuter.

nonameboy007, 2019-07-09:
i ll try tonight. i ll keep you informed 

regards

tx

nonameboy007, 2019-07-12:
Hi,

So, i have remove the DNS of all interface,

reboot, router and vps, but no luck... Iptv on ISP ==> OK, with 4G  ==> OK, with mptcp ==> DOWN.

maybe i do something wrong?i activate some dns trace and i see that on mptcprouter via logread command:

sometimes i got this: 

> Fri Jul 12 02:19:33 2019 daemon.info dnsmasq[25627]: 23 192.168.100.230/42084 query[A] premium-tv.com from 192.168.100.230
Fri Jul 12 02:19:33 2019 daemon.info dnsmasq[25627]: 23 192.168.100.230/42084 forwarded premium-tv.com to 127.0.0.1
Fri Jul 12 02:19:33 2019 daemon.info dnsmasq[25627]: 24 192.168.100.230/42084 query[AAAA] premium-tv.com from 192.168.100.230
Fri Jul 12 02:19:33 2019 daemon.info dnsmasq[25627]: 24 192.168.100.230/42084 forwarded premium-tv.com to 127.0.0.1
Fri Jul 12 02:19:33 2019 daemon.info dnsmasq[25627]: 23 192.168.100.230/42084 ipset add omr_dst_bypass_all 220.200.94.94 premium-tv.com
Fri Jul 12 02:19:33 2019 daemon.info dnsmasq[25627]: 23 192.168.100.230/42084 ipset add omr6_dst_bypass_all 220.200.94.94 premium-tv.com
Fri Jul 12 02:19:33 2019 daemon.info dnsmasq[25627]: 23 192.168.100.230/42084 reply premium-tv.com is 220.200.94.94

sometimes i got that:

> Fri Jul 12 02:26:17 2019 daemon.info dnsmasq[25627]: 72 192.168.100.230/49831 query[A] premium-ott.com from 192.168.100.230
Fri Jul 12 02:26:17 2019 daemon.info dnsmasq[25627]: 72 192.168.100.230/49831 cached premium-ott.com is 220.200.94.94
Fri Jul 12 02:26:17 2019 daemon.info dnsmasq[25627]: 73 192.168.100.230/49831 query[AAAA] premium-ott.com from 192.168.100.230
Fri Jul 12 02:26:17 2019 daemon.info dnsmasq[25627]: 73 192.168.100.230/49831 forwarded premium-ott.com to 127.0.0.1

when i made a netstat i got

> Server:		127.0.0.1
Address:	127.0.0.1#53
Name:      google.fr
Address 1: 172.217.22.131
Address 2: 2a00:1450:4007:815::2003

> root@OpenMPTCProuter:~# uci show omr-bypass
omr-bypass.all=interface
omr-bypass.@domains[0]=domains
omr-bypass.@domains[0].name='premium-tv.com'
omr-bypass.@domains[0].interface='wan1'
omr-bypass.lo=interface
omr-bypass.lo.id='1'
omr-bypass.eth0=interface
omr-bypass.eth0.id='2'
omr-bypass.wan1=interface
omr-bypass.wan1.id='3'
omr-bypass.wan2=interface
omr-bypass.wan2.id='4'
omr-bypass.tun0=interface
omr-bypass.tun0.id='5'

> root@OpenMPTCProuter:~# uci show network
network.loopback=interface
network.loopback.ifname='lo'
network.loopback.proto='static'
network.loopback.ipaddr='127.0.0.1'
network.loopback.netmask='255.0.0.0'
network.loopback.multipath='off'
network.loopback.macaddr='00:00:00:00:00:00'
network.loopback.metric='1'
network.globals=globals
network.globals.ula_prefix='fd33:1edd:402a::/48'
network.globals.multipath='enable'
network.globals.mptcp_path_manager='fullmesh'
network.globals.mptcp_scheduler='default'
network.globals.congestion='bbr'
network.globals.mptcp_checksum='0'
network.globals.mptcp_debug='0'
network.globals.mptcp_syn_retries='1'
network.globals.mptcp_fullmesh_num_subflows='1'
network.globals.mptcp_fullmesh_create_on_err='1'
network.globals.mptcp_ndiffports_num_subflows='1'
network.lan=interface
network.lan.ifname='eth0'
network.lan.proto='static'
network.lan.ipaddr='192.168.100.1'
network.lan.netmask='255.255.255.0'
network.lan.ip6assign='60'
network.lan.delegate='0'
network.lan.ip4table='lan'
network.lan.macaddr='b8:27:xx:xx:xx:bf'
network.lan.metric='2'
network.lan.ipv6='0'
network.lan.multipath='off'
network.lan_rule=rule
network.lan_rule.lookup='lan'
network.lan_rule.priority='100'
network.wan1=interface
network.wan1.proto='static'
network.wan1.ip4table='wan'
network.wan1.defaultroute='0'
network.wan1.type='macvlan'
network.wan1.ifname='wan1'
network.wan1.masterintf='eth0'
network.wan1.metric='3'
network.wan1.peerdns='0'
network.wan1.ipv6='0'
network.wan1.ipaddr='192.168.1.10'
network.wan1.netmask='255.255.255.0'
network.wan1.gateway='192.168.1.1'
network.wan1.macaddr='e6:f1:xx:xx:xx:f2'
network.wan1.multipath='master'
network.wan1.delegate='0'
network.wan2=interface
network.wan2.proto='static'
network.wan2.ip4table='wan'
network.wan2.defaultroute='0'
network.wan2.type='macvlan'
network.wan2.ifname='wan2'
network.wan2.masterintf='eth0'
network.wan2.metric='4'
network.wan2.peerdns='0'
network.wan2.ipv6='0'
network.wan2.ipaddr='192.168.10.10'
network.wan2.netmask='255.255.255.0'
network.wan2.gateway='192.168.10.1'
network.wan2.macaddr='f6:70:xx:xx:xxca'
network.wan2.multipath='on'
network.wan2.delegate='0'
network.wan2.force_link='0'
network.wan1_dev=device
network.wan1_dev.name='wan1'
network.wan1_dev.type='macvlan'
network.wan1_dev.ifname='eth0'
network.wan1_dev.macaddr='e6:f1:xx:xx:xx:f2'
network.wan2_dev=device
network.wan2_dev.name='wan2'
network.wan2_dev.type='macvlan'
network.wan2_dev.ifname='eth0'
network.wan2_dev.macaddr='f6:70:0x:xx:0d:ca'
network.omrvpn=interface
network.omrvpn.ifname='tun0'
network.omrvpn.ip4table='vpn'
network.omrvpn.leasetime='12h'
network.omrvpn.txqueuelen='1000'
network.omrvpn.metric='5'
network.omrvpn.ipv6='0'
network.omrvpn.proto='none'
network.omrvpn.multipath='off'
network.omrvpn.delegate='0'
network.omr6in4=interface
network.omr6in4.proto='6in4'
network.omr6in4.ip4table='vpn'
network.omr6in4.ipaddr='10.255.255.2'
network.omr6in4.peeraddr='10.255.255.1'
network.omr6in4.gateway='fe80::a00:1'
network.omr6in4.ip6addr='fe80::a00:2'
network.omr6in4.metric='6'
network.omr6in4.multipath='off'

Thanks for help.

Ysurac, 2019-07-12:
Should be working with _premium-tv.com_ but you didn't add _premium-ott.com_ in omr-bypass.
And from where is the netstat ?

nonameboy007, 2019-07-12:
hi,
for the domain name, it's an error on my part, I changed the names to hide all that and I forgot to change when copying / pasting

the netstat were from the openmptcprouter, from the vu+ uno 4k se, the netstat show the mptcprouter Ip address.

nonameboy007, 2019-07-14:
Yannick,

does its normal that on the diagnostic utility, the nslookup to openmptcprouter.com return

> Server:		127.0.0.1
Address:	127.0.0.1#53
Name:      openmptcprouter.com
Address 1: 94.23.252.192
*** Can't find openmptcprouter.com: No answer

Thanks for all

Ysurac, 2019-07-15:
Yes. I didn't set IPv6 entry for openmptcprouter.com

nonameboy007, 2019-07-16:
so,

i ve reinstall  everything:
- VPS (debian 9 OpenMPTCProuter VPS 0.996) from OVH
- Raspberry pi 3 32bit (openmptcprouter v0.50.2 r0+10127-3209f5ae3d / LuCI Master (f138fc93)) with only settings wizard and OMR bypass modified

This is what i get from the system log after enabled Log queries in DNS & DHCP Server Settings/General Settings:

> Tue Jul 16 19:28:51 2019 daemon.info dnsmasq[1359]: 74 192.168.100.230/51862 query[A] vuuno4kse from 192.168.100.230
Tue Jul 16 19:28:51 2019 daemon.info dnsmasq[1359]: 74 192.168.100.230/51862 config vuuno4kse is NODATA-IPv4
Tue Jul 16 19:28:51 2019 daemon.info dnsmasq[1359]: 75 192.168.100.230/51862 query[AAAA] vuuno4kse from 192.168.100.230
Tue Jul 16 19:28:51 2019 daemon.info dnsmasq[1359]: 75 192.168.100.230/51862 config vuuno4kse is NODATA-IPv6

> Tue Jul 16 19:37:43 2019 daemon.info dnsmasq[1359]: 273 192.168.100.230/37144 query[A] premium-tv.com from 192.168.100.230
Tue Jul 16 19:37:43 2019 daemon.info dnsmasq[1359]: 273 192.168.100.230/37144 cached premium-tv.com is 193.XX.XX.XX
Tue Jul 16 19:37:43 2019 daemon.info dnsmasq[1359]: 274 192.168.100.230/37144 query[AAAA] premium-tv.com from 192.168.100.230
Tue Jul 16 19:37:43 2019 daemon.info dnsmasq[1359]: 274 192.168.100.230/37144 forwarded premium-tv.com to 127.0.0

No video at all when using OMR... (but from ISP (free) or 4G no problem at all), 1 month ago, it was working..

i not able to see whats going wrong.

Some help will be appreciated.

Regards

Ysurac, 2019-07-16:
You can try beta release on https://www.openmptcprouter.com/release/4.19/

nonameboy007, 2019-07-16:
not better...

on the vps i got martian source 10.255.254.1 from 10.255.254.2, on dev gt-tun0

and since no video thrue OMR

nonameboy007, 2019-07-17:
Hi,

today i ve saw that and vps syslog log:

> Jul 17 12:45:22 vps641946 systemd[1]: Started Shadowsocks-libev Default Server Service.
Jul 17 12:45:22 vps641946 ss-server[32169]:  2019-07-17 12:45:22 INFO: resolving hostname to IPv6 address first
Jul 17 12:45:22 vps641946 ss-server[32169]:  2019-07-17 12:45:22 INFO: using tcp fast open
Jul 17 12:45:22 vps641946 ss-server[32169]:  2019-07-17 12:45:22 INFO: UDP relay enabled
Jul 17 12:45:22 vps641946 ss-server[32169]:  2019-07-17 12:45:22 INFO: enable TCP no-delay
Jul 17 12:45:22 vps641946 ss-server[32169]:  2019-07-17 12:45:22 INFO: initializing ciphers... chacha20
Jul 17 12:45:22 vps641946 ss-server[32169]:  2019-07-17 12:45:22 ERROR: Invalid key for your chosen cipher!
Jul 17 12:45:22 vps641946 ss-server[32169]:  2019-07-17 12:45:22 ERROR: It requires a 32-byte key encoded with URL-safe Base64
Jul 17 12:45:22 vps641946 ss-server[32169]:  2019-07-17 12:45:22 ERROR: Generating a new random key: MGDva4k3tRAdiQeZB13QdS9HcrQXpH0HKknf_YS77Xg=
Jul 17 12:45:22 vps641946 ss-server[32169]:  2019-07-17 12:45:22 ERROR: Please use the key above or input a valid key
Jul 17 12:45:22 vps641946 systemd[1]: shadowsocks-libev.service: Main process exited, code=exited, status=255/n/a

so i take a look to the configuration of this file /etc/shadowsocks-libev/config.json

I saw that one character of the password was been modified ? / change to _ ? i dont know if it was normal, and its not my modification?

so i try the new password with this command: 

ss-server -c /etc/shadowsocks-libev/config.json

All good,

 but now i got this ERROR: remote recv: Connection reset by peer  on vps syslog log when i change a channel, but still no image or sound...

and 

> Tue Jul 16 19:28:51 2019 daemon.info dnsmasq[1359]: 74 192.168.100.230/51862 query[A] vuuno4kse from 192.168.100.230
> Tue Jul 16 19:28:51 2019 daemon.info dnsmasq[1359]: 74 192.168.100.230/51862 [...]

on OMR Syslog

i will try tonight with this image https://www.openmptcprouter.com/release/4.19/rpi3/targets/brcm2708/bcm2710/openmptcprouter-v0.50.2-83-g3629820-r0%2B10518-6d61fcfa9e-brcm2708-bcm2710-rpi-3-ext4-factory.img.gz

and i will clean the vps too. Maybe i do something wrong?

We are OK that i only have to configure the settings wizard, and add my domain premium-tv.com to OMR Bypass?  nothing more todo with shadowsocks or recursive DNS or DHCP and DNS? 

Thanks

Ysurac, 2019-07-17:
You can try, from a client computer, a traceroute: if this go out with the defined wan than it's working.

nonameboy007, 2019-07-17:
the traceroute seems working from a computer client, but there is no translation from ip to name: for exemple:

From 4g connection:

>  traceroute nis.nsf.net.
> traceroute to nis.nsf.net. (192.64.147.192), 30 hops max, 60 byte packets
>  1  OpenMPTCProuter.lan (192.168.100.1)  3.701 ms  2.842 ms  4.385 ms
>  2  10.255.255.1 (10.255.255.1)  81.707 ms  80.934 ms  82.394 ms
>  3  51.75.133.1 (51.75.133.1)  85.806 ms  92.459 ms  86.571 ms
>  4  192.168.143.254 (192.168.143.254)  91.635 ms  94.758 ms  86.491 ms
>  5  10.69.87.254 (10.69.87.254)  98.440 ms  99.197 ms  102.036 ms
>  6  10.69.86.8 (10.69.86.8)  105.479 ms  105.343 ms 10.69.86.10 (10.69.86.10)  140.427 ms
>  7  10.69.64.22 (10.69.64.22)  139.544 ms 10.69.64.18 (10.69.64.18)  114.918 ms 10.69.64.16 (10.69.64.16)  107.775 ms
>  8  10.17.193.66 (10.17.193.66)  84.204 ms  82.203 ms 10.17.193.60 (10.17.193.60)  81.398 ms
>  9  10.73.8.56 (10.73.8.56)  79.759 ms 10.73.8.58 (10.73.8.58)  80.977 ms 10.73.8.60 (10.73.8.60)  81.695 ms
> 10  10.95.48.8 (10.95.48.8)  148.515 ms 10.95.48.10 (10.95.48.10)  82.451 ms 10.95.48.8 (10.95.48.8)  85.032 ms
> 11  be100-1258.gsw-1-a9.fr.eu (91.121.215.219)  91.258 ms  89.534 ms be100-1254.th2-1-a9.fr.eu (94.23.122.139)  92.028 ms
> 12  be100-1345.ash-1-a9.va.us (94.23.122.244)  169.659 ms  174.192 ms be100-1346.ash-5-a9.va.us (37.187.36.199)  173.048 ms
> 13  mia-mi1-bb1-a9.fl.us (142.44.208.188)  199.492 ms be100-1290.mia-mi1-bb1-a9.fl.us (192.99.146.112)  194.537 ms mia-mi1-bb1-a9.fl.us (142.44.208.188)  190.556 ms
> 14  * * *
> 15  * * *
> 16  * * *
> 17  96-31-83-106.static.hvvc.us (96.31.83.106)  207.153 ms  220.261 ms  219.430 ms
> 18  * * *
> 19  * * *
> 20  * * *
> 21  * * *
> 22  * * *
> 23  * * *
> 24  * * *
> 25  * * *
> 26  * * *
> 27  * * *
> 28  * * *
> 29  * * *
> 30  * * *

From OMR connection:

> traceroute nis.nsf.net.
> traceroute to nis.nsf.net. (192.64.147.192), 30 hops max, 60 byte packets
>  1  _gateway (192.168.10.1)  0.859 ms  1.229 ms  1.479 ms
>  2  192.168.225.1 (192.168.225.1)  2.180 ms  3.086 ms  4.229 ms
>  3  192.168.254.30 (192.168.254.30)  31.278 ms  31.579 ms  35.217 ms
>  4  192.168.255.10 (192.168.255.10)  37.503 ms  37.922 ms  38.220 ms
>  5  ppp22-metz.isdnet.net (194.149.166.149)  35.619 ms  35.970 ms  36.249 ms
>  6  ppp10-metz.isdnet.net (194.149.166.137)  36.633 ms *  90.582 ms
>  7  194.149.166.54 (194.149.166.54)  90.809 ms  91.313 ms  91.527 ms
>  8  yankee-6k-1.po2.intf.routers.proxad.net (212.27.58.26)  109.637 ms  113.574 ms  113.872 ms
>  9  * newyork-6k-1-po2.intf.routers.proxad.net (212.27.58.209)  117.577 ms *
> 10  * * paloalto-6k-1-po1.intf.routers.proxad.net (212.27.58.222)  189.587 ms
> 11  * * *
> 12  10ge7-5.core1.sjc2.he.net (72.52.92.70)  214.285 ms 100ge9-2.core1.sjc2.he.net (72.52.92.157)  192.465 ms 10ge7-5.core1.sjc2.he.net (72.52.92.70)  184.337 ms
> 13  100ge2-2.core1.lax1.he.net (72.52.92.121)  184.578 ms 100ge15-1.core1.lax1.he.net (184.104.193.50)  189.275 ms  188.853 ms
> 14  hivelocity-ventures-corp.10gigabitethernet12-19.core1.lax1.he.net (64.71.134.142)  189.244 ms  189.226 ms  189.205 ms
> 15  10.254.0.32 (10.254.0.32)  190.217 ms  192.311 ms  191.050 ms
> 16  10.254.0.10 (10.254.0.10)  211.029 ms *  207.963 ms
> 17  96-31-83-106.static.hvvc.us (96.31.83.106)  198.924 ms  198.852 ms  202.823 ms
> 18  * * *
> 19  * * *
> 20  * * *
> 21  * * *
> 22  * * *
> 23  * * *
> 24  * * *
> 25  * * *
> 26  * * *
> 27  * * *
> 28  * * *
> 29  * * *
> 30  * * *

Ysurac, 2019-07-17:
first traceroute use the VPN and should bypass, second traceroute bypass all and is correct.

nonameboy007, 2019-07-17:
the first traceroute were witout MPTCP rasp, and the second were with mptcp.

nonameboy007, 2019-07-17:
Now, throught OMR, with VLC, when i try to open the network stream, i got 

if OMR-bypass not set with premium-tv.com
[00007f7ea801a470] mjpeg demux error: cannot peek

if OMR-bypass  set with premium-tv.com for wan1
[00007f7ea8002f60] access stream error: HTTP 401 error
[00007f7ea8002f60] http stream error: authentication failed without realm

nonameboy007, 2019-07-19:
i Yannick,

I found the problem...  finally.

if i put premium-tv.com on omr bypass, its not working, but, i see trought netstat command on my vuuno, some ip that cannot be established, so i add the ip/24 on omr bypass ip and all good !!

what a work, thanks again ;-)

closed 2019-07-19 by nonameboy007