Ysurac/openmptcprouter #546: Kernel CONFIG_MEMCG: missing

dur3x, 2019-07-13:
Can you please apply the fix of this issue 
https://github.com/multipath-tcp/mptcp/issues/321

Currently with your kernel it's impossible to use memory cgroup. 
It's just a flag enabled normally as default. 
It's already fixed in the official MPTCP project but not yet here.

Thanks !

Ysurac, 2019-07-13:
I think you mean on the VPS ?
I use default Debian config and I don't need cgroup on VPS part, but why not...

dur3x, 2019-07-13:
Exactly it's for the vps part.
Thanks in advance :-)

dur3x, 2019-09-04:
@Ysurac Did you get time to add the cgroup?
It will be great if I can directly use your kernel image instead of the one available on https://dl.bintray.com/cpaasch/deb

Ysurac, 2019-09-04:
This will be available in next release with an updated kernel. Should be out soon (I hope)

dur3x, 2019-11-15:
> This will be available in next release with an updated kernel. Should be out soon (I hope)

Just a gentle reminder :-)

closed 2019-12-20 by Ysurac