Ysurac/openmptcprouter #547: Best way to replace ShadowSock (MPTCP not supported...)

PxxBxx, 2019-07-19:
Since some weeks, one of my ISPs disabled MPTCP support :'(
So I have to find a new way to aggregate lines... I read in the wiki that one must desactivate Shadowsock in order for glorytun to take all trafic, but I can't manage..

## Expected Behavior

To be able to disable Shadowsock and fallback to default VPN (GloryTUN with aggregation)

## Actual Behavior

* disable Shadowsock -> OK, it disables, but the checkbox stays unchecked, and at the next _save and apply_ it will be re-activated

* after disabling SS, I can only Ping, but I can't do anything else (SSH, HTTP, ...)

I'm sorry, but it's a long time since Ive digged into the logs, you'll have to tell me what to look.

## Steps to Reproduce the Problem

* Go to System > OMR > Server > Advanced
* disable Shadowsocks
* let GloryTun TCP as default VPN

## Specifications

  - OpenMPTCProuter version: Last ^^ (0.50.2)
  - OpenMPTCProuter VPS version: Last ^^ (0.99.6)
  - OpenMPTCProuter platform: x64_64 ext4

tranquiliza, 2019-07-21:
Im having this issue as well. But also having constant issues with the software refusing to save settings.

Ysurac, 2019-07-21:
Fixed in latest release (at least the shadowsocks disabled checkbox).

closed 2019-07-21 by Ysurac