Ysurac/openmptcprouter #548: YouTube appears to be capped at 1Mbps?

jwthecolorist, 2019-07-21:
This may simply be my server (I'm using Vultr) but I've noticed on the latest releases (0.50.0 onward) that YouTube appears to be capped at 1Mbps. Is this a feature or setting builtin that I cannot seem to locate? Just wondering before I try changing hosting etc.

## Specifications

  - OpenMPTCProuter version: v0.51 r0+10518-6d61fcfa9e
  - OpenMPTCProuter VPS version: 0.996 4.14.127-mptcp
  - OpenMPTCProuter platform: x86_64

Ysurac, 2019-07-21:
No support on not yet released version (and VPS is not updated).
And no problem on Vultr for me.

closed 2019-07-21 by Ysurac

jwthecolorist, 2019-07-21:
Thanks Ysurac. Sorry to open an issue about this, I had exhausted everything I could try without changing hosting. I thought perhaps it was something I was missing somewhere.