Ysurac/openmptcprouter #549: 4G modem same MAC addresses

Bestle, 2019-07-21:
When connecting multiple Huawei USB 4G dongles, they all have the same MAC address. OMR works correctly with one connected, but freaks out when two are connected.

https://i.imgur.com/oIctX1u.jpg

## Specifications

  - OpenMPTCProuter version:  openmptcprouter-v0.51
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter platform: x86_64

Ysurac, 2019-07-22:
You can try to change the mac address using the network->interfaces and advanced parameters on the wan tab.

Bestle, 2019-07-22:
That tab is not available when using NCM protocol.

Ysurac, 2019-07-22:
You can try to change it via SSH using ```ip link set wan2 address 02:01:02:03:04:08```

Bestle, 2019-07-22:
How would I make this persist across reboots and not cause issues with OMR?

Ysurac, 2019-07-22:
This work or not ?

Bestle, 2019-07-22:
Yes the change of MAC works but it goes back to original duplicate MAC when router is restarted.

Ysurac, 2019-07-22:
If you try :
```
uci set network.wan2.macaddr="02:01:02:03:04:08"
uci commit network
```
This work after reboot ?

Bestle, 2019-07-22:
ifconfig wwan0 shows the original MAC after reboot, but the option is still set in network config

Bestle, 2019-07-22:
OMR cannot find the device now either

Ysurac, 2019-07-22:
I can't really help more about that, I don't have any NCM keys. You can check if you find something about that on OpenWrt doc/forum, because that part is not modified at all from OpenWrt version.

rstanislav, 2019-07-22:
NCM is bad idea@linux.
For mac address part - you can change it using AT commands i think, what model of modems are you using ?

Bestle, 2019-07-22:
The modems are Huawei E3372-153 LTE/4G non-hilink. Is there a protocol I should be using instead?

tranquiliza, 2019-07-22:
I'm having issues, that using the Override MAC in the Advanced tab, is not overriding the MAC (Also Duplicate Mac, from 2 Huawei Modems)

[image]

[image]

rstanislav, 2019-07-22:
Well, i used over 20 of them (3372h) and never saw duplicate mac.. It can be ncm driver problem i think.. Try to flash them with hilink firmware, but it is danger process if you don't know how to do it, if you do something wrong its hard to recover modem.
There is detailed description how to do it with links to all needed tools and firmwares(http://4pda.ru/forum/index.php?showtopic=582284&st=750), but its in russian.. maybe somewhere else there is guide in english..

tranquiliza, 2019-07-22:
Unfortunately I am not the owner of one of the Modems (LTE Provider's property) So I cannot do this for that one. Besides, I have no clue what this is :(

Bestle, 2019-07-22:
I can try flashing mine with hi link firmware. Which protocol should I use after that? Will I not have double NAT issues?

Bestle, 2019-07-22:
If you search your MAC on google @tranquiliza you will see lots of people have devices with the same MAC address... It seems to be a common problem.

tranquiliza, 2019-07-22:
That's wild. So only fix is buy another Modem? (which is not a Huawei? )

Bestle, 2019-07-22:
[image]

I have 4 more modems on the way, I will check the MACs and let you know.

karjaj, 2019-08-07:
Is it MAC addr 0c:5b:8f:27:9a:64 ? I have E5377T and Vodafone stick wich is rebranded E3372. Both have same MAC, but they working nice with openmptcprouter. They recognized without problem as eth1 and eth2 as cdc-ethernet devices.

closed 2019-12-20 by Ysurac