Ysurac/openmptcprouter #551: "Use default gateway" disables on reboot

Bestle, 2019-07-22:
## Expected Behavior

For the "use default gateway" checkbox setting to persist across reboots.

## Actual Behavior

After a reboot, "use default gateway" is unchecked for WAN connection. I have installed OMR to an SSD that has a ton of free space and no drive errors.

## Steps to Reproduce the Problem

  1. Set up an NCM protocol WAN connection
  2. Tick "use default gateway" and save
  3. Verify setting by reloading and checking it is still ticked
  4. Reboot router
  5. Check WAN settings and "use default gateway" is now unchecked

## Specifications

  - OpenMPTCProuter version: openmptcprouter-v0.51-r010518-6d61fcfa9e-x86-64-combined
  - OpenMPTCProuter VPS version: N/A
  - OpenMPTCProuter platform: x86_64

Bestle, 2019-07-22:
This can also be reproduced by manually changing the network config.

Set option defaultroute to '1', reboot, value is now '0' and wan connection is not active.

Bestle, 2019-07-22:
This gets a little weirder... After a reboot of OMR there is no connection to the internet at all. But if I open WAN1 and click "save & apply" _without changing any settings at all_ the internet is available after a few seconds.

System log (I clicked save & apply at 3:40:31, there was no internet until I clicked it): https://pastebin.com/68RjkV2r

Bestle, 2019-07-22:
More findings... it doesn't seem to happen when using shadowsocks, only glorytun.

Ysurac, 2019-07-22:
True : "use default gateway" is removed when detected because this can't work well with OpenMPTCProuter. If you need it then you have a configuration problem.
Default route is the VPN when available, else it's working connection with best RTT.

Bestle, 2019-07-22:
Please check my updated replies. I don't think it has anything to do with default gateway now, as just clicking on "save and apply" fixes the WAN connection _without_ enabling the default gateway option.

Ysurac, 2019-07-22:
As I said, I think you have a configuration problem.
You can also check status page.

Bestle, 2019-07-22:
Let me try and explain this issue 100% clearly because it is not a configuration problem...

When using glorytun (this does not happen with just shadowsocks), the connection does not work after a router reboot (status page shows everythingok). To fix this I go into WAN connection, click "save and apply" **without changing any options at all** and the internet now works. How can it be a configuration issue if it fixes without me changing anything?

See this system log: https://pastebin.com/68RjkV2r
No internet available until Mon Jul 22 03:40:31 when I clicked "save and apply" in WAN without changing any options.

closed 2019-12-20 by Ysurac