Ysurac/openmptcprouter #553: OpenMPTCPRouter v0.51-r010518 doesn't restore or save settings.

xicciodj, 2019-07-22:
Hi!
Today I wanted to update my OpenMPTCPRouter on my Raspberry Pi 3B to the latest version (v0.51-r010518), so I backed up my settings from the web interface, then I flashed the factory image to the SD Card. After that, I put my SD into my Raspberry Pi 3 and it booted. At this point, I went to the web interface and tried to restore settings, it rebooted as expected but then all the settings were gone. I thought that maybe the backup was corrupted in some way, so I reflashed again the image on the SD just to be sure and I tried to set up manually everything. But there were problems:
First of all, it says that "Multipath seems to be blocked on the connection", which is an error I've never seen before. And then, what I think is the most important error, is "Filesystem is readonly". After I've read that, I tried to reboot the Raspberry Pi and all the settings were gone again. So it just doesn't save settings.
I also tried another SD Card but it didn't work either.
Here's a screenshot of the error (there is only one wan because I didn't have the second one "handy" at that moment):
[image]

So, how can I fix this?
## Specifications

  - OpenMPTCProuter version: v0.51-r010518
  - OpenMPTCProuter VPS version: 0.9962
  - OpenMPTCProuter platform: RPI3 (64bit image)

Ysurac, 2019-07-22:
Squashfs or ext4 ?
Using ```dmesg``` via SSH you can find why there is a FS problem.

xicciodj, 2019-07-22:
Squashfs
I'm seeing this in dmesg, I don't know if this could be the cause:
VFS: Mounted root (squashfs filesystem) readonly on device 179:2.

Ysurac, 2019-07-22:
squashfs is always readonly, there is a f2fs overlay that must be read/write. Check about F2FS log.

xicciodj, 2019-07-22:
The only error I'm seeing about F2FS (in dmesg) is this:
`[    6.958099] mount_root: failed to mount -t f2fs /dev/loop0 /tmp/overlay: No such device
[    6.969346] mount_root: overlay filesystem has not been fully initialized yet
[    6.978302] mount_root: switching to f2fs overlay
[    6.984687] mount_root: switching to f2fs failed - fallback to ramoverlay`

Ysurac, 2019-07-22:
Nothing more before that ?
You can try to run via ssh ```mount_root done```

xicciodj, 2019-07-22:
I can't see anything related to f2fs before that, but maybe I'm searching for the wrong thing. If needed, I can copy all the dmesg output.
I'm not sure what mount_root done should do, but I tried it. The command gave no output, and there was no new messages in dmesg. I also tried to restore configurations again, but it didn't work.

Ysurac, 2019-07-22:
You should try to reboot (via reboot command) after that, and check if you see the same message about overlay filesystem not initialized.

xicciodj, 2019-07-22:
Ok, tried it. Same exact error.

Ysurac, 2019-07-22:
Use ext4 version ;)

xicciodj, 2019-07-22:
Ok, the ext4 version is restoring and saving settings correctly!

closed 2019-07-22 by xicciodj

mycanaletto, 2019-07-23:
Bonjour, 
Idem sous ESX
openmptcprouter-v0.51-r010518-6d61fcfa9e-x86-64-combined-squashfs.img.gz

La mise à jour provoque un clean de la config et quand on refait la config elle ne résiste pas à un reboot...

Si la solution est de refaire la VM en EXT4 autant virer les images HFS ?

Un conseil sur quoi utiliser sur ESX ? Merci :-)

Ysurac, 2019-07-23:
La mise à jour efface la config du fait du changement du partitionnement et de la taille de l'image.

ext4 est actuellement plus stable que squashfs.

mycanaletto, 2019-07-23:
J'ai remonté une VM avec :
openmptcprouter-v0.51-r010518-6d61fcfa9e-x86-64-combined-ext4.vmdk
Et c'est OK

Impossible de le faire en EFI par contre, j'ai peut être loupé qq chose... Le mieux peut être serait à terme de proposer un OVF ?

Ysurac, 2019-07-23:
C'est prévu, il faut que je prenne les 5 minutes pour le faire :)