Ysurac/openmptcprouter #554: Version compatibility

JorgeMoort89, 2019-07-22:
Hi, can I use the VPS at the latest version (0.997) with the router on 0.49?
I've seen a kernel and MPTCP update, can that cause some problems?
I've seen quite often some "Can't get public IP address from ShadowSocks" on the status paga, can that be a problem? It disappear at the next refresh and come back in some minutes, this situation repeat continously.

## Specifications

  - OpenMPTCProuter version: 0.49
  - OpenMPTCProuter VPS version: 0.997
  - OpenMPTCProuter platform: x86_64

Ysurac, 2019-07-22:
Better when all is updated a the same time.

For the can't get public IP... you should check the system log about shadowsocks problems. If there is no problems then it's only that this take too much time to reach the website to get public IP.

JorgeMoort89, 2019-07-22:
Where can I check the SS logs?
I haven't the physical access to the router in this moment and I don't wan't to update it remotely.

Ysurac, 2019-07-22:
In Status->System log

JorgeMoort89, 2019-07-22:
Only a lot of this `daemon.err /usr/bin/ss-redir[9408]: send: Connection reset by peer` and the normale DHCP renew

closed 2019-07-24 by JorgeMoort89