Ysurac/openmptcprouter #555: Problème de débit et Shadowsocks

Randdrick, 2019-07-22:
## Expected Behavior

Mon débit agrégé (WAN1 + WAN2) devrait être supérieur à mon débit sur mon WAN2 (4G)

## Actual Behavior
Mon débit ( lorsque mon pc est configuré OMRBYPASS ) sur ma connexion 4G est deux fois plus importante que mon débit agrégé (Wan1 : ADLS + WAN2 : 4G)

## Steps to Reproduce the Problem

  1. Lancer un test un Nperf (hors OMRBYPASS) : Débit moyen constaté : 30 Mb/s
  2. Configurer son ordinateur en OMRBYPASS sur son WAN2 (4G) ou son interface WAN la plus rapide
  3. Lancer un test Nperf : Débit moyen constaté : 60 Mb/s

## Specifications

  - OpenMPTCProuter version: 0.51
  - OpenMPTCProuter VPS version: 0.997
  - OpenMPTCProuter platform: RPI3

Il est a noté aussi que je rencontre un problème avec les Shadowsocks. Message d'erreur très fréquent : "can't get Ip public adresse". C'est la raison pour laquelle je me pose la question si mon problème de débit n'est pas lié avec ce problème Shadowsocks.

En version 49.5 et 0.994 pour le VPS, mon débit moyen était de 60 mb/s (hors OMRBYPASS)

Ysurac, 2019-07-22:
Le VPS a été redémarrer après la mise à jour ?
Si c'est le cas il faut regarder dans les logs systèmes.

Randdrick, 2019-07-22:
Oui, j'ai bien redémarré le VPS.
Dans les logs système du client, je n'ai rien de particulier. Dois-je regarder dans les logs du VPS ?

Ysurac, 2019-07-22:
Etonnant qu'aucune erreur lié à shadowsocks n'apparaisse dans les logs.
Tu peux regarder ceux du VPS aussi oui, mais il ne doit pas y avoir grand chose.

Randdrick, 2019-07-22:
La seule erreur que j'ai eu est celle là :
Mon Jul 22 22:25:16 2019 daemon.err /usr/bin/ss-redir[30519]: send: Broken pipe
Mon Jul 22 22:25:16 2019 daemon.err /usr/bin/ss-redir[30521]: send: Broken pipe
Mon Jul 22 22:25:16 2019 daemon.err /usr/bin/ss-redir[30522]: send: Broken pipe
Mon Jul 22 22:25:16 2019 daemon.err /usr/bin/ss-redir[30521]: send: Broken pipe
Mon Jul 22 22:25:16 2019 daemon.err /usr/bin/ss-redir[30522]: send: Broken pipe
Mon Jul 22 22:25:16 2019 daemon.err /usr/bin/ss-redir[30522]: send: Broken pipe

Mais le problème s’est reproduit plusieurs fois déjà, sans message d'erreur.
J'ai bien eu un problème lié au mot de passe, que j'ai résolu car la case "Enable obfuscation" s'était décochée après la mise à jour du routeur.

Ysurac, 2019-07-22:
Pourquoi tu utilises l'obfuscation ? Si tu habites pas en Chine ça ne sert à rien.

Randdrick, 2019-07-22:
Si je ne coche pas cette case, j'ai ce message d'erreur : 

Mon Jul 22 22:00:11 2019 daemon.err /usr/bin/ss-local[25257]: crypto: stream: repeat IV detected
Mon Jul 22 22:00:11 2019 daemon.err /usr/bin/ss-local[25257]: invalid password or cipher
Mon Jul 22 22:00:15 2019 daemon.err /usr/bin/ss-local[25257]: crypto: stream: repeat IV detected
Mon Jul 22 22:00:15 2019 daemon.err /usr/bin/ss-local[25257]: invalid password or cipher

Randdrick, 2019-07-22:
**Je confirme le problème de débit.**

Le problème des Shadowsocks est "réglé".  Du moins, je n'ai plu eu le message "can't get Ip public adresse".

Ysurac, 2019-07-23:
J'aurais besoin de la configuration (sans les clefs) disponible dans Système->OpenMPTCProuter->Voir tous les paramètres

Randdrick, 2019-07-23:
ddns.global=ddns
ddns.global.ddns_dateformat='%F %R'
ddns.global.ddns_loglines='250'
ddns.global.upd_privateip='0'
ddns.myddns_ipv4=service
ddns.myddns_ipv4.lookup_host='yourhost.example.com'
ddns.myddns_ipv4.domain='yourhost.example.com'
ddns.myddns_ipv4.username='your_username'
ddns.myddns_ipv4.password='your_password'
ddns.myddns_ipv4.interface='wan'
ddns.myddns_ipv4.ip_source='network'
ddns.myddns_ipv4.ip_network='wan'
ddns.myddns_ipv4.service_name='dyn.com'
ddns.myddns_ipv6=service
ddns.myddns_ipv6.update_url='http://[USERNAME]:[PASSWORD]@your.provider.net/nic/update?hostname=[DOMAIN]&myip=[IP]'
ddns.myddns_ipv6.lookup_host='yourhost.example.com'
ddns.myddns_ipv6.domain='yourhost.example.com'
ddns.myddns_ipv6.username='your_username'
ddns.myddns_ipv6.password='your_password'
ddns.myddns_ipv6.use_ipv6='1'
ddns.myddns_ipv6.interface='wan6'
ddns.myddns_ipv6.ip_source='network'
ddns.myddns_ipv6.ip_network='wan6'
dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].server='127.0.0.1#5353'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].quietdhcp='1'
dhcp.@dnsmasq[0].notinterface='wan1'
dhcp.@dnsmasq[0].rebind_domain='plex.direct'
dhcp.@dnsmasq[0].ipset='/googlevideo.com/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/shadow.cdn.prismic.io/omr-cs6' '/shadow.cdn.prismic.io/omr-cs4' '/nflxvideo.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/iptvtool.es/omr-cs4' '/nflxvideo.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googlevideo.com/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googlevideo.com/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googlevideo.com/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googlevideo.com/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googlevideo.com/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googlevideo.com/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googlevideo.com/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googlevideo.com/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googlevideo.com/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr-cs4,omr_dscp-cs4,omr_ [... cut ...]
wireless.radio0.disabled='1'
wireless.radio1=wifi-device
wireless.radio1.type='mac80211'
wireless.radio1.channel='11'
wireless.radio1.hwmode='11g'
wireless.radio1.path='platform/soc/3f300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
wireless.radio1.htmode='HT20'
wireless.radio1.disabled='1'
wireless.default_radio1=wifi-iface
wireless.default_radio1.device='radio1'
wireless.default_radio1.network='lan'
wireless.default_radio1.mode='ap'
wireless.default_radio1.ssid='OpenWrt'
wireless.default_radio1.encryption='none

Randdrick, 2019-07-28:
Post mis à jour

closed 2019-12-20 by Ysurac