Ysurac/openmptcprouter #556: omr-tracker Can't create route

dnwk, 2019-07-23:
The following keeps showing up in log. Not sure what these means. My connection is generally OK but will suffer sudden slow down.

Tue Jul 23 02:18:47 2019 daemon.err /usr/bin/ss-redir[12426]: accept: No file descriptors available
Tue Jul 23 02:18:47 2019 daemon.err /usr/bin/ss-redir[12426]: accept: No file descriptors available
Tue Jul 23 02:18:47 2019 daemon.err /usr/bin/ss-redir[12426]: accept: No file descriptors available
Tue Jul 23 02:18:47 2019 daemon.err /usr/bin/ss-redir[12426]: accept: No file descriptors available
Tue Jul 23 02:18:47 2019 daemon.err /usr/bin/ss-redir[12426]: accept: No file descriptors available
Tue Jul 23 02:18:47 2019 daemon.err /usr/bin/ss-redir[12426]: accept: No file descriptors available
Tue Jul 23 02:18:47 2019 daemon.err /usr/bin/ss-redir[12426]: accept: No file descriptors available
Tue Jul 23 02:18:47 2019 daemon.err /usr/bin/ss-redir[12426]: accept: No file descriptors available
Tue Jul 23 02:18:47 2019 daemon.err /usr/bin/ss-redir[12426]: accept: No file descriptors available
Tue Jul 23 02:18:47 2019 daemon.err /usr/bin/ss-redir[12426]: accept: No file descriptors available
Tue Jul 23 02:18:47 2019 daemon.err /usr/bin/ss-redir[12426]: accept: No file descriptors available
Tue Jul 23 02:18:47 2019 daemon.err /usr/bin/ss-redir[12426]: accept: No file descriptors available
Tue Jul 23 02:18:47 2019 daemon.err /usr/bin/ss-redir[12426]: accept: No file descriptors available
Tue Jul 23 02:18:47 2019 daemon.err /usr/bin/ss-redir[12426]: accept: No file descriptors available

Tue Jul 23 02:20:19 2019 user.notice omr-tracker: Can't create route to 80.67.169.12 via 184.217.*.* dev 3g-WAN3 src 173.132.*.*. waiting...
Tue Jul 23 02:20:21 2019 user.notice omr-tracker: Can't create route to 80.67.169.40 via 184.217.*.* dev 3g-WAN3 src 173.132.*.*. waiting...
Tue Jul 23 02:20:23 2019 user.notice omr-tracker: Can't create route to 4.2.2.1 via 184.217.*.* dev 3g-WAN3 src 173.132.*.*. waiting...
Tue Jul 23 02:20:25 2019 user.notice omr-tracker: Can't create route to 8.8.8.8 via 184.217.*.* dev 3g-WAN3 src 173.132.*.*. waiting...
Tue Jul 23 02:20:27 2019 user.notice omr-tracker: Can't create route to 8.8.4.4 via 184.217.*.* dev 3g-WAN3 src 173.132.*.*. waiting...
Tue Jul 23 02:20:29 2019 daemon.info omr-tracker-ss: Shadowsocks is down (can't contact 1.1.1.1, 1.0.0.1, 212.27.48.10)
Tue Jul 23 02:20:29 2019 user.notice omr-tracker: Can't create route to 9.9.9.9 via 184.217.*.* dev 3g-WAN3 src 173.132.*.*. waiting...
Tue Jul 23 02:20:29 2019 user.notice Shadowsocks: Rules DOWN
Tue Jul 23 02:20:31 2019 user.notice omr-tracker: Can't create route to 1.1.1.1 via 184.217.*.* dev 3g-WAN3 src 173.132.*.*. waiting...
Tue Jul 23 02:20:33 2019 daemon.err omr-tracker-ss[12195]: Failed to establish connection
Tue Jul 23 02:20:33 2019 user.notice omr-tracker: Can't create route to 1.0.0.1 via 184.217.*.* dev 3g-WAN3 src 173.132.*.*. waiting...
Tue Jul 23 02:20:35 2019 user.notice omr-tracker: Can't create route to 80.67.169.12 via 184.217.*.* dev 3g-WAN3 src 173.132.*.*. waiting...
Tue Jul 23 02:20:38 2019 user.notice omr-tracker: Can't create route to 80.67.169.40 via 184.217.*.* dev 3g-WAN3 src 173.132.*.*. waiting...
Tue Jul 23 02:20:40 2019 user.notice omr-tracker: Can't create route to 4.2.2.1 via 184.217.*.* dev 3g-WAN3 src 173.132.*.*. waiting...
Tue Jul 23 02:20:40 2019 daemon.err omr-tracker-ss[12195]: Failed to establish connection
Tue Jul 23 02:20:42 2019 user.notice omr-tracker: Can't create route to 8.8.8.8 via 184.217.*.* dev 3g-WAN3 src 173.132.*.*. waiting...
Tue Jul 23 02:20:44 2019 user.notice omr-tracker: Can't create route to 8.8.4.4 via 184.217.*.* dev 3g-WAN3 src 173.132.*.*. waiting...
Tue Jul 23 02:20:46 2019 user.notice omr-tracker: Can't create route to 9.9.9.9 via 184.217.*.* dev 3g-WAN3 src 173.132.*.*. waiting...
Tue Jul 23 02:20:48 2019 user.notice omr-tracker: Can't create route to 1.1.1.1 via 184.217.*.* dev 3g-WAN3 src 173.132.*.*. waiting...
Tue Jul 23 02:20:48 2019 daemon.info omr-tracker-ss: Shadowsocks is up (can contact 104.16.1.1)
Tue Jul 23 02:20:48 2019 daemon.info omr-tracker-ss: Reload Shadowsocks rules
Tue Jul 23 02:20:48 2019 user.notice Shadowsocks: Rules UP
Tue Jul 23 02:20:50 2019 user.notice omr-tracker: Can't create route to 1.0.0.1 via 184.217.*.* dev 3g-WAN3 src 173.132.*.*. waiting...
Tue Jul 23 02:20:50 2019 user.notice Shadowsocks: Reload omr-bypass rules
Tue Jul 23 02:20:51 2019 user.notice omr-bypass: Starting OMR-ByPass...
Tue Jul 23 02:20:52 2019 user.notice omr-tracker: Can't create route to 80.67.169.12 via 184.217.*.* dev 3g-WAN3 src 173.132.*.*. waiting...
Tue Jul 23 02:20:54 2019 user.notice omr-tracker: Can't create route to 80.67.169.40 via 184.217.*.* dev 3g-WAN3 src 173.132.*.*. waiting...
Tue Jul 23 02:20:57 2019 user.notice omr-tracker: Can't create route to 4.2.2.1 via 184.217.*.* dev 3g-WAN3 src 173.132.*.*. waiting...
Tue Jul 23 02:21:00 2019 user.notice omr-tracker: Can't create route to 8.8.8.8 via 184.217.*.* dev 3g-WAN3 src 173.132.*.*. waiting...
Tue Jul 23 02:21:02 2019 user.notice omr-tracker: Can't create route to 8.8.4.4 via 184.217.*.* dev 3g-WAN3 src 173.132.*.*. waiting...
Tue Jul 23 02:21:07 2019 user.notice omr-tracker: Can't create route to 9.9.9.9 via 184.217.*.* dev 3g-WAN3 src 173.132.*.*. waiting...
Tue Jul 23 02:21:15 2019 user.notice omr-tracker: Can't create route to 1.1.1.1 via 184.217.*.* dev 3g-WAN3 src 173.132.*.*. waiting...
Tue Jul 23 02:21:31 2019 user.notice omr-tracker: Can't create route to 1.0.0.1 via 184.217.*.* dev 3g-WAN3 src 173.132.*.*. waiting...
Tue Jul 23 02:21:49 2019 user.notice omr-tracker: Can't create route to 80.67.169.12 via 184.217.*.* dev 3g-WAN3 src 173.132.*.*. waiting...
Tue Jul 23 02:22:06 2019 user.notice omr-tracker: Can't create route to 80.67.169.40 via 184.217.*.* dev 3g-WAN3 src 173.132.*.*. waiting...
Tue Jul 23 02:22:16 2019 user.notice omr-tracker: Can't create route to 4.2.2.1 via 184.217.*.* dev 3g-WAN3 src 173.132.*.*. waiting...
Tue Jul 23 02:22:21 2019 user.notice omr-tracker: Can't create route to 8.8.8.8 via 184.217.*.* dev 3g-WAN3 src 173.132.*.*. waiting...

## Specifications

  - OpenMPTCProuter version: (Last is not a version) 0.51
  - OpenMPTCProuter VPS version: (Last is not a version) 0.997
  - OpenMPTCProuter platform: (RPI2/RPI3/x86/x86_64) WRT32X

Ysurac, 2019-07-23:
The " accept: No file descriptors available" will be fixed in next release.
For the omr-tracker error I would need the config available in System->OpenMPTCProuter->Show all settings (without the keys). And, even if it's possible, check that gateway for 3g-WAN3 is really not in same range than it's ip and check the name of this iface.

dnwk, 2019-07-23:
3g-wan3 is USB connection. What do you mean by not in the same range?

Ysurac, 2019-07-23:
gateway : 184.217.x.x
ip: 173.132.x.x

dnwk, 2019-07-23:
I turned off omr-tracking at the moment

dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].server='1.1.1.1#53' '127.0.0.1#5353'
dhcp.@dnsmasq[0].rebind_domain='plex.direct'
dhcp.@dnsmasq[0].ipset='/googlevideo.com/omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.cot/omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr-cs2,omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr-cs2,omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr-cs2,omr_dscp-cs2,omr_dscp6-cs2' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/steamcontent.com/omr-cs1,omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.lan.ra='server'
dhcp.lan.ra_preference='high'
dhcp.lan.ndp='relay'
dhcp.lan.ra_management='1'
dhcp.lan.dhcpv6='server'
dhcp.lan.ra_default='1'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dhcp.@host[0]=host
dhcp.@host[0].name='qnap'
dhcp.@host[0].dns='1'
dhcp.@host[0].mac='24:5E:BE:08:FF:C8'
dhcp.@host[0].ip='192.168.100.176'
dhcp.@host[1]=host
dhcp.@host[1].name='tracker'
dhcp.@host[1].dns='1'
dhcp.@host[1].mac='B8:27:EB:57:05:E5'
dhcp.@host[1].ip='192.168.100.125'
dhcp.@host[2]=host
dhcp.@host[2].name='tv'
dhcp.@host[2].dns='1'
dhcp.@host[2].mac='C0:D2:F3:9A:F5:40'
dhcp.@host[2].ip='192.168.100.152'
dropbear.@dropbear[0]=dropbear
dropbear.@dropbear[0].PasswordAuth='on'
dropbear.@dropbear[0].Port='22'
dscp.@classify[0]=classify
dscp.@classify[0].direction='both'
dscp.@classify[0].proto='icmp'
dscp.@classify[0].class='cs5'
dscp.@classify[0].comment='ICMP'
dscp.@classify[1]=classify
dscp.@classify[1].direction='both'
dscp.@classify[1].proto='udp'
dscp.@classify[1].class='cs5'
dscp.@classify[1].src_port='53,123,5353'
dscp.@classify[1].comment='DNS udp and NTP'
dscp.@classify[2]=classify
dscp.@classify[2].direction='both'
[... 1361 lines cut ...]
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.channel='36'
wireless.radio0.hwmode='11a'
wireless.radio0.path='soc/soc:pcie/pci0000:00/0000:00:01.0/0000:01:00.0'
wireless.radio0.htmode='VHT80'
wireless.radio0.disabled='1'
wireless.radio0.country='US'
wireless.radio1=wifi-device
wireless.radio1.type='mac80211'
wireless.radio1.channel='11'
wireless.radio1.hwmode='11g'
wireless.radio1.path='soc/soc:pcie/pci0000:00/0000:00:02.0/0000:02:00.0'
wireless.radio1.htmode='HT20'
wireless.radio1.disabled='1'
wireless.radio1.country='US'
wireless.radio2=wifi-device
wireless.radio2.type='mac80211'
wireless.radio2.channel='36'
wireless.radio2.hwmode='11a'
wireless.radio2.path='platform/soc/soc:internal-regs/f10d8000.sdhci/mmc_host/mmc0/mmc0:0001/mmc0:0001:1'
wireless.radio2.htmode='VHT80'
wireless.radio2.disabled='1'

Powered by openmptcprouter v0.51 r0+10518-6d61fcfa9e

dnwk, 2019-07-23:
They are not in the same range. Also, when I try to go to status page, OpenMPTCPRouter will crash and reboot. I can't find anything interesting in log about the reboot.

Ysurac, 2019-07-24:
The problem appear when gateway can't be used. So connection down or wrong gateway.
At least the gateway can't be ping from the router.

closed 2019-07-25 by dnwk