Ysurac/openmptcprouter #557: MPTCP not enabled in ss-redir(possible also on ss-server)

rstanislav, 2019-07-23:
Some time ago shadowsocks-libev changed its logic, current version allows to enable mptcp only by passing "--mptcp" command line agr, config variable does not work, i noticed this few days ago on my server, and today tested rpi4 image of openmptcprouter - same thing, i did quick workaround fix - in file /etc/init.d/shadowsocks-libev in line 108 added to init string --mptcp, and now it works, not using VPS part (have my custom version) so can't tell if VPS part also affected.

This can be tested very easy - during speedtest check interface utilisation - it will always use only master without --mptcp option.

rstanislav, 2019-07-23:
Also i think tcp-fast open should be off by default, its causing alot of "connection reset by peer" "Socket not connected" and other errors for many people, it is not recommended to use in production env worldwide due to huge problems of NAT traversal..

Ysurac, 2019-07-23:
mptcp is not ignored in json : https://github.com/shadowsocks/shadowsocks-libev/blob/master/src/jconf.c

For tcp-fast-open it's enabled by default in many Linux distribution on outgoing connections. I've enabled it in both directions by default. There is a checkbox to disable it for the few that it's not working for.

closed 2019-07-23 by Ysurac

rstanislav, 2019-07-23:
Well, if you  start ss-server for example like this: 
INFO: enable multipath TCP
/usr/local/bin/ss-server -c /etc/ss-config.json

with "mptcp": true in json, there is no message in log..
but if you start it like this:
/usr/local/bin/ss-server -c /etc/ss-config.json --mptcp
in log it says "INFO: enable multipath TCP"
And also tests with speedtest http download shows that without --mptcp only 1 interface used (others could have activity, but its kb/sec, when main interface have mbits/s...)

Ysurac, 2019-07-23:
The message is only displayed with command line: https://github.com/shadowsocks/shadowsocks-libev/blob/master/src/server.c#L1622
But json conf is read a few line after: https://github.com/shadowsocks/shadowsocks-libev/blob/master/src/server.c#L1712 and mptcp set from config here: https://github.com/shadowsocks/shadowsocks-libev/blob/master/src/server.c#L1749

rstanislav, 2019-07-23:
I have found that also, but still i see that mptcp not used in shadowsocks without --mptcp option, maybe its a bug in shadowsocks mptcp init code(will try to investigate this tomorrow/day after tomorrow)