Ysurac/openmptcprouter #559: OMR-Bypass does not work as expected

warenbe, 2019-07-23:
## Expected Behavior

When adding domain or lan IP in the OMR-Bypass, traffic should be sent through selected WAN.

## Actual Behavior

Traffic is sent through VPN

## Steps to Reproduce the Problem

  1. Add your PC local IP (v4) to the OMR-Bypass and select 1 wan device (like Free ADSL in my case)
  2. go to a website like adresseip.com or ifconfig.me
  3. check the IP: it's the IP of the MTCP server, not the ip of the selected WAN

## Specifications

  - OpenMPTCProuter version: v0.51 r0+10518-6d61fcfa9e
  - OpenMPTCProuter VPS version: 0.997 4.19.56
  - OpenMPTCProuter platform: RPI3 (64b)

In my case, the goal is to force the freebox player (last one) to go through the ADSL directly, not the raspberry
The ADSL modem + 4G modem are in the same local network as the raspberry (yep. it was like this when i was using Overthebox. the goal is to replace OTB by OpenMTCP)

warenbe, 2019-07-23:
Config: 

dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].server='127.0.0.1#5353'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].rebind_domain='plex.direct'
dhcp.@dnsmasq[0].ipset='/googlevideo.com/omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.cot/omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/steamcontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1' '/adresseip.com/omr_dst_bypass_wan2,omr6_dst_bypass_wan2'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.leasetime='12h'
dhcp.lan.force='1'
dhcp.lan.limit='100'
dhcp.lan.netmask='255.255.255.0'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dhcp.@host[0]=host
dhcp.@host[0].name='Olivier-PC'
dhcp.@host[0].dns='1'
dhcp.@host[0].mac='14:DA:E9:06:40:61'
dhcp.@host[0].ip='192.168.0.1'
dhcp.@host[0].leasetime='12h'
dhcp.@host[1]=host
dhcp.@host[1].name='RTX'
dhcp.@host[1].dns='1'
dhcp.@host[1].mac='00:08:7B:0F:AF:73'
dhcp.@host[1].ip='192.168.0.201'
dhcp.@host[1].leasetime='12h'
dhcp.@host[2]=host
dhcp.@host[2].name='FreeboxPlayer'
dhcp.@host[2].dns='1'
dhcp.@host[2].mac='34:27:92:91:2A:C4'
dhcp.@host[2].ip='192.168.0.249'
dhcp.@host[2].leasetime='12h'
dhcp.@host[3]=host
dhcp.@host[3].name='PIPower'
dhcp.@host[3].dns='1'
dhcp.@host[3].mac='50:3E:AA:3B:86:7D'
dhcp.@host[3].ip='192.168.0.130'
dhcp.@host[3].leasetime='12h'
dhcp.@host[4]=host
dhcp.@host[4].name='APWifiSalon'
dhcp.@host[4].dns='1'
dhcp.@host[4].mac='FC:EC:DA:10:F4:45'
dhcp.@host[4].ip='192.168.0.211'
dhcp.@host[4].leasetime='12h'
dhcp.@host[5]=host
dhcp.@host[5].name='APWifiCuisine'
dhcp.@host[5].dns='1'
dhcp.@host[5].mac='FC:EC:DA:10:F4:67'
dhcp.@host[5].ip='192.168.0.212'
[... 1303 lines cut ...]
upnpd.config.port='5000'
upnpd.config.upnp_lease_file='/var/run/miniupnpd.leases'
upnpd.config.enabled='1'
upnpd.config.uuid='8993fd9f-559d-42ea-b9ca-7b814ac3b0af'
upnpd.@perm_rule[0]=perm_rule
upnpd.@perm_rule[0].action='allow'
upnpd.@perm_rule[0].ext_ports='1024-65535'
upnpd.@perm_rule[0].int_addr='0.0.0.0/0'
upnpd.@perm_rule[0].int_ports='1024-65535'
upnpd.@perm_rule[0].comment='Allow high ports'
upnpd.@perm_rule[1]=perm_rule
upnpd.@perm_rule[1].action='deny'
upnpd.@perm_rule[1].ext_ports='0-65535'
upnpd.@perm_rule[1].int_addr='0.0.0.0/0'
upnpd.@perm_rule[1].int_ports='0-65535'
upnpd.@perm_rule[1].comment='Default deny'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='wan1' 'wan2'
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.channel='36'
wireless.radio0.hwmode='11a'
wireless.radio0.path='platform/soc/3f300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
wireless.radio0.htmode='VHT80'
wireless.radio0.disabled='1'

warenbe, 2019-07-24:
Reinstalled everything, moved modem to another subnet, and same issue :(

i tryed to reload the omr-bypass config from ssh and got this result: 

```
root@OpenMPTCProuter:~# /etc/init.d/omr-bypass reload
ip6tables-restore v1.8.3 (legacy): host/network `192.168.0.1' not found
Error occurred at line: 2
Try `ip6tables-restore -h' or 'ip6tables-restore --help' for more information.
udhcpc: started, v1.31.0
udhcpc: sending discover
udhcpc: no lease, failing
```

is it normal?

Ysurac, 2019-07-26:
This error message will be fixed in next release (with a typo bug in OMR-Bypass), CircleCI compile it right now.

warenbe, 2019-07-26:
Excellent! i'm back on otb waiting for next release. i'll test it as soon as it's available.

ArChlOgY, 2019-07-26:
I have similar issue with bypass not working but different error : 

root@OpenMPTCProuter:~# /etc/init.d/omr-bypass reload
/etc/rc.common: /etc/init.d/omr-bypass: line 445: syntax error: unexpected end o                                                           f file (expecting "}")

 VERSION:     v0.51.2

Ysurac, 2019-07-27:
It's what I was speaking about typo. This will be fixed in 0.51.3 (I compile it again, I have a kernel crashloop).

ArChlOgY, 2019-07-27:
I added the "}" at the end of the file (easy fix 😋 ) it seems to work but not...

root@OpenMPTCProuter:~# /etc/init.d/omr-bypass reload
udhcpc: started, v1.31.0
udhcpc: sending discover
udhcpc: no lease, failing

Ysurac, 2019-07-27:
These messages are not a problem. This should work now.

ArChlOgY, 2019-07-27:
NOp :(

Bypass: https://imgur.com/GZZRxgG

Routes : https://imgur.com/NQz77QO

IP check : https://imgur.com/F58ns4g

Ysurac, 2019-07-27:
What is the result of ```iptables-save``` on the router ?

ArChlOgY, 2019-07-27:
> root@OpenMPTCProuter:~# iptables-save
# Generated by iptables-save v1.8.3 on Sat Jul 27 10:49:55 2019
*nat
:PREROUTING ACCEPT [1355:285564]
:INPUT ACCEPT [145:12630]
:OUTPUT ACCEPT [1701:132870]
:POSTROUTING ACCEPT [60:3600]
:MINIUPNPD - [0:0]
:MINIUPNPD-POSTROUTING - [0:0]
:postrouting_lan_rule - [0:0]
:postrouting_rule - [0:0]
:postrouting_vpn_rule - [0:0]
:postrouting_wan_rule - [0:0]
:prerouting_lan_rule - [0:0]
:prerouting_rule - [0:0]
:prerouting_vpn_rule - [0:0]
:prerouting_wan_rule - [0:0]
:ss_rules_dst - [0:0]
:ss_rules_forward - [0:0]
:ss_rules_local_out - [0:0]
:ss_rules_pre_src - [0:0]
:ss_rules_src - [0:0]
:zone_lan_postrouting - [0:0]
:zone_lan_prerouting - [0:0]
:zone_vpn_postrouting - [0:0]
:zone_vpn_prerouting - [0:0]
:zone_wan_postrouting - [0:0]
:zone_wan_prerouting - [0:0]
-A PREROUTING -p tcp -j ss_rules_pre_src
-A PREROUTING -i tun0 -p tcp -m tcp --dport 5006 -j DNAT --to-destination 192.168.100.80:5006
-A PREROUTING -m comment --comment "!fw3: Custom prerouting rule chain" -j prerouting_rule
-A PREROUTING -i eth0 -m comment --comment "!fw3" -j zone_lan_prerouting
-A PREROUTING -i wan1 -m comment --comment "!fw3" -j zone_wan_prerouting
-A PREROUTING -i wan2 -m comment --comment "!fw3" -j zone_wan_prerouting
-A PREROUTING -i wan3 -m comment --comment "!fw3" -j zone_wan_prerouting
-A PREROUTING -i eth1 -m comment --comment "!fw3" -j zone_wan_prerouting
-A PREROUTING -i tun0 -m comment --comment "!fw3" -j zone_vpn_prerouting
-A PREROUTING -i 6in4-omr6in4 -m comment --comment "!fw3" -j zone_vpn_prerouting
-A OUTPUT -p tcp -j ss_rules_local_out
-A POSTROUTING -m comment --comment "!fw3: Custom postrouting rule chain" -j postrouting_rule
-A POSTROUTING -o eth0 -m comment --comment "!fw3" -j zone_lan_postrouting
-A POSTROUTING -o wan1 -m comment --comment "!fw3" -j zone_wan_postrouting
-A POSTROUTING -o wan2 -m comment --comment "!fw3" -j zone_wan_postrouting
-A POSTROUTING -o wan3 -m comment --comment "!fw3" -j zone_wan_postrouting
-A POSTROUTING -o eth1 -m comment --comment "!fw3" -j zone_wan_postrouting
-A POSTROUTING -o tun0 -m comment --comment "!fw3" -j zone_vpn_postrouting
-A POSTROUTING -o 6in4-omr6in4 -m comment --comment "!fw3" -j zone_vpn_postrouting
-A ss_rules_dst -m set --match-set omr_dst_bypass_all dst -j RETURN
-A ss_rules_dst -m set --match-set omr_dst_bypass_6in4-omr6in4 dst -j RETURN
-A ss_rules_dst -m set --match-set omr_dst_bypass_tun0 dst -j RETURN
-A ss_rules_dst -m set --match-set omr_dst_bypass_wan2 dst -j RETURN
-A ss_rules_dst -m set --match-set omr_dst_bypass_eth1 dst -j RETURN
-A ss_rules_dst -m set --match-set omr_dst_bypass_wan3 dst -j RETURN
-A ss_rules_dst -m set --match-set omr_dst_bypass_wan1 dst -j RETURN
-A ss_rules_dst -m set --match-set omr_dst_bypass_eth0 dst -j RETURN
-A ss_rules_dst -m set --match-set omr_dst_bypass_lo dst -j RETURN
-A ss_rules_dst -m set --match-set ss_rules_dst_bypass_all dst -j RETURN
-A ss_rules_dst -m set --match-set ss_rules_dst_bypass dst -j RETURN
-A ss_rules_dst -m set --match-set ss_rules_dst_forward dst -j ss_rules_forward
-A ss_rules_dst -m comment --comment "dst_default: forward" -j ss_rules_forward
-A ss_rules_forward -p tcp -j REDIRECT --to-ports 1100
-A ss_rules_local_out -m set --match-set omr_dst_bypass_all dst -j RETURN
-A ss_rules_local_out -m mark --mark 0x539 -j RETURN
-A ss_rules_local_out -m set --match-set omr_dst_bypass_6in4-omr6in4 dst -j RETURN
-A ss_rules_local_out -m mark --mark 0x5398 -j RETURN
-A ss_rules_local_out -m set --match-set omr_dst_bypass_tun0 dst -j RETURN
-A ss_rules_local_out -m mark --mark 0x5397 -j RETURN
-A ss_rules_local_out -m set --match-set omr_dst_bypass_wan2 dst -j RETURN
-A ss_rules_local_out -m mark --mark 0x5396 -j RETURN
-A ss_rules_local_out -m set --match-set omr_dst_bypass_eth1 dst -j RETURN
[... 308 lines cut ...]
-A zone_wan_dest_ACCEPT -o eth1 -m conntrack --ctstate INVALID -m comment --comment "!fw3: Prevent NAT leakage" -j DROP
-A zone_wan_dest_ACCEPT -o eth1 -m comment --comment "!fw3" -j ACCEPT
-A zone_wan_dest_REJECT -o wan1 -m comment --comment "!fw3" -j reject
-A zone_wan_dest_REJECT -o wan2 -m comment --comment "!fw3" -j reject
-A zone_wan_dest_REJECT -o wan3 -m comment --comment "!fw3" -j reject
-A zone_wan_dest_REJECT -o eth1 -m comment --comment "!fw3" -j reject
-A zone_wan_forward -m comment --comment "!fw3: Custom wan forwarding rule chain" -j forwarding_wan_rule
-A zone_wan_forward -p esp -m comment --comment "!fw3: Allow-IPSec-ESP" -j zone_lan_dest_ACCEPT
-A zone_wan_forward -p udp -m udp --dport 500 -m comment --comment "!fw3: Allow-ISAKMP" -j zone_lan_dest_ACCEPT
-A zone_wan_forward -m conntrack --ctstate DNAT -m comment --comment "!fw3: Accept port forwards" -j ACCEPT
-A zone_wan_forward -m comment --comment "!fw3" -j zone_wan_dest_REJECT
-A zone_wan_input -m comment --comment "!fw3: Custom wan input rule chain" -j input_wan_rule
-A zone_wan_input -p udp -m udp --dport 68 -m comment --comment "!fw3: Allow-DHCP-Renew" -j ACCEPT
-A zone_wan_input -p icmp -m icmp --icmp-type 8 -m comment --comment "!fw3: Allow-Ping" -j ACCEPT
-A zone_wan_input -p igmp -m comment --comment "!fw3: Allow-IGMP" -j ACCEPT
-A zone_wan_input -m conntrack --ctstate DNAT -m comment --comment "!fw3: Accept port redirections" -j ACCEPT
-A zone_wan_input -m comment --comment "!fw3" -j zone_wan_src_REJECT
-A zone_wan_output -m comment --comment "!fw3: Custom wan output rule chain" -j output_wan_rule
-A zone_wan_output -m comment --comment "!fw3" -j zone_wan_dest_ACCEPT
-A zone_wan_src_REJECT -i wan1 -m comment --comment "!fw3" -j reject
-A zone_wan_src_REJECT -i wan2 -m comment --comment "!fw3" -j reject
-A zone_wan_src_REJECT -i wan3 -m comment --comment "!fw3" -j reject
-A zone_wan_src_REJECT -i eth1 -m comment --comment "!fw3" -j reject
COMMIT
# Completed on Sat Jul 27 10:49:55 2019

ArChlOgY, 2019-07-28:
Well, Bypass is working again! The issue was that custom rules  in firewall were applied following this issue #562 

As SNAT module is back again I have deleted the custom rules and created it with SNAT, as a result the Bypass is back as normal.

In conclusion bypass is not working if you apply custom rules in firewall, right ?

Ysurac, 2019-07-28:
No, but custom rules can conflit with omr-bypass rules.

ArChlOgY, 2019-07-28:
ok, those rules below were in conflit with omr-bypass : 

iptables -A PREROUTING -t nat -i tun0 -p tcp --dport 5006 -j DNAT --to 192.168.100.166:80
iptables -A FORWARD -p tcp -d 192.168.100.166 --dport 5006 -j ACCEPT

issue closed for me

warenbe, 2019-07-28:
not for me :(
still now working

updated to v0.51.3 r0+10518-6d61fcfa9e 
update vps to 0.999

now i have this message in status: 
[image]

but, i get connection to internet.
if i disable shadowshock i loose internet connection :( (glorytun not working ??)

bypass stil now working, i get my vps public ip instead of wan1 public ip after adding this: 
[image]

i rebooted both part (VPS, router) without success.

ArChlOgY, 2019-07-28:
can you sreenshot the "Interfaces settings" of wan 1 & wan 2 ? (System > OpenMPTCProuter)

warenbe, 2019-07-28:
that's what you want ?

[image]

[image]

Here are more settings

warenbe, 2019-07-28:
can this be because of omr6in4?

Thu Jul 25 01:04:44 2019 daemon.notice netifd: Interface 'omr6in4' is setting up now
Thu Jul 25 01:04:44 2019 daemon.notice netifd: Interface 'omr6in4' is now down
Thu Jul 25 01:04:44 2019 daemon.notice netifd: Interface 'omr6in4' is setting up now
Thu Jul 25 01:04:44 2019 daemon.notice netifd: Interface 'omr6in4' is now down
Thu Jul 25 01:04:44 2019 daemon.notice netifd: Interface 'omr6in4' is setting up now
Thu Jul 25 01:04:44 2019 daemon.notice netifd: Interface 'omr6in4' is now down
Thu Jul 25 01:04:44 2019 daemon.notice netifd: Interface 'omr6in4' is setting up now
Thu Jul 25 01:04:44 2019 daemon.notice netifd: Interface 'omr6in4' is now down
Thu Jul 25 01:04:44 2019 daemon.notice netifd: Interface 'omr6in4' is setting up now
Thu Jul 25 01:04:44 2019 daemon.notice netifd: Interface 'omr6in4' is now down
Thu Jul 25 01:04:44 2019 daemon.notice netifd: Interface 'omr6in4' is setting up now
Thu Jul 25 01:04:44 2019 daemon.notice netifd: Interface 'omr6in4' is now down
Thu Jul 25 01:04:44 2019 daemon.notice netifd: Interface 'omr6in4' is setting up now
Thu Jul 25 01:04:44 2019 daemon.notice netifd: Interface 'omr6in4' is now down
Thu Jul 25 01:04:44 2019 daemon.notice netifd: Interface 'omr6in4' is setting up now
Thu Jul 25 01:04:44 2019 daemon.notice netifd: Interface 'omr6in4' is now down
Thu Jul 25 01:04:44 2019 daemon.notice netifd: Interface 'omr6in4' is setting up now
Thu Jul 25 01:04:44 2019 daemon.notice netifd: Interface 'omr6in4' is now down
Thu Jul 25 01:04:44 2019 daemon.notice netifd: Interface 'omr6in4' is setting up now
Thu Jul 25 01:04:44 2019 daemon.notice netifd: Interface 'omr6in4' is now down
Thu Jul 25 01:04:44 2019 daemon.notice netifd: Interface 'omr6in4' is setting up now
Thu Jul 25 01:04:44 2019 daemon.notice netifd: Interface 'omr6in4' is now down
Thu Jul 25 01:04:44 2019 daemon.notice netifd: Interface 'omr6in4' is setting up now
Thu Jul 25 01:04:45 2019 daemon.notice netifd: Interface 'omr6in4' is now down
Thu Jul 25 01:04:45 2019 daemon.notice netifd: Interface 'omr6in4' is setting up now

ArChlOgY, 2019-07-28:
not an expert and not Ysurac but are you sure about your ip and gateway configuration ? I'd prefer to set-up as follow : 

WAN1:
192.168.0.240
192.168.0.1

WAN2:
192.168.1.240
192.168.1.1

warenbe, 2019-07-28:
well, i use the same subnet for each modem yes
is it a problem?
i mean, multipath work correctly, just bypass is not working.

ArChlOgY, 2019-07-28:
based on standard configuration recommended you may use different gateway : https://github.com/Ysurac/openmptcprouter/wiki/OpenMPTCProuter-configuration

I never tried both on same since basically I started to set-up my OMR based on standard configuration then adapt as much as possible to my needs ...this is a step by step process with many tests in between  ;)

warenbe, 2019-07-29:
Reinstalled from scratch, set freebox on network 1.254, set 4G on 2.254, set home network to 0.0/24
and now it works!
i think you can close.

warenbe, 2019-07-29:
Question: is it normal that bypass does not work as soon as you use DSCP ?
i tried to add a computer with CS7 for test, and as soon as i added in DSCP, the bypass was not working anymore.

mycanaletto, 2019-09-06:
Il y a des conditions pour que le bypass fonctionne ? Install clean, tout est OK sauf le bypass (PPTP ou Netflix).

ArChlOgY, 2019-09-06:
try Network > Firewall > Custom Rules
click Restart Firewall button

mycanaletto, 2019-09-06:
No more result...

mycanaletto, 2019-09-06:
Je crois avoir trouvé, il faut configurer la time zone... Et il doit y avoir une raison...

mycanaletto, 2019-09-07:
Contournement par adresse MAC. Je pense que le bypass par protocole est lié à une table locale pas à jour ? Ça peut s'expliquer pour Netflix, mais pas pour PPTP.

Ysurac, 2019-09-07:
@mycanaletto please when an issue is in english, use only english in this issue. If you can't use english, open a new issue. And there is not enough info, you speak about MAC, Netflix and PPTP ? Did you test any domain bypass too ?

mycanaletto, 2019-09-07:
OK, I am sorry. I dont use OpemMPTCP as default DNS, perhaps is the problem ?

Ysurac, 2019-09-07:
Yes, at least for netflix or any domain you use, you have to use OpenMPTCProuter as DNS.

mycanaletto, 2019-09-07:
This is a problem with the DNS Active Directory or if openMPTCP is behind a router type USG in which case it will be necessary for the DNS used to use that of openMPTCP relays ... I will do some tests.

This morning without using the DNS of openMPTCP Netflix works, curious! (Netflix protocol bypass)

closed 2019-12-20 by Ysurac