Ysurac/openmptcprouter #561: Usage in redundant mode?

ir-fuel, 2019-07-24:
I am trying to use openmptcprouter for a project that requires data streaming over a tcp connection. What I want to do is to have zero interruptions. What that means is: I have 2 network connections and disabling one of them should simply keep the stream going. I should be able to plug and unplug any connection (not within the same second of course) and the stream should not be interrupted.

I have a running setup on a Raspberry Pi using 2 4G modems and the ethernet port as input, but when testing the above scenario (or simply running "ping" in an infinite loop) I get quite a lot of interruptions when disconnecting one of the modems. 

Wouldn't it be possible to configure it fully redundant whereby everything is sent over both connections and on the VPN server side the fastest packet wins? Similar to 

https://speedify.com/features/fix-packet-loss-redundant-internet-mode/

Thanks

Ysurac, 2019-07-24:
I'm adding a new mode for master route that should give 0 downtime. This will be available in next release, but need more tests.

jwthecolorist, 2019-07-24:
Awesome, looking forward to trying the new mode when it comes out! 

Speaking of which, @Ysurac what do you think about Forward Error Correction for handling lossy links? Is that something that might be possible at some point? Just curious.

Ysurac, 2019-07-24:
I would need to check how I can do that.
FEC can give more stable links but with less available bandwidth.

ir-fuel, 2019-07-24:
I think that's a trade off that's acceptable.
Also how will the system handle the case where one of the links isn't actually down, but extremely slow, as can happen with mobile connections?

Ysurac, 2019-07-25:
With fullmesh patch manager yes because RTT is used as path priority.

A2Razor, 2019-07-26:
> I would need to check how I can do that.
> FEC can give more stable links but with less available bandwidth.

One possible option would be the [UDPSpeeder](https://github.com/wangyu-/UDPspeeder) + OpenVPN + ifenslave.

OpenVPN {TUN, to drop the duplicate UDP-packets} => ifenslave {broadcast mode, mirroring} => OpenVPN TAP {one per connection} => UDP-Speeder > <internet connections> --- reverse on the VPS side

^ I've had very good luck with this combo in the past for gaming, and also for no-loss, low jitter / mission critical situations for businesses.

rstanislav, 2019-07-26:
> I'm adding a new mode for master route that should give 0 downtime. This will be available in next release, but need more tests.

Can you please explain how it works and what difference vs old route logic ?

Ysurac, 2019-07-26:
With new "redundant" mode (it's a bad name, I know) available in current release, you have that in the route table:
```
51.11.111.111 
        nexthop via 192.168.0.1 dev wan1 weight 1 
        nexthop via 192.168.2.1 dev wan2 weight 1 
        nexthop via 192.168.3.1 dev wan3 weight 1 
```
Where 51.11.111.111 is the VPS IP address.

Old logic is using route with different metric to VPS IP address.
I need to make more tests with this...

closed 2019-12-20 by Ysurac