Ysurac/openmptcprouter #562: Port Forwarding - SNAT Interface Missing

ArChlOgY, 2019-07-24:
## Expected Behavior

Apply a Port-forwarding rule as per the wiki : https://github.com/Ysurac/openmptcprouter/wiki/Port-forwarding

At this step on the wiki : "It's a webserver, we need SNAT here. A a new SNAT rule :"

## Actual Behavior

The source NAT Interface is not showed in Firewall - Traffic Rules
https://imgur.com/9ziARsn

Redirects all ports from server to this router is checked

## Steps to Reproduce the Problem

  1. Go to Network > Firewall > Traffic Rules
  2. Scroll to the bottom of the page
  3. No NAT source to configure a New source nat

## Specifications

  - OpenMPTCProuter version: v0.51 r0+10518-6d61fcfa9e
  - OpenMPTCProuter VPS version: 0.997
  - OpenMPTCProuter platform: (RPI2)

reedog117, 2019-07-24:
Same issue occurring

OpenMPTCProuter version: v0.51
OpenMPTCProuter VPS version: 0.9962
OpenMPTCProuter platform: (RPI3B+)

bjoerrrrn, 2019-07-25:
You can try to add a userdefined rule:

iptables -A PREROUTING -t nat -i tun0 -p tcp --dport 80 -j DNAT --to 192.168.100.166:80
iptables -A FORWARD -p tcp -d 192.168.100.166 --dport 80 -j ACCEPT

Adjust 192.168.100.166 to your needs...

ArChlOgY, 2019-07-25:
thanks bjoerrrrn, it did the trick and finally much easier until you know what you are doing with iptable :)

I think that the Source NAT disapear with the last update

Ysurac, 2019-07-25:
Seems that OpenWrt removed SNAT in latest luci-app-firewall package. I will add it again.

Ysurac, 2019-07-25:
luci-app-firewall will be reverted to previous version, current release doesn't work.

Ysurac, 2019-07-26:
Done in 0.51.2

closed 2019-07-26 by Ysurac