Ysurac/openmptcprouter #563: MPTCP blocked?

Bestle, 2019-07-25:
Hopefully a quick one.

I am trying to debug some WAN aggregation issues (mainly slow speeds) and am trying to figure out if MPTCP is blocked by Vodafone UK (which omr-mptcp-intf seems to claim is the case here).

omr-mptcp-intf:
`Nay, Nay, Nay, your have an old computer that does not speak MPTCP. Shame on you!`

MPTCP Support Check:
https://pastebin.com/jL0AYfFt

MPTCP Full Mesh:
`Index, Address-ID, Backup, IP-address, if-idx
IPv4, next v4-index: 3
2, 3, 0, 10.17.192.44 8
IPv6, next v6-index: 0`

Status page:
[image]

So my question is, is MPTCP really disabled, or is omr-mptcp-intf wrong in this case?

  - OpenMPTCProuter version: v0.51
  - OpenMPTCProuter VPS version: N/A
  - OpenMPTCProuter platform: x86_64

Bestle, 2019-07-25:
Speed tests:

`root@OpenMPTCProuter:~# omr-test-speed wwan0
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  2 10.0G    2  229M    0     0  3990k      0  0:43:47  0:00:58  0:42:49 3940k
root@OpenMPTCProuter:~# omr-test-speed wwan1
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  1 10.0G    1  160M    0     0  2370k      0  1:13:43  0:01:09  1:12:34 2732k
root@OpenMPTCProuter:~# omr-test-speed
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  3 10.0G    3  369M    0     0  5980k      0  0:29:13  0:01:03  0:28:10 5785k`

Ysurac, 2019-07-25:
MPTCP seems to work and aggregation is not so bad.

Bestle, 2019-07-25:
Thanks for the quick reply :)

Is omr-mptcp-intf sometimes wrong then? It seems to work fine for all of my WANs except the one I am posting about.

closed 2019-12-20 by Ysurac