Ysurac/openmptcprouter #564: VPS update error

mycanaletto, 2019-07-25:
Erreur sur un VPS 0.997.4. lorsque je lance le script...

2019-07-25 14:12:20 (17.0 MB/s) - '/tmp/linux-headers-4.19.56-mptcp_1.2+a289cca_amd64.deb' saved [9496666/9496666]

Reading package lists... Done
Building dependency tree
Reading state information... Done
curl is already the newest version (7.52.1-5+deb9u9).
rename is already the newest version (0.20-4).
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
(Reading database ... 66846 files and directories currently installed.)
Preparing to unpack .../linux-image-4.19.56-mptcp_1.2+a289cca_amd64.deb ...
Ok, aborting, since modules for this image already exist.
dpkg: error processing archive /tmp/linux-image-4.19.56-mptcp_1.2+a289cca_amd64.deb (--install):
 subprocess new pre-installation script returned error exit status 1
Examining /etc/kernel/postrm.d .
run-parts: executing /etc/kernel/postrm.d/initramfs-tools 4.19.56-mptcp /boot/vmlinuz-4.19.56-mptcp
run-parts: executing /etc/kernel/postrm.d/zz-update-grub 4.19.56-mptcp /boot/vmlinuz-4.19.56-mptcp
**Errors were encountered while processing:**
 /tmp/linux-image-4.19.56-mptcp_1.2+a289cca_amd64.deb

Ysurac, 2019-07-27:
Fixed. Back to previous kernel for now.

closed 2019-07-27 by Ysurac

mycanaletto, 2019-07-27:
Yes, OK for last update