Ysurac/openmptcprouter #565: Problème de redirection de port

Slivikk-fr, 2019-07-25:
## Actual Behavior
Et bien je vais la faire à la Française, déjà bonjour, j'ai actuellement un problème avec les ports d'ouvert en principe mais je n'arrive pas à me connecter en SSH sur mon serveur qui est chez moi, pourtant tout est ouvert que ce soit box (je pense que c'est inutile si je passe par OVH) mais quand j'essaye de me connecter au serveur ça me fait "Connexion refused" alors que j'ai fait la redirection de port, ajout d'un SNAT, et vérifier que tout est fait en sorte que les ports sois ouverts [](https://media.discordapp.net/attachments/572830242444673024/603938039496048643/unknown.png?width=843&height=474)

## Steps to Reproduce the Problem

  1.--
  2.
  3.

## Specifications

  - OpenMPTCProuter version: (0.51.1)
  - OpenMPTCProuter VPS version: (Last is not a version)
  - OpenMPTCProuter platform: (RPI3)

bjoerrrrn, 2019-07-25:
Check Out:
https://github.com/Ysurac/openmptcprouter/issues/562

Slivikk-fr, 2019-07-25:
Je l'ai déjà vu et ce n'est pas pour autant que le problème est résolu

Slivikk-fr, 2019-07-25:
J’essaierais tout de même même si les règles apparaissent bien

Slivikk-fr, 2019-07-25:
Même après avoir fait la même chose que dans le topic précédent je vais envoyer tout le fichier de configuration ici : `dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].server='127.0.0.1#5353'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].rebind_domain='plex.direct'
dhcp.@dnsmasq[0].ipset='/googlevideo.com/omr_dscp-cs4,omr_dscp6-cs4' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/nflxvideo.net/omr_dscp-cs4,omr_dscp6-cs4' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/s3.ll.dash.row.aiv-cdn.net/omr_dscp-cs4,omr_dscp6-cs4' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/d25xi40x97liuc.cloudfront.net/omr_dscp-cs4,omr_dscp6-cs4' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/aiv-delivery.net/omr_dscp-cs4,omr_dscp6-cs4' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/steamcontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/audio-fa.scdn.cot/omr_dscp-cs4,omr_dscp6-cs4' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.lan.dhcpv6='server'
dhcp.lan.ra='server'
dhcp.lan.ra_default='1'
dhcp.lan.ra_preference='high'
dhcp.lan.ra_management='1'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dhcp.@host[0]=host
dhcp.@host[0].name='Mailserver'
dhcp.@host[0].dns='1'
dhcp.@host[0].mac='00:0C:29:D0:E4:94'
dhcp.@host[0].ip='192.168.100.231'
dhcp.@host[1]=host
dhcp.@host[1].name='adky-vps-linux-1'
dhcp.@host[1].dns='1'
dhcp.@host[1].mac='00:0C:29:CF:3A:5B'
dhcp.@host[1].ip='192.168.100.139'
dhcp.@domain[0]=domain
dhcp.@domain[0].name='techrp.fr'
dhcp.@domain[0].ip='192.168.100.139'
dhcp.@domain[1]=domain
dhcp.@domain[1].name='adkynet.com'
dhcp.@domain[1].ip='192.168.100.139'
dhcp.@domain[2]=domain
dhcp.@domain[2].name='lewizooo.eu'
dhcp.@domain[2].ip='192.168.100.139'
dropbear.@dropbear[0]=dropbear
dropbear.@dropbear[0].PasswordAuth='on'
dropbear.@dropbear[0].Port='22'
dropbear.@dropbear[0].Interface='lan'
dscp.@classify[0]=classify
dscp.@classify[0].direction='both'
dscp.@classify[0].proto='icmp'
dscp.@classify[0].class='cs5'
dscp.@classify[0].comment='ICMP'
dscp.@classify[1]=classify
dscp.@classify[1].direction='both'
dscp.@classify[1].proto='udp'
dscp.@classify[1].class='cs5'
dscp.@classify[1].src_port='53,123,5353'
dscp.@classify[1].comment='DNS udp and NTP'
[... 1256 lines cut ...]
upnpd.@perm_rule[0].comment='Allow high ports'
upnpd.@perm_rule[1]=perm_rule
upnpd.@perm_rule[1].action='deny'
upnpd.@perm_rule[1].ext_ports='0-65535'
upnpd.@perm_rule[1].int_addr='0.0.0.0/0'
upnpd.@perm_rule[1].int_ports='0-65535'
upnpd.@perm_rule[1].comment='Default deny'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='wan1'
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.hwmode='11a'
wireless.radio0.path='platform/soc/3f300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
wireless.radio0.htmode='VHT80'
wireless.radio0.disabled='0'
wireless.radio0.channel='auto'
wireless.radio0.legacy_rates='1'
wireless.@wifi-iface[0]=wifi-iface
wireless.@wifi-iface[0].network='wwan'
wireless.@wifi-iface[0].ssid='AdkyNet.com'
wireless.@wifi-iface[0].encryption='psk2'
wireless.@wifi-iface[0].device='radio0'
wireless.@wifi-iface[0].mode='sta'
wireless.@wifi-iface[0].bssid='18:D2:76:37:BA:31'
wireless.@wifi-iface[0].key='57894565789'`

Ysurac, 2019-07-25:
je vois un "openmptcprouter.vps.admin_error='1'' donc un soucis de communication avec le VPS.
Tu peux vérifier sur le VPS, dans /etc/shorewall/rules que les régles de DNAT sont décommentées.

Slivikk-fr, 2019-07-25:
Et comment je peut voir ça ? Directement en SSH si oui comment ?

Ysurac, 2019-07-25:
oui via SSH : ```cat /etc/shorewall/rules```

Slivikk-fr, 2019-07-25:
cat: can't open '/etc/shorewall/rules': No such file or directory

Ysurac, 2019-07-25:
Sur le VPS :)

Slivikk-fr, 2019-07-25:
ah ok il est sur l'image directement c'est quoi les identifiants ?

Ysurac, 2019-07-25:
Sur la partie serveur tu as normalement déjà lancé un script d'installation via SSH.

mycanaletto, 2019-07-25:
DNAT            net             vpn:$OMR_ADDR   tcp     1-64999
DNAT            net             vpn:$OMR_ADDR   udp     1-64999

de-commentées et mm problème

Slivikk-fr, 2019-07-25:
Bon je vais check tout ça :)

Slivikk-fr, 2019-07-25:
Faudrais-t-il déjà que j'arrive à me connecter au VPS

mycanaletto, 2019-07-25:
> Faudrais-t-il déjà que j'arrive à me connecter au VPS

Si tu n'arrive pas à te connecter à ton VPS en SSH soit il est down, soit tu as loupé qq chose car tu t'y est nécessairement connecté lors de l'installation...

Ysurac, 2019-07-25:
Port SSH du VPS : 65222

Slivikk-fr, 2019-07-25:
Bon le VPS à merder je réinstalle le tout je vous tient au courant ici 👍

mycanaletto, 2019-07-25:
DNAT net vpn:$OMR_ADDR tcp 1-64999
DNAT net vpn:$OMR_ADDR udp 1-64999
Coté VPS mais redirection de ports NOK

Ysurac, 2019-07-25:
@mycanaletto tu l'avais déjà dit...

Ysurac, 2019-07-25:
Il y a un soucis côté firewall sur OpenMPTCProuter. J'essaye de trouver pourquoi.

Slivikk-fr, 2019-07-25:
Ah D'accord

Ysurac, 2019-07-25:
L'interface ne fonctionne pas du tout. Je vais revenir à la version précédente dans la prochaine version d'OpenMPTCProuter qui sortira rapidement.

Slivikk-fr, 2019-07-25:
Donc indirectement ce n'est pas de ma faute ?

Ysurac, 2019-07-25:
Non, j'ai été peut être un peu trop rapide pour utiliser ces versions ;)

Slivikk-fr, 2019-07-25:
Ah d'accord, maintenant je n'ai plus de message "connection refused" mais un chargement infini c'est peut être bon signe

Slivikk-fr, 2019-07-26:
J'ai mis à jour et j'ai toujours un timed out :(

mycanaletto, 2019-07-26:
Toujours aussi réactif, bravo et merci pour la .52 👍

Slivikk-fr, 2019-07-26:
J'ai dé-commenter les lignes et j'ai des erreurs :( [](https://cdn.discordapp.com/attachments/465155797517598740/604273900502908928/sdzzd.PNG)

Slivikk-fr, 2019-07-26:
J'ai plus précisément les erreur 

Côté router : Filesystem IS readonly

Et côté vos j'ai : Can't connect server admin script (no token) 

Je regarderais tout a l'heure je dois m'absenter

Slivikk-fr, 2019-07-26:
Tout est bon merci beaucoup pour votre aide 😏

closed 2019-07-26 by Slivikk-fr