Ysurac/openmptcprouter #566: How to enable WiFi AP mode on RPi 4

rstanislav, 2019-07-25:
Hello! Can some one please give advice, how to enable internal wifi of rpi4 in AP mode ?
I have configured country code, ssid, even started interface and checked that can connect to it in raspbian, but in openmptcprouter in luci there is no even wireless menu, iw list shows nothing, ifconfig -a also (only wired interfaces and vpn).

rstanislav, 2019-07-25:
Answer: via ssh do:
rmmod brcmfmac
insmod brcmfmac
openwrt creates base config, and wireless tab becomes available, but AP should be configured in "Legacy" mode.

Ysurac, 2019-07-25:
With 32bits or 64bits image ?

rstanislav, 2019-07-26:
64 bit, i have found that after reboot it does not work until i do 

rmmod brcmfmac
insmod brcmfmac
sleep 1
/sbin/wifi down
sleep 1
/sbin/wifi up

i have put this to startup script, but its durty workaround :/

Ysurac, 2019-07-26:
I don't have this wifi issue on my RPI4.

There is already a check on ext4 at boot, and a message in status page if it's RO. If you don't want RO, don't unplug power of the RPI, it really don't like that.

You can disable IP check on external site, but for now it's not possible to change it.

Ysurac, 2019-07-26:
What is the RPI4 you have ? 1Go, 2Go or 4Go ? Can you give me the result of dmesg via SSH on the router ?

rstanislav, 2019-07-29:
`dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 4.19.57 (root@build-test) (gcc version 7.4.0 (OpenWrt GCC 7.4.0 r0+10518-6d61fcfa9e)) #0 SMP Tue Jul 16 03:33:46 2019
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.1
[    0.000000] cma: Reserved 256 MiB at 0x0000000008000000
[    0.000000] On node 0 totalpages: 504832
[    0.000000]   DMA32 zone: 8192 pages used for memmap
[    0.000000]   DMA32 zone: 0 pages reserved
[    0.000000]   DMA32 zone: 504832 pages, LIFO batch:63
[    0.000000] random: get_random_bytes called from start_kernel+0x88/0x3a0 with crng_init=0
[    0.000000] percpu: Embedded 21 pages/cpu s48152 r8192 d29672 u86016
[    0.000000] pcpu-alloc: s48152 r8192 d29672 u86016 alloc=21*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: enabling workaround for EL2 vector hardening
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 496640
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 cma=64M bcm2708_fb.fbwidth=0 bcm2708_fb.fbheight=0 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32:07:25:3D vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 kgdboc=ttyS0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=squashfs,ext4 coherent_pool=128M cma=256M@128M swiotlb=force rootwait
[    0.000000] Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes)
[    0.000000] Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes)
[    0.000000] software IO TLB: mapped [mem 0x79a00000-0x7da00000] (64MB)
[    0.000000] Memory: 1644884K/2019328K available (6462K kernel code, 466K rwdata, 1776K rodata, 384K init, 508K bss, 112300K reserved, 262144K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: 	CONFIG_RCU_FANOUT set to non-default value of 32.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000005] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000175] Console: colour dummy device 80x25
[    0.000393] console [tty1] enabled
[    0.000422] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=540000)
[    0.000446] pid_max: default: 32768 minimum: 301
[    0.000584] Mount-cache hash table entries: 4096 (order: 3, 32768 bytes)
[    0.000622] Mountpoint-cache hash table entries: 4096 (order: 3, 32768 bytes)
[    0.001847] ASID allocator initialised with 32768 entries
[    0.001935] rcu: Hierarchical SRCU implementation.
[    0.002643] smp: Bringing up secondary CPUs ...
[    0.002995] Detected PIPT I-cache on CPU1
[    0.003045] CPU1: Booted secondary processor 0x0000000001 [0x410fd083]
[    0.003458] Detected PIPT I-cache on CPU2
[    0.003495] CPU2: Booted secondary processor 0x0000000002 [0x410fd083]
[    0.003895] Detected PIPT I-cache on CPU3
[    0.003930] CPU3: Booted secondary processor 0x0000000003 [0x410fd083]
[    0.004005] smp: Brought up 1 node, 4 CPUs
[    0.004066] SMP: Total of 4 processors activated.
[    0.004080] CPU features: detected: 32-bit EL0 Support
[    0.006365] CPU: All CPU(s) started at EL2
[    0.006395] alternatives: patching kernel code
[    0.007076] devtmpfs: initialized
[    0.012994] Enabled cp15_barrier support
[    0.013028] Enabled setend support
[    0.013188] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.013216] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.019098] pinctrl core: initialized pinctrl subsystem
[    0.019389] reg-dummy reg-dummy: no of_node; not parsing pinctrl DT
[    0.019969] NET: Registered protocol family 16
[    0.020895] cpuidle: using governor ladder
[    0.020928] cpuidle: using governor menu
[    0.021201] vdso: 2 pages (1 code @ (____ptrval____), 1 data @ (____ptrval____))
[    0.104327] DMA: preallocated 131072 KiB pool for atomic allocations
[    0.104448] Serial: AMBA PL011 UART driver
[    0.106197] bcm2835-mbox fe00b880.mailbox: mailbox enabled
[    0.125968] cryptd: max_cpu_qlen set to 1000
[    0.126412] bcm2835-dma: probe of fe007000.dma failed with error -5
[    0.127923] of_get_named_gpiod_flags: can't parse 'gpio' property of node '/fixedregulator_3v3[0]'
[    0.128119] of_get_named_gpiod_flags: can't parse 'gpio' property of node '/fixedregulator_5v0[0]'
[    0.128394] of_get_named_gpiod_flags: can't parse 'enable-gpio' property of node '/sd_io_1v8_reg[0]'
[    0.128672] SCSI subsystem initialized
[... 740 lines cut ...]
[   20.713488] uas 2-1.2:1.2: no of_node; not parsing pinctrl DT
[   20.713527] usb-storage 2-1.2:1.2: no of_node; not parsing pinctrl DT
[   20.713541] usb-storage 2-1.2:1.2: USB Mass Storage device detected
[   20.720432] scsi host2: usb-storage 2-1.2:1.2
[   21.192120] scsi 1:0:0:0: Direct-Access     HUAWEI   TF CARD Storage  2.31 PQ: 0 ANSI: 2
[   21.200584] sd 1:0:0:0: no of_node; not parsing pinctrl DT
[   21.201276] sd 1:0:0:0: Power-on or device reset occurred
[... cut ...]
Mon Jul 29 19:00:08 2019 kern.info kernel: [  164.958766] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
Mon Jul 29 19:00:08 2019 kern.info kernel: [  164.976831] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Mar  1 2015 07:29:38 version 7.45.18 (r538002) FWID 01-6a2c8ad4
Mon Jul 29 19:00:08 2019 kern.debug kernel: [  165.024630] brcmfmac mmc1:0001:3: no of_node; not parsing pinctrl DT
Mon Jul 29 19:00:08 2019 kern.info kernel: [  165.024887] usbcore: registered new interface driver brcmfmac`

bournehlm, 2019-08-27:
I also have that problem, I can't see the wifi item in the Luci interface.
I have 2GB rpi4 and installed 32bit ext4 image (openmptcprouter-v0.51.5-r010518-6d61fcfa9e-brcm2708-bcm2709-rpi-2-ext4-factory.img.gz ).

closed 2019-12-20 by Ysurac

iamdivyanshverma, 2022-04-08:
Mon Jul 29 18:49:45 2019 daemon.err omr-tracker[11902]: /bin/omr-tracker: line 221: can't create 1: Read-only file system
Mon Jul 29 18:49:45 2019 daemon.err omr-tracker[11902]: /bin/omr-tracker: line 217: can't create 1: Read-only file system
Mon Jul 29 18:49:45 2019 user.notice omr-tracker: Can't create route to 9.9.9.9 via 192.168.161.1 dev eth3 src 192.168.161.100. waiting...

""you can fix this with fsck""