Ysurac/openmptcprouter #567: eth0: hw csum failure

dnwk, 2019-07-25:
Kernel warnings in System log:

:24 2019 kern.err kernel: [  944.727567] eth0: hw csum failure
Thu Jul 25 07:58:53 2019 kern.warn kernel: [ 1093.901790]  dump_backtrace+0x0/0x130
Thu Jul 25 07:58:53 2019 kern.warn kernel: [ 1093.906280]  show_stack+0x14/0x20
Thu Jul 25 07:58:53 2019 kern.warn kernel: [ 1093.910410]  dump_stack+0x9c/0xbc
Thu Jul 25 07:58:53 2019 kern.warn kernel: [ 1093.914518]  netdev_rx_csum_fault+0x38/0x48
Thu Jul 25 07:58:53 2019 kern.warn kernel: [ 1093.919518]  __skb_checksum_complete+0x6c/0xc0
Thu Jul 25 07:58:53 2019 kern.warn kernel: [ 1093.924787]  tcp_v6_rcv+0x1c8/0xdf8
Thu Jul 25 07:58:53 2019 kern.warn kernel: [ 1093.929099]  ip6_input_finish+0x260/0x368
Thu Jul 25 07:58:53 2019 kern.warn kernel: [ 1093.933931]  ip6_input+0x70/0x80
Thu Jul 25 07:58:53 2019 kern.warn kernel: [ 1093.937961]  ipv6_rcv+0x94/0xb8
Thu Jul 25 07:58:53 2019 kern.warn kernel: [ 1093.941888]  __netif_receive_skb_one_core+0x3c/0x48
Thu Jul 25 07:58:53 2019 kern.warn kernel: [ 1093.947612]  __netif_receive_skb+0x6c/0x78
Thu Jul 25 07:58:53 2019 kern.warn kernel: [ 1093.952530]  process_backlog+0xa8/0x170
Thu Jul 25 07:58:53 2019 kern.warn kernel: [ 1093.957185]  net_rx_action+0x110/0x298
Thu Jul 25 07:58:53 2019 kern.warn kernel: [ 1093.961751]  __do_softirq+0xe0/0x1dc
Thu Jul 25 07:58:53 2019 kern.warn kernel: [ 1093.966141]  irq_exit+0x60/0xb8
Thu Jul 25 07:58:53 2019 kern.warn kernel: [ 1093.970087]  handle_IPI+0xac/0x140
Thu Jul 25 07:58:53 2019 kern.warn kernel: [ 1093.974291]  bcm2836_arm_irqchip_handle_irq+0x8c/0xb8
Thu Jul 25 07:58:53 2019 kern.warn kernel: [ 1093.980181]  el1_irq+0xb0/0x140
Thu Jul 25 07:58:53 2019 kern.warn kernel: [ 1093.984097]  arch_cpu_idle+0x10/0x18
Thu Jul 25 07:58:53 2019 kern.warn kernel: [ 1093.988427]  do_idle+0x10c/0x228
Thu Jul 25 07:58:53 2019 kern.warn kernel: [ 1093.992367]  cpu_startup_entry+0x20/0x28
Thu Jul 25 07:58:53 2019 kern.warn kernel: [ 1093.997025]  secondary_start_kernel+0x1ac/0x1c0

Anything to worry?

  - OpenMPTCProuter version: (Last is not a version) 0.51.1
  - OpenMPTCProuter VPS version: (Last is not a version) 0.999
  - OpenMPTCProuter platform: (RPI2/RPI3/x86/x86_64)RPI 3B+ ext4 64bit

Ysurac, 2019-07-25:
Try ```ethtool --offload eth0 rx off tx off```. This should fix the issue. I disabled it for 4.19 kernel, but the problem seems to be still here.

dnwk, 2019-07-26:
Yes. after the command, the kern.wan went away. Not sure what that error means.

closed 2019-07-26 by dnwk