Ysurac/openmptcprouter #568: OVA ESX

mycanaletto, 2019-07-25:
Merci pour les OVA !

La version EFI ne démarre pas. Par contre la version non EFI EXT4 OK. Il manque juste open-vmware tools... Et on a une erreur non critique sur le choix de l'OS, en acceptant ça passe en other Linux 64 (du au fait que je suis en ESX 5.5).

Ysurac, 2019-07-25:
Effectivement, j'ai utilisé le même OVF pour les 2 images, j'ai donc oublié de remplacer le "bios" par "efi" :)

mycanaletto, 2019-07-25:
Merci pour la prochaine release. Ce serait top aussi d'ajouter open-vmaware-tools..

Ysurac, 2019-07-25:
Il est dispo dans les packages à installer mais je peux pas le mettre par défaut car il plante si il n'est pas dans une VM vmware.

mycanaletto, 2019-07-25:
C'est bizzare, avant hier suie au problème hfs j'ai rééinstllé une nouvelle vm à partir de ça _openmptcprouter-v0.51-r010518-6d61fcfa9e-x86-64-combined-ext4.vmdk_ et les vmware tools sont déjà présents...

Ysurac, 2019-07-25:
Oui, j'ai viré dernièrement.

mycanaletto, 2019-07-25:
C'est du cosmétique. Pas le plus grave :-)

closed 2019-12-20 by Ysurac