Ysurac/openmptcprouter #569: Compile Error on v2ray Plugin

iamfreaky, 2019-07-25:
So Tried to Compile OpenMptcp for my own Router ( Edge Router X ) but sadly it fails. Does hanybody have a solution for it ?.

The Router is not offical supported by openmptcp but supported from openwrt. 

Cleaning module download cache (golang/go#27455)
go: finding github.com/shadowsocks/v2ray-plugin/.go_work/build latest
go: finding github.com/shadowsocks/v2ray-plugin/.go_work latest
can't load package: package github.com/shadowsocks/v2ray-plugin/.go_work/build: unknown import path "github.com/shadowsocks/v2ray-plugin/.go_work/build": cannot find module providing package github.com/shadowsocks/v2ray-plugin/.go_work/build

make[3]: *** [Makefile:44: /home/chris/drive/edge-router-x/openmptcprouter/myplatform/source/build_dir/target-mipsel_24kc_musl/v2ray-plugin-1e5b3780/.built] Error 2
make[3]: Leaving directory '/home/chris/drive/edge-router-x/openmptcprouter/feeds/openmptcprouter/shadowsocks-v2ray-plugin'
time: package/feeds/openmptcprouter/shadowsocks-v2ray-plugin/compile#17.38#3.90#34.44
make[2]: *** [package/Makefile:109: package/feeds/openmptcprouter/shadowsocks-v2ray-plugin/compile] Error 2
make[2]: Leaving directory '/home/chris/drive/edge-router-x/openmptcprouter/myplatform/source'
make[1]: *** [package/Makefile:103: /home/chris/drive/edge-router-x/openmptcprouter/myplatform/source/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/chris/drive/edge-router-x/openmptcprouter/myplatform/source'
make: *** [/home/chris/drive/edge-router-x/openmptcprouter/myplatform/source/include/toplevel.mk:218: world] Fehler 2

iamfreaky, 2019-07-27:
@Ysurac You dont have a solution for it ?

Ysurac, 2019-07-27:
Not for now. There is several problems with compilation for MIPS

svdwalt007, 2019-08-28:
Any updates to fix this yet ?

closed 2019-12-20 by Ysurac

2muaddlb, 2023-07-19:
Hi, Ysurac!
Did you manage to fix it?

Ysurac, 2023-07-19:
Shadowsocks v2ray-plugin is not available anymore. Better to use directly V2Ray proxy.

2muaddlb, 2023-07-19:
Do you mean there is a V2Ray proxy for OpenWRT/OpenMptcp out there as a standalone proxy server?

Ysurac, 2023-07-19:
I mean that V2Ray is integrated in OpenMPTCProuter and can be used as proxy instead of ShadowSocks via System->OpenMPTCProuter, and "advanced settings" checkbox in the wizard.