Ysurac/openmptcprouter #570: Problème avec 4G Box (Bouygues Télécom)

Slivikk-fr, 2019-07-26:
## Expected Behavior

Bonjour j'ai reçu ma 4G Box aujourd'hui et j'ai configurer comme indiqué sur le wiki mais ça me met Gateway Down :-( 

## Actual Behavior

## Steps to Reproduce the Problem

  1. Suivre le wiki
  2. Mettre les mêmes IP avec le DHCP de désactivé
  3.Gateway down

## Specifications

  - OpenMPTCProuter version: (0.51.2)
  - OpenMPTCProuter VPS version: (0.999)
  - OpenMPTCProuter platform: (RPI3B+)

Ysurac, 2019-07-26:
Le gateway down correspond à un ping vers la passerelle, si elle répond pas on a ce message.
Donc soit la configuration est mauvaise, soit le ping est bloqué sur la box.
Tu peux mettre une capture d'écran de la page status ?

Slivikk-fr, 2019-07-26:
[image]

Slivikk-fr, 2019-07-26:
[image]

Ysurac, 2019-07-26:
Donc pour le wwan il arrive à trouver l'adresse WAN, donc le modem ne répond surement pas au ping. Pas forcément de soucis à ce niveau.
Pour l'interface 4GBOX il est fort possible qu'il y a un soucis. Tu peux tester via SSH sur le routeur un ping vers 192.168.10.1 et sinon revérifie ta configuration et tes câbles.

Slivikk-fr, 2019-07-26:
Dès que je met ma 4G BOX sur mon switch relié au RPI il ce met en rouge (le signale passe en rouge) mais dès que je met mon pc portable c'est bleu ;/ C'est peut être du à la première ligne qui fait chier la 4G Box ?

Slivikk-fr, 2019-07-26:
La 4G Box ne répond pas au ping :/

Slivikk-fr, 2019-07-26:
J'ai beau autoriser le ping sur le wan , changer de switch, câble ethernet rien n'y fait ça veut pas du tout

Slivikk-fr, 2019-07-26:
Quand y'a le raspberry pi sur le switch ça fonctionne pas et dès qu'il n'est plus sur le switch ça fonctionne

Ysurac, 2019-07-26:
Ca ressemble à un conflit d'IP ça. Tu as bien mis 192.168.10.1 sur la box comme IP fixe et pas .2 ?
Et il n'y a rien déjà présent en 192.168.10.2 ?

Slivikk-fr, 2019-07-26:
OUI tout est bon de ce côté

Breizor, 2019-07-27:
si le "wwan" correspond au wifi de la 4G box, l'adresse wan commune entre "wwan" et 4GBOX" pourrait causer le conflit?

Sinon pour dépanner: utilise le wifi de la 4Gbox, le DHCP peut être activé sur wwan (pas le même réseau physique)...pratique si tu partage la connection avec un smartphone.
[image]

[image]

Slivikk-fr, 2019-07-27:
En fait le WWAN c'est mon téléphone :/

Slivikk-fr, 2019-07-27:
Alors si j'avais la possibilité de mettre mes deux réseaux sur la même carte wifi (Utilisant la bande 5Ghz pour le routeur 4G et la bande 2.4Ghz pour le téléphone) où sur un adaptateur USB j'aimerais bien comme ça ça m'éviterai d'avoir ces problème embêtant :(

Slivikk-fr, 2019-07-27:
J'ai réussi à tout faire fonctionner mais sérieux problème de débit je devrais être en théorie à 70-100Mbits et je n'atteint pas 10Mbits :(

Ysurac, 2019-07-27:
Dans _réseau->MPTCP->MPTCP fullmesh_ tu as bien toute tes IPs ?

Slivikk-fr, 2019-07-27:
J'ai deux adresses IP effectivement

Slivikk-fr, 2019-07-27:
[image]
ARRRRRRRRRRRRRRRRRRRRRRRRRRRRRRG

Ysurac, 2019-07-27:
Il arrive pas à utiliser wan1 la, vu le rouge.

Slivikk-fr, 2019-07-27:
Pourtant tout à l'heure tout étais ok tout à l'heure

closed 2019-07-28 by Slivikk-fr

PierreGranger, 2020-05-30:
Bonjour,

Ayant eu un problème équivalent moi même, j'ai pu le régler en allant sur la 4Gbox dans Paramètres réseau > Ethernet > Paramètres Ethernet > Mode de connexion > LAN uniquement.

Il semblerait que le port Ethernet de la box ne soit au départ pas destiné à se relier à un réseau local mais plutôt comme fournisseur de la connexion Internet (un peu comme les box utilisant du Wifimax, dont un port Ethernet est destiné à relier la box à l'antenne).

En passant en LAN uniquement je pense qu'on destine le port Ethernet de la box à une utilisation classique.