Ysurac/openmptcprouter #572: bug VPS kernel panic

cayenne17, 2019-07-27:
Bonjour,
J'ai la dernière version du VPS 0.999 et du routeur 0.51.2 r0+10518-6d61fcfa9e.

Je viens de tester de rajouter une connexion partagé depuis un Google Nexus 5x 4G Bouygues Telecom en plus de la Freebox en VDSL2 et mon modem router Zyxel LTE4506 avec une sim free. 

Quand je lance quelque speedtest nperf, j'ai mon shadowsocks qui coupe. Quand je vais voir sur le VPS (machine virtual proxmox VE) j'ai kernel panique + cpu/RAM à 100%.
j'ai tester et 3 fois kernel panique / cpu/RAM à 100%

j'envoi des screenshots.

[image]
[image]
[image]
[image]

Ysurac, 2019-07-27:
Un soucis de mémoire, 1Go pour le VPS c'est vraiment pas beaucoup.
Tu peux installer htop et vérifier ce qu'il se passe vraiment.

cayenne17, 2019-07-27:
j'ai une monté d'un coup et plantage 
voici un htop :
[image]

je vais mettre la VM à 2 Go

cayenne17, 2019-07-27:
htop avec 2 Go de RAM pendant 2 nperf en même temps :
[image]

cayenne17, 2019-07-27:
Kernel panic encore. une fuite mémoire ?

[image]

[image]

Ysurac, 2019-07-27:
What is the result if you stop mlvpn  ? ```systemctl stop mlvpn@mlvpn0```

cayenne17, 2019-07-27:
Voici ce qui se passe

[image]

[image]
 je ne perd pas la connexion. dés que je relance plusieurs nperf, monté en RAM puis crash.

Ysurac, 2019-07-27:
Plusieurs nperf simultanément ?

cayenne17, 2019-07-27:
oui, car si j'en lance un seul, il me faut un gros débit comme ce matin pour que ça crash

closed 2019-12-20 by Ysurac