Ysurac/openmptcprouter #573: RPI4 0.51.3 Sysupgrade - Filesystem is Read-Only

bjoerrrrn, 2019-07-27:
## Actual Behavior

After the Update via Webgui the Filesystem is Read-Only - after another Reboot the Configuration is gone.

For testing purpose i rebuilt everything and tested via direct sd-card-flashing. Same for me.

Back on 0.51.2 everything is fine.

## Steps to Reproduce the Problem

  1. Firmware Upgrade on RPI4 from 0.51.2 to 0.51.3 sysupgrade Image (WebGUI Flash and SD-Card-Flash makes no difference for me)

## Specifications

  - OpenMPTCProuter version: 0.51.3 
  - OpenMPTCProuter VPS version: 0.9962 on Debian 9
  - OpenMPTCProuter platform: RPI4

Ysurac, 2019-07-28:
Ext4 or squashfs ?
Squashfs with F2FS seems top have more problem after a hard shutdown or with faulty SD card.
There is also less problems with good quality sd card.

Dragonox77, 2019-07-28:
Same prob here.

rstanislav, 2019-07-29:
squashfs in mounted RO for most part of system(overlay partition), changes are stored in ram, while ext4 is rw for all files, so any change in system & poweroff without proper shutdown = filesystem corruption. openmptcprouter is storing realtime info in wrong places i think(not in /tmp ?), because even routes for dynamic checks can't be added and there is error in syslog if sd is RO.
I think this is not good, router should be able to work without any problems without proper shutdown, its not a PC after all..

Ysurac, 2019-07-29:
For squashfs image, squashfs is always RO, but a F2FS overlay is created and should be RW. If it's not the case, check dmesg and you should see that F2FS have error and can't be mounted, so a RAMFS is used.
For ext4 image, ext4 is always RW, with journal on x86_64 and x86 platform, without journal for others platforms.

The problem is not an OpenMPTCProuter problem, OpenWrt have same problem, and any system have same problem and don't like power shutdown. Some platform are more sensible to that than others.

bjoerrrrn, 2019-07-29:
In my case its the ext4 image. 
I flashed sysupgrade, after boot it worked fine, but in the openmptcp statuspage it says that filesystem is ro - after another reboot config was gone. 

I never did an hard reset with power plug.  

Will try a factory flash with config restore before i finally virtualise my router after vacation. :-)

closed 2019-12-20 by Ysurac

Bumpalert, 2021-05-16:
I got this issue in 0.58beta2 two times. Is it root caused already @Ysurac ?
[image]

Bumpalert, 2021-05-16:
Another suspect I have is I tried it on PI4B 4gb version, while earlier experiments were on pi4b 2gb version

Ysurac, 2021-05-16:
As I answered it's due to power cut, bad sdcard,...

Bumpalert, 2021-05-23:
added a powered usb hub and problem solved in my case

eothein, 2021-08-09:
If multiple people are experciencing the problem, I guess the "Power cut, bad sdcard" does not seem to be the correct answer. I'm facing the same issue and don't know how to proceed now.

Dragonox77, 2021-08-09:
Try to reboot your pi and tell us if the problem still occur.

Ysurac, 2021-08-09:
It's the right answer. Problem is always due to power cut, bad sdcard or not enough power due to USB use.
RPI are all sensible to this issue. If you have no power problems and a good sdcard you will never face this issue on RPI.
Also check in system->Mount points that "Check filesystems before mount" is checked.

Adorfer, 2021-10-14:
1) 4,7V is below the lower limit. Starting fom 4.8V i see all strange of effect on RPI4 (on Raspberrian doing other stuff with multiple USB devices). I strongly recommend a power supply which delivers 5.2V to the Raspi, so that the USB (and system voltage) never drops below 4.9V. 
2) when using an external USB HUB, make sure it is a) running of 12V PSU b) not feeding back 5V to the Raspi (via the USB A)
3) squashfs is (YMMV) more robust against defects than ext4.

spacesuitdiver, 2021-10-15:
For some reason it's not listed in system however, here's the core temp command output:

```
root@OpenMPTCProuter:~# cat /sys/class/thermal/thermal_zone0/temp
43816
root@OpenMPTCProuter:~# vcgencmd measure_temp
temp=44.3'C
```

spacesuitdiver, 2021-10-16:
@Ysurac firmware update didn’t help. I can try underclocking next.

spacesuitdiver, 2021-10-18:
> For the FS Read Only, can you try to lower CPU speed, in System->OpenMPTCProuter, "Advanced settings" tab. Set it to 1000000 for both mini and max.

@Ysurac adjusting to `1000000` made the pi unreachable 😐

MRedZac, 2025-03-19:
The story with power or SD card is bullshit. Even with 5Ampere of power and a bunch of high quality SD cards it runs into the same issue... What is the sense of having bonded connections if not even the router is able to run stable on RPi4... I cannot tell you in words how angry I am, wasting hours of precious time on this crap !!!!

Dragonox77, 2025-03-19:
Hey, keep in mind this is an open source project.

I had this router working for 1y +, and most of the times it worked fine.
You should consider opening a new ticket, specifying which version you are running on. The dev behind this project is active, and i'm sure he will help you out.

Ysurac, 2025-03-20:
@MRedZac putting comment on closed issue only to put aggressive comment doesn't help.
I have a RPI4 running with OpenMPTCProuter and it's stable.