Ysurac/openmptcprouter #575: Problèmes de débit

Slivikk-fr, 2019-07-29:
## Actual Behavior
J'ai un débit qui est très bas (1.9mbits) alors que j'ai ma 4G Box de connecté. Comment je pourrais faire pour avoir le débit max de la 4G Box ?

## Speed Test
Résultat Speedtest 👍 
Ping : 60 MS
Débit Descendant : 3mbits (aussi rapide que mon ADSL génial 👍 )
Débit Montant : 0Mbits

## Steps to Reproduce the Problem

  1.--
  2.--
  3.--

## Specifications

  - OpenMPTCProuter version: (0.51.3)
  - OpenMPTCProuter VPS version: (0.999)
  - OpenMPTCProuter platform: (RPI3B+)

Ysurac, 2019-07-29:
Il y a des chances donc que l'agrégation ne fonctionne pas correctement.
Tout est vert dans la page status ?
Le VPS à l'autre bout est assez rapide ? avec le kernel MPTCP activé ? Et situé en France aussi ?
La configuration est celle par défaut avec shadowsocks et glorytun TCP ? sans obfuscation d'activer ? 
MPTCP est bien en fullmesh ? Les IPs bien présentent dans Réseau->MPTCP->MPTCP fullmesh ? Dans "Established connections" il y a bien des connexions ?
Sinon speedtest ne fournit pas des résultat très cohérent pour MPTCP. Il y a ```omr-test-speed``` sur le routeur qui permet de mesurer la vitesse de façon un peu plus fiable.

Slivikk-fr, 2019-07-29:
Oui tout est vert pourtant y'a que la 4G Box en ligne de mis pour l'instant :/ Et comment je regarde sir le kernel MPTCP est activé, VPS Situé dans le DC de OVH à Gravelines, je n'ai rien toucher à la config de shadowsocks et glotyrun, je ne sais pas si l'obfuscation est activé, MPTCP Fullmesh ok, les connexions sont bien établis partout.

Ysurac, 2019-07-29:
pour voir si le noyau est bien mis sur le VPS : ```uname -a```
Tu as que la 4G box pour le moment ?
Tu peux tester la vitesse de la 4G box toute seule en faisant ```omr-test-speed wan1```

Slivikk-fr, 2019-07-29:
Je n'ai pas encore ajouter mon téléphone, car j'ai reset encore une fois OpenMPTCPRouter

Slivikk-fr, 2019-07-29:
Point de vue VPS on est bon

Slivikk-fr, 2019-07-29:
Sur le Raspberry Pi je viens de voir que j'ai "r8152 1-1.1.3:1.0 :eth1: Promiscuous mode enabled"

Slivikk-fr, 2019-07-29:
Point de vu omr j'ai la même chose que Speedtest :(

Slivikk-fr, 2019-07-29:
Voici la config au cas où :

dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].boguspriv='1'
dhcp.@dnsmasq[0].filterwin2k='0'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].resolvfile='/tmp/resolv.conf.auto'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].server='127.0.0.1#5353'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].rebind_domain='plex.direct'
dhcp.@dnsmasq[0].ipset='/googlevideo.com/omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.cot/omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/steamcontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dropbear.@dropbear[0]=dropbear
dropbear.@dropbear[0].PasswordAuth='on'
dropbear.@dropbear[0].RootPasswordAuth='on'
dropbear.@dropbear[0].Port='22'
dscp.@classify[0]=classify
dscp.@classify[0].direction='both'
dscp.@classify[0].proto='icmp'
dscp.@classify[0].class='cs5'
dscp.@classify[0].comment='ICMP'
dscp.@classify[1]=classify
dscp.@classify[1].direction='both'
dscp.@classify[1].proto='udp'
dscp.@classify[1].class='cs5'
dscp.@classify[1].src_port='53,123,5353'
dscp.@classify[1].comment='DNS udp and NTP'
dscp.@classify[2]=classify
dscp.@classify[2].direction='both'
dscp.@classify[2].proto='tcp'
dscp.@classify[2].class='cs5'
dscp.@classify[2].src_port='53,5353'
dscp.@classify[2].comment='DNS tcp'
dscp.@classify[3]=classify
dscp.@classify[3].direction='both'
dscp.@classify[3].proto='tcp'
dscp.@classify[3].class='cs6'
dscp.@classify[3].dest_port='65001,65301'
dscp.@classify[3].comment='OMR vpn'
dscp.@classify[4]=classify
dscp.@classify[4].direction='both'
dscp.@classify[4].proto='udp'
dscp.@classify[4].class='cs6'
dscp.@classify[4].dest_port='65001'
dscp.@classify[4].comment='OMR vpn'
dscp.@domains[0]=domains
[... 1110 lines cut ...]
upnpd.config.internal_iface='lan'
upnpd.config.port='5000'
upnpd.config.upnp_lease_file='/var/run/miniupnpd.leases'
upnpd.config.igdv1='0'
upnpd.@perm_rule[0]=perm_rule
upnpd.@perm_rule[0].action='allow'
upnpd.@perm_rule[0].ext_ports='1024-65535'
upnpd.@perm_rule[0].int_addr='0.0.0.0/0'
upnpd.@perm_rule[0].int_ports='1024-65535'
upnpd.@perm_rule[0].comment='Allow high ports'
upnpd.@perm_rule[1]=perm_rule
upnpd.@perm_rule[1].action='deny'
upnpd.@perm_rule[1].ext_ports='0-65535'
upnpd.@perm_rule[1].int_addr='0.0.0.0/0'
upnpd.@perm_rule[1].int_ports='0-65535'
upnpd.@perm_rule[1].comment='Default deny'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='wan1' 'eth1'
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.channel='36'
wireless.radio0.hwmode='11a'
wireless.radio0.path='platform/soc/3f300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
wireless.radio0.htmode='VHT80'
wireless.radio0.disabled='1'

Ysurac, 2019-07-29:
Et ```omr-test-speed wan2``` donne quoi comme débit ?

Slivikk-fr, 2019-07-29:
Comme sur Speed-test.net  avec mon téléphone le débit est mieux mais je ne peut pas tout le temps mettre mon téléphone en partage de connexion histoire de compenser le débit qui manque encore 👎

Ysurac, 2019-07-29:
C'est à dire ? Quel est le résultat pour ```omr-test-speed wan2``` et celui pour ```omr-test-speed``` ?

Slivikk-fr, 2019-07-29:
Je suis obligé d'attendre qu'il fasse ces heures de speed test ?

Ysurac, 2019-07-29:
non tu laisses 1 ou 2 minutes puis control+c pour couper

Slivikk-fr, 2019-07-29:
~900K sur omr-test-speed
Et aucune idée pour eth1 ça me met des nombres comme 44205 - 24319 etc

Ysurac, 2019-07-29:
Avec un copier-coller ça aiderait ;)

Slivikk-fr, 2019-07-29:
Bah je fait que depuis le RPI directement je vais le faire en SSH alors ?

Ysurac, 2019-07-29:
oui

Slivikk-fr, 2019-07-29:
Ok je fait ça alors

Slivikk-fr, 2019-07-29:
ETH 1 (WAN2):

  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 10.0G    0 61.7M    0     0   809k      0  3:35:51  0:01:18  3:34:33 1028k

Sans rien (omr-test-speed)
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 10.0G    0 50.9M    0     0   816k      0  3:33:58  0:01:03  3:32:55  412k

Ysurac, 2019-07-29:
ok, donc même vitesse moyenne en direct ou en passant via le VPS. Donc OpenMPTCProuter fait ce qu'il peut avec ce qu'il a la ;)

Le soucis ne semble pas venir d'OpenMPTCProuter vu comme ça... Donc il faut vérifier les câbles, le placement de la box,...

Slivikk-fr, 2019-07-29:
Point de vue câble c'est neuf, bon je vais faire avec merci quand même :)

closed 2019-07-29 by Slivikk-fr