Ysurac/openmptcprouter #576: How to bypass MPTCP

JorgeMoort89, 2019-07-29:
I have one 4G mobile connection that block MPTCP aggregation, how can I move to other type of aggregation bypassing MPTCP? 
Can you make a guide in the Wiki for it?

## Specifications

  - OpenMPTCProuter version: 0.996
  - OpenMPTCProuter VPS version: 0.49
  - OpenMPTCProuter platform: x86_64

Ysurac, 2019-07-29:
First you have to be sure that it's really blocked.
If it's really the case, on the wizard disable shadowsocks and use Glorytun UDP.

JorgeMoort89, 2019-07-29:
I've checked with this command:
`root@router:~# omr-mptcp-intf eth6
Nay, Nay, Nay, your have an old computer that does not speak MPTCP. Shame on you!`

Ysurac, 2019-07-29:
This doesn't always give an right result.
You can also try ```omr-tracebox-mptcp your_vps_ip eth6```

JorgeMoort89, 2019-07-29:
Shadowsocks is disabled, Glorytun on UDP and switched to Glorytun UDP but is using only one connettivity with direct output and I don't know why

JorgeMoort89, 2019-07-29:
`root@router:~# omr-tracebox-mptcp VPS_IP eth6

`
Done this command multiple times, it return no response

Ysurac, 2019-07-29:
No answer = not sure.
For glorytun UDP there is a bug, this will be fixed in next release.

JorgeMoort89, 2019-07-29:
Ok, so MPTCP maybe isn't blocked by the ISP? I'm going to try with the latest relese (0.51.3).
How can temporary bond the connections without MPTCP and Glorytun UDP?

closed 2019-12-20 by Ysurac