Ysurac/openmptcprouter #577: Issues with Modems not Bonding

tranquiliza, 2019-07-29:
## Expected Behavior
Both modems are being used, splitting data evenly or at least just splitting some data.
  In English please.

## Actual Behavior
Only one modem is being used 

## Specifications

  - OpenMPTCProuter version: 0.49.7
  - OpenMPTCProuter VPS version: 0.9962 4.19.56 MPTCP
  - OpenMPTCProuter platform: Raspberry PI M3+ 

Why am I on V 0.49.7? A lot of UI has been rearranged on 0.50, 0.51, and I havnt seen 0.52. But I spent a week trying to figure out why half of the things werent working, had a lot of issues with 0.50+ versions unfortunately. 

I'm not super firm on this whole thing, so any help is appreciated. And I will provide all the information you need to solve this issue, just be very specific. 

If you need, I can provide direct access to the PI, but that will be shared in a private convo somewhere and not here.

tranquiliza, 2019-07-29:
[image]
dont worry too much about the "Me" block, I'm currently connected through the VPN from the interwebs (Easier than sitting on phone local network style)

tranquiliza, 2019-07-29:
TCP Checks: 

1: 192.168.8.1 IP::DiffServicesCP IP::CheckSum 
2: 10.253.7.246  [PARTIAL] +PartialTCP IP::DiffServicesCP IP::TTL IP::CheckSum 
3: 94.144.63.189  [PARTIAL] +PartialTCP IP::DiffServicesCP IP::TTL IP::CheckSum 
4: 148.122.9.26  [PARTIAL] +PartialTCP IP::DiffServicesCP IP::TTL IP::CheckSum 
5: 148.122.9.25  [PARTIAL] +PartialTCP IP::DiffServicesCP IP::TTL IP::CheckSum 
6: 146.172.105.21  [PARTIAL] +PartialTCP IP::DiffServicesCP IP::TTL IP::CheckSum 
7: * 
8: * 
9: * 
10: * 
11: * 
12: * 
13: * 
14: * 
15: * 
16: * 
17: * 
18: * 
19: * 
20: * 
21: 78.141.203.175 TCP::SrcPort TCP::DstPort TCP::SeqNumber TCP::AckNumber TCP::DataOffset TCP::Flags TCP::WindowsSize TCP::CheckSum IP::TotalLength IP::Identification IP::TTL IP::CheckSum IP::SourceIP IP::DestinationIP +TCPOptionMaxSegSize TCPOptionMPTCPCapable::Sender's Key 

Result:
< Ethernet (14 bytes) :: DestinationMAC = 0c:5b:8f:27:9a:64 , SourceMAC = dc:09:4c:29:69:4b , Type = 0x800 , >
< IP (20 bytes) :: Version = 4 , HeaderLength = 5 , DiffServicesCP = 0 , ExpCongestionNot = 0 , TotalLength = 56 , Identification = 0x0 , Flags = 2 , FragmentOffset = 0 , TTL = 54 , Protocol = 0x6 , CheckSum = 0x61d9 , SourceIP = 78.141.203.175 , DestinationIP = 192.168.8.2 , >
< TCP (20 bytes) :: SrcPort = 65101 , DstPort = 64194 , SeqNumber = 253116765 , AckNumber = 1764536800 , DataOffset = 9 , Reserved = 0 , Flags = ( SYN ACK ) , WindowsSize = 28800 , CheckSum = 0x8280 , UrgPointer = 0 , >
< TCPOptionMaxSegSize (4 bytes) :: Kind = 2 , Length = 4 , MaxSegSize = 1460 , >
< TCPOptionMPTCPCapable (12 bytes) :: Kind = 30 , Length = 12 , Subtype = 0 , Version = 0 , Checksum = 1 (Checksum Enabled) , Flags = 0 , Crypto = 1 (HMAC-SHA1) , Sender's Key = Sender's Key = 11458013342367409684 , >

And Second: 

1: 192.168.9.1 IP::DiffServicesCP IP::CheckSum 
2: * 
3: 62.44.164.113 IP::TTL IP::CheckSum 
4: 62.44.166.56  [PARTIAL] +PartialTCP IP::TTL IP::CheckSum 
5: 212.97.200.73  [PARTIAL] +PartialTCP IP::TTL IP::CheckSum 
6: 213.248.66.145  [PARTIAL] +PartialTCP IP::TTL IP::CheckSum 
7: 213.248.99.149  [PARTIAL] +PartialTCP IP::TTL IP::CheckSum 
8: * 
9: * 
10: * 
11: * 
12: * 
13: * 
14: * 
15: * 
16: * 
17: * 
18: * 
19: * 
20: * 
21: * 
22: 78.141.203.175 TCP::SrcPort TCP::DstPort TCP::SeqNumber TCP::AckNumber TCP::DataOffset TCP::Flags TCP::WindowsSize TCP::CheckSum IP::TotalLength IP::Identification IP::TTL IP::CheckSum IP::SourceIP IP::DestinationIP +TCPOptionMaxSegSize TCPOptionMPTCPCapable::Sender's Key 

Result:
< Ethernet (14 bytes) :: DestinationMAC = 48:73:c0:41:07:9f , SourceMAC = d0:d7:83:6e:36:01 , Type = 0x800 , >
< IP (20 bytes) :: Version = 4 , HeaderLength = 5 , DiffServicesCP = 0 , ExpCongestionNot = 0 , TotalLength = 56 , Identification = 0x0 , Flags = 2 , FragmentOffset = 0 , TTL = 48 , Protocol = 0x6 , CheckSum = 0x66d9 , SourceIP = 78.141.203.175 , DestinationIP = 192.168.9.2 , >
< TCP (20 bytes) :: SrcPort = 65101 , DstPort = 7170 , SeqNumber = 3181906823 , AckNumber = 1189479447 , DataOffset = 9 , Reserved = 0 , Flags = ( SYN ACK ) , WindowsSize = 28800 , CheckSum = 0xfbb4 , UrgPointer = 0 , >
< TCPOptionMaxSegSize (4 bytes) :: Kind = 2 , Length = 4 , MaxSegSize = 1400 , >
< TCPOptionMPTCPCapable (12 bytes) :: Kind = 30 , Length = 12 , Subtype = 0 , Version = 0 , Checksum = 1 (Checksum Enabled) , Flags = 0 , Crypto = 1 (HMAC-SHA1) , Sender's Key = Sender's Key = 10167035833971140169 , >

Ysurac, 2019-07-30:
I have no idea if this can work with your config. You should really update, because release < 0.50 are 4.14 based.

How do you check that it's not working ? What are the result, via SSH on the router, of ```omr-test-speed```, ```om-test-speed wan1``` and ```om-test-speed wan2``` ?

tranquiliza, 2019-07-30:
i look at the realtime graphs displaying the usage of upload/download on mptcp, and its only loading on one of them

what does it mean that its 4.14 based? remember i do not how the same insider knowledge as ypu do.

And all configuration is done via UI.

I would upgrade to higher, if it wasnt due to the amount of problems introduced on the later versions. Like missing UI for configuration of port forwarding etc.

tranquiliza, 2019-07-30:
[image]

With shadowsocks turned off, using MLVPN. Still only one modem of the 2 being used.

Ysurac, 2019-07-30:
I asked for some info, I don't have the result. How do you want me to help on this old no more supported release ?

tranquiliza, 2019-07-30:
Apologies: Running those tests now: 

OMR-TEST-SPEED:          
[image]

OM-TEST-SPEED WAN1: Command does not work?
OM-TEST-SPEED WAN2: Command does not work?

Note looking at UI while speedtest: 
[image]

Will try to see if I can figure out why commands doesnt work.

tranquiliza, 2019-07-30:
I understand: Typos. 

[image]

Both modems seem to work fine on their own. But they never combine?

tranquiliza, 2019-07-30:
So I just found a spare SD, and installed latest (0.51.3) and reconfigured everything. I have internet access. 

This version claims that one of the modems is MPTCP blocked somewhere in the network
Going to try to turn off ShadowSocks and put Glorytun UDP. 

Does these changes require VPS changes too?

Ysurac, 2019-07-30:
Current release is 0.51.4 that fixes some problems with Glorytun UDP.
Latest VPS release is 0.999

tranquiliza, 2019-07-30:
Oh apologies, I did install 0.51.4 (Double checking the file name haha) (openmptcprouter-v0.51.4-r010518-6d61fcfa9e-brcm2708-bcm2710-rpi-3-ext4-factory) specifically

tranquiliza, 2019-07-30:
VPS is running outdated version, will hear host if possible to update soon too.

tranquiliza, 2019-07-30:
Additionally: Now im running into the issue I had prior to downgrading, while running ShadowSocks Off and Glorytun UDP: 

I can reach Google and open the page, and search, but I am unable to navigate to anything else. 
Any ideas?

Ysurac, 2019-07-30:
This can be a MTU problem, even this should be set automatically, you can set them in  network->interface. Try a low value like 1300 if this work try 1400.

tranquiliza, 2019-07-30:
I went away from the setup for a few, came back and tried accessing things and now I have access. Without changing any settings. 
However, doing the speed test, it still looks like just one Modem is being used and not both. 

Console is however, when im doing the omr-test-speed commands, no replying: 

ipset v7.1: The set with the given name does not exist. 

Not sure what this means.

tranquiliza, 2019-07-30:
Tried the MTU, no changes. After reboot, I have no connection at all now.

closed 2019-12-20 by Ysurac