Ysurac/openmptcprouter #578: How to change Shadowsocks to port 443?

Bestle, 2019-07-29:
I am attempting to run shadowsocks through port 443 but for some reason OMR cannot connect to it after I do the following steps...

## Steps to Reproduce the Problem

  1. Change the server port on VPS in /etc/shadowsocks-libev/config.json to 443
  2. Reboot VPS
  3. Change the shadowsocks port to 443 via /luci/admin/services/shadowsocks-libev/servers
  4. Reboot router

After router reboot the status page shows that shadowsocks is unavailable. Am I just missing a step somewhere here? If I change them both back to 65101 everything works again.

## Specifications

  - OpenMPTCProuter version: 0.51
  - OpenMPTCProuter VPS version: VPS 0.9962
  - OpenMPTCProuter platform: x86_64

maxou56, 2019-08-12:
In SSH
VPS: change port in "/root/openmptcprouter_config.txt" and "/etc/shadowsocks-libev/config.json"
Router: change port in "/etc/config/shadowsocks-libev"

Ysurac, 2019-08-13:
Changing the port via the interface should be enough. I need to make some test...

closed 2019-12-20 by Ysurac